linux-nvme/nvme-cli #1272: Configurator Segmentation Fault during make

jeffreyalien, 2021-12-09:
I'm seeing the following segmentation fault on 2 different Redhat systems when doing a make:
[root@ddrh-jlien nvmeCliJal]# make
NVME_VERSION = 1.14.265.g1a60.dirty
    CC ccan/tools/configurator/configurator
ccan/tools/configurator/configurator > ccan/config.h
sh: line 1: 708930 Segmentation fault      (core dumped) ./configurator.out 2>&1

On one system (running a customized 5.10.0 kernel), the make continues on with out error and completes successfully.   On the other system (running the standard 4.18.0-305.7.1.el8 kernel), the make fails with a bunch of conflicting types errors (see example below).  I'm not sure if they segmentation fault is causing these errors or if they are a separate problem, but would like to understand and fix the configurator issue first.   

**Conflict types error message example**
    CC nvme.o
In file included from common.h:4,
                 from nvme.c:60:
ccan/ccan/endian/endian.h:243:25: error: conflicting types for cpu_to_le64
 static inline leint64_t cpu_to_le64(uint64_t native)
                         ^~~~~~~~~~~
In file included from libnvme/src/nvme/ioctl.h:15,
                 from libnvme/src/nvme/tree.h:20,
                 from nvme.c:27:
libnvme/src/nvme/types.h:71:22: note: previous definition of cpu_to_le64 was here
 static inline __le64 cpu_to_le64(uint64_t x)
                      ^~~~~~~~~~~
In file included from common.h:4,
                 from nvme.c:60:
ccan/ccan/endian/endian.h:252:25: error: redefinition of cpu_to_le32
 static inline leint32_t cpu_to_le32(uint32_t native)
                         ^~~~~~~~~~~

igaw, 2021-12-10:
Can you try the meson build instead? The plan is to drop make support for meson.

jeffreyalien, 2021-12-10:
I was able to build successfully with meson.  So if the plan is to drop make support for meson, what is the long term plan?  Are you planning to fix the makefile so the "make" command will work in the master branch as it does in the monolithic branch?

keithbusch, 2021-12-10:
Is it possible to just have "make" be a wrapper for the meson build system? Just let the default target for "make" execute "meson" with sane defaults?

igaw, 2021-12-10:
Correct, the monolithic branch will stay on make. It's in pure maintenance mode (at least from my point of view).

I am building libnvme and nvme-cli since a few weeks only with meson and I hope I was able to addressed most of the problems. I also verified that it's possible to package it (https://build.opensuse.org/package/show/home:wagi:branches:openSUSE:Factory). 

That said I think it would be possible to drop the Makefiles and just use meson. And yes having a small makefile wrapper around is no problem.

```
BUILD_DIR=.build

all: $(BUILD_DIR)/build.ninja
	ninja -C $(BUILD_DIR)

check:
	meson test -C $(BUILD_DIR) -v

clean:
	rm -rf $(BUILD_DIR)

$(BUILD_DIR)/build.ninja:
	meson $(BUILD_DIR)

.PHONY: check clean
```

jeffreyalien, 2021-12-13:
This plan sounds good to me.

sc108-lee, 2021-12-15:
In my case, the segmentation fault happens from HAVE_POINTER_SAFE_MAKECONTEXT test.

johnmeneghini, 2021-12-22:
I think the problem with the configurator utility has to do with a problem in the library.

```
[55182.648896] configurator.ou[2799]: segfault at 403f90 ip 00007f1aa927573b sp 0000000000403f80 error 7 in ld-linux-x86-64.so.2[7f1aa9]
[55182.663867] Code: 24 08 e9 e8 fb ff ff 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 41 57 41 56 41 55 41 54 49 89 fc 55 48 89 d5 53 48 81 ec 98 00 00 00 <48> 89 74 24 f
```

igaw, 2021-12-23:
I was not able to reproduce this problem so far. It seems to appear on RHEL/Centos systems. There was also a bug report on memory leaks on the configurator utility. Maybe you could run it with valgrind etc to see what's triggering the crash.

PR #1306 (2022-01-07): build: Replacing make by meson.
This involves replacing all Makefiles by meson.build files.
Also the top-level configure script now simply invokes meson.
The top-level Makefile is just a wrapper for meson. It supports
the following operations:

make
make test
make install
make clean    # Remove build artifacts but keep configuration
make purge    # Remove build artifacts and configuration

Based on Martin Belanger's libnvme work.

Fix #1272
closed 2022-01-07 by igaw