linux-nvme/nvme-cli #1289: 'nvme list-subsys' shows connecting state wrong

igaw, 2021-12-20:
When a path is down (controller state is in connecting), 'nvme list-subsys' does not show the state correctly:

```
# nvme list-subsys                                                                                                                   
nvme-subsys8 - NQN=nqn.2014-08.org.nvmexpress:NVMf:uuid:de63429f-50a4-4e03-ade6-0be27b75be77                                                                     
\                                                                                                                                                                
 +- (null) tcp traddr=10.161.8.24,trsvcid=4420 (null)                                                                                                            
 +- nvme8 tcp traddr=10.161.8.24,trsvcid=4421 live     

```

This is what version 1.x shows
```
# nvme list-subsys
nvme-subsys8 - NQN=nqn.2014-08.org.nvmexpress:NVMf:uuid:de63429f-50a4-4e03-ade6-0be27b75be77
\
 +- nvme8 tcp traddr=10.161.8.24 trsvcid=4421 live 
 +- nvme9 tcp traddr=10.161.8.24 trsvcid=4420 connecting 
```

hreinecke, 2022-01-11:
It's not so much the missing commits, it's a quirk in libnvme.
nvme_ns_open() is calling nvme_ns_init(), which will fail if it can't send an 'identify' command. So scan_namespace will fail, and the namespace won't be found when searching the topology tree.

hreinecke, 2022-01-21:
This is now https://github.com/linux-nvme/libnvme/pull/190

hreinecke, 2022-01-21:
Ah. It's the _controller_ which fails to initialize, not the namespace. Of course.

hreinecke, 2022-01-21:
Fixed now in https://github.com/linux-nvme/libnvme/pull/194

hreinecke, 2022-01-21:
Merged now; guess we'll need to update libnvme.wrap.

hreinecke, 2022-01-21:
With the latest update I get

```
./nvme list-subsys
nvme-subsys0 - NQN=nqn.1996-04.de.suse:albeniz:vm-2
\
 +- nvme1 tcp traddr=192.168.20.11,trsvcid=4420 live
 +- nvme0 tcp traddr=192.168.10.11,trsvcid=4420 connecting

```

closed 2022-01-21 by hreinecke