linux-nvme/nvme-cli #1293: Build failing on Centos 6.10

brendon, 2021-12-21:
### CentOS release 6.10 (Final)

I'm getting an error when trying to run `make`.

```
git clone --recurse-submodules https://github.com/linux-nvme/nvme-cli
Cloning into 'nvme-cli'...
remote: Enumerating objects: 14184, done.
remote: Counting objects: 100% (53/53), done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 14184 (delta 20), reused 38 (delta 17), pack-reused 14131
Receiving objects: 100% (14184/14184), 6.67 MiB | 10.71 MiB/s, done.
Resolving deltas: 100% (10448/10448), done.
Submodule 'libnvme' (https://github.com/linux-nvme/libnvme) registered for path 'libnvme'
Cloning into '/root/nvme-cli/libnvme'...
remote: Enumerating objects: 2646, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 2646 (delta 5), reused 9 (delta 3), pack-reused 2627
Receiving objects: 100% (2646/2646), 913.93 KiB | 16.03 MiB/s, done.
Resolving deltas: 100% (1689/1689), done.
Submodule path 'libnvme': checked out 'cfc9f9be07f365e07a8c5a27d2ddb9e7bbae5eae'
```

```
[~/nvme-cli]# make
NVME_VERSION = 1.14.329.g9472
    CC ccan/tools/configurator/configurator
ccan/tools/configurator/configurator > ccan/config.h
    CC nvme.o
make -C libnvme/
make[1]: Entering directory `/root/nvme-cli/libnvme'
Makefile:34: config-host.mak: No such file or directory
Running configure ...
prefix                        /usr
includedir                    /usr/include
libdir                        /usr/lib
mandir                        /usr/man
datadir                       /usr/share
libuuid                       no
libjson-c                     no
cpp                           yes
make[1]: Leaving directory `/root/nvme-cli/libnvme'
make[1]: Entering directory `/root/nvme-cli/libnvme'
sed -e "s%@prefix@%/usr%g" -e "s%@libdir@%/usr/lib%g" -e "s%@includedir@%/usr/include%g" -e "s%@NAME@%libnvme%g" -e "s%@VERSION@%0.1%g" libnvme.pc.in >libnvme.pc
make[2]: Entering directory `/root/nvme-cli/libnvme/src'
cc -D_GNU_SOURCE    ../ccan/tools/configurator/configurator.c   -o ../ccan/tools/configurator/configurator
../ccan/tools/configurator/configurator > ../ccan/ccan-config.h
     CC ../ccan/ccan/list/list.ol
     CC ../ccan/ccan/str/debug.ol
     CC ../ccan/ccan/str/str.ol
     CC nvme/linux.ol
     CC nvme/ioctl.ol
     CC nvme/filters.ol
     CC nvme/fabrics.ol
nvme/fabrics.c: In function ‘uuid_from_product_uuid’:
nvme/fabrics.c:909: error: ‘for’ loop initial declarations are only allowed in C99 mode
nvme/fabrics.c:909: note: use option -std=c99 or -std=gnu99 to compile your code
make[2]: *** [nvme/fabrics.ol] Error 1
make[2]: Leaving directory `/root/nvme-cli/libnvme/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/root/nvme-cli/libnvme'
make: *** [libnvme] Error 2
```

brendon, 2021-12-21:
I've tested your fix in linux-nvme/libnvme#153 and the output is now:

```
$ make

NVME_VERSION = 1.14.329.g9472.dirty
make -C libnvme/
make[1]: Entering directory `/root/nvme-cli/libnvme'
make[2]: Entering directory `/root/nvme-cli/libnvme/src'
     CC nvme/fabrics.ol
     CC nvme/util.ol
     CC nvme/tree.ol
nvme/tree.c: In function ‘nvme_ctrl_alloc’:
nvme/tree.c:1209: warning: ‘e’ may be used uninitialized in this function
     CC nvme/log.ol
     CC nvme/cleanup.ol
     AR libnvme.a
ar: creating libnvme.a
 RANLIB libnvme.a
     CC nvme/linux.os
     CC nvme/ioctl.os
     CC nvme/filters.os
     CC nvme/fabrics.os
     CC nvme/util.os
     CC nvme/tree.os
nvme/tree.c: In function ‘nvme_ctrl_alloc’:
nvme/tree.c:1209: warning: ‘e’ may be used uninitialized in this function
     CC nvme/log.os
     CC nvme/cleanup.os
     CC ../ccan/ccan/list/list.os
     CC ../ccan/ccan/str/debug.os
     CC ../ccan/ccan/str/str.os
     CC libnvme.so.1.0.1
nvme/ioctl.os: In function `nvme_feature_decode_arbitration':
/root/nvme-cli/libnvme/src/nvme/util.h:120: multiple definition of `nvme_feature_decode_arbitration'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:120: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_power_mgmt':
/root/nvme-cli/libnvme/src/nvme/util.h:131: multiple definition of `nvme_feature_decode_power_mgmt'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:131: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_lba_range':
/root/nvme-cli/libnvme/src/nvme/util.h:139: multiple definition of `nvme_feature_decode_lba_range'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:139: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_temp_threshold':
/root/nvme-cli/libnvme/src/nvme/util.h:150: multiple definition of `nvme_feature_decode_temp_threshold'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:150: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_error_recovery':
/root/nvme-cli/libnvme/src/nvme/util.h:159: multiple definition of `nvme_feature_decode_error_recovery'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:159: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_volatile_write_cache':
/root/nvme-cli/libnvme/src/nvme/util.h:167: multiple definition of `nvme_feature_decode_volatile_write_cache'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:167: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_number_of_queues':
/root/nvme-cli/libnvme/src/nvme/util.h:175: multiple definition of `nvme_feature_decode_number_of_queues'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:175: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_interrupt_coalescing':
/root/nvme-cli/libnvme/src/nvme/util.h:184: multiple definition of `nvme_feature_decode_interrupt_coalescing'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:184: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_interrupt_config':
/root/nvme-cli/libnvme/src/nvme/util.h:193: multiple definition of `nvme_feature_decode_interrupt_config'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:193: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_write_atomicity':
/root/nvme-cli/libnvme/src/nvme/util.h:201: multiple definition of `nvme_feature_decode_write_atomicity'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:201: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_async_event_config':
/root/nvme-cli/libnvme/src/nvme/util.h:218: multiple definition of `nvme_feature_decode_async_event_config'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:218: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_auto_power_state':
/root/nvme-cli/libnvme/src/nvme/util.h:231: multiple definition of `nvme_feature_decode_auto_power_state'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:231: first defined here
nvme/ioctl.os: In function `nvme_feature_decode_host_memory_buffer':
[... 337 lines cut ...]
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:300: first defined here
nvme/tree.os: In function `nvme_feature_decode_software_progress_marker':
/root/nvme-cli/libnvme/src/nvme/util.h:308: multiple definition of `nvme_feature_decode_software_progress_marker'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:308: first defined here
nvme/tree.os: In function `nvme_feature_decode_host_identifier':
/root/nvme-cli/libnvme/src/nvme/util.h:315: multiple definition of `nvme_feature_decode_host_identifier'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:315: first defined here
nvme/tree.os: In function `nvme_feature_decode_reservation_notification':
/root/nvme-cli/libnvme/src/nvme/util.h:325: multiple definition of `nvme_feature_decode_reservation_notification'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:325: first defined here
nvme/tree.os: In function `nvme_feature_decode_reservation_persistance':
/root/nvme-cli/libnvme/src/nvme/util.h:334: multiple definition of `nvme_feature_decode_reservation_persistance'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:334: first defined here
nvme/tree.os: In function `nvme_feature_decode_namespace_write_protect':
/root/nvme-cli/libnvme/src/nvme/util.h:341: multiple definition of `nvme_feature_decode_namespace_write_protect'
nvme/linux.os:/root/nvme-cli/libnvme/src/nvme/util.h:341: first defined here
collect2: ld returned 1 exit status
make[2]: *** [libnvme.so.1.0.1] Error 1
make[2]: Leaving directory `/root/nvme-cli/libnvme/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/root/nvme-cli/libnvme'
make: *** [libnvme] Error 2
```

I can't quite make out what the error is here. Let me know if you want me to run anything further :)

igaw, 2021-12-22:
It's probably due to the inline function.  They should properly declared as 'static inline'.

brendon, 2021-12-23:
After fixing that additional line:

```
static inline void nvme_feature_decode_host_identifier(__u32 value, bool *exhid)
{
        *exhid = NVME_FEAT_HOSTID_EXHID(value);
}
```

The compliation gets much further but complains about the missing realtime library akin to this I think:

https://stackoverflow.com/questions/2418157/c-error-undefined-reference-to-clock-gettime-and-clock-settime

Probably due to an older gcc being used. Not sure if you want to explicitly include this library?

```
$ make
make -C libnvme/
make[1]: Entering directory `/root/nvme-cli/libnvme'
make[2]: Entering directory `/root/nvme-cli/libnvme/src'
     CC nvme/linux.ol
     CC nvme/ioctl.ol
     CC nvme/filters.ol
     CC nvme/fabrics.ol
     CC nvme/util.ol
     CC nvme/tree.ol
nvme/tree.c: In function ‘nvme_ctrl_alloc’:
nvme/tree.c:1209: warning: ‘e’ may be used uninitialized in this function
     CC nvme/log.ol
     CC nvme/cleanup.ol
     AR libnvme.a
ar: creating libnvme.a
 RANLIB libnvme.a
     CC nvme/linux.os
     CC nvme/ioctl.os
     CC nvme/filters.os
     CC nvme/fabrics.os
     CC nvme/util.os
     CC nvme/tree.os
nvme/tree.c: In function ‘nvme_ctrl_alloc’:
nvme/tree.c:1209: warning: ‘e’ may be used uninitialized in this function
     CC nvme/log.os
     CC nvme/cleanup.os
     CC libnvme.so.1.0.1
make[2]: Leaving directory `/root/nvme-cli/libnvme/src'
make[2]: Entering directory `/root/nvme-cli/libnvme/test'
     CC test
test.c: In function ‘test_ctrl’:
test.c:44: warning: missing braces around initializer
test.c:44: warning: (near initialization for ‘ns_list.ns’)
test.c:45: warning: missing braces around initializer
test.c:45: warning: (near initialization for ‘cfx.acs’)
test.c:52: warning: missing braces around initializer
test.c:52: warning: (near initialization for ‘uuid.rsvd0’)
../src//libnvme.a(log.ol): In function `__nvme_msg':
/root/nvme-cli/libnvme/src/nvme/log.c:63: undefined reference to `clock_gettime'
collect2: ld returned 1 exit status
make[2]: *** [test] Error 1
make[2]: Leaving directory `/root/nvme-cli/libnvme/test'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/root/nvme-cli/libnvme'
make: *** [libnvme] Error 2
```

igaw, 2021-12-23:
glibc seems to made the -lrt optional in 2012. That is almost 10 years ago. 

In theory this should `LDFLAGS=-lrt make` do the trick.

Given that we are about to replace the Makefile based build system with meson, could you check what happens if you build it with meson instead?

brendon, 2021-12-23:
I installed the system version of meson (Centos 6 - I wish I didn't have to use this OS but I'm stuck with it for now and am trying to switch to a newer EC2 type) and ran it but I suspect it's too ancient:

```
$ meson compile
The Meson build system
Version: 0.44.1
Source dir: /root/nvme-cli
Build dir: /root/nvme-cli/compile
Build type: native build

Meson encountered an error in file meson_options.txt, line 7, column 0:
Unknown type feature.

A full log can be found at /root/nvme-cli/compile/meson-logs/meson-log.txt
```

I'll look into installing a newer version of meson. Installing a newer version of python seems like a challenge too lol!

As far as the regular compile goes, it gets  this far now:

```
$ LDFLAGS=-lrt make
NVME_VERSION = 1.14.332.gefed8.dirty
make -C libnvme/
make[1]: Entering directory `/root/nvme-cli/libnvme'
make[2]: Entering directory `/root/nvme-cli/libnvme/src'
     CC nvme/linux.ol
     CC nvme/ioctl.ol
     CC nvme/filters.ol
     CC nvme/fabrics.ol
     CC nvme/util.ol
     CC nvme/tree.ol
nvme/tree.c: In function ‘nvme_ctrl_alloc’:
nvme/tree.c:1209: warning: ‘e’ may be used uninitialized in this function
     CC nvme/log.ol
     CC nvme/cleanup.ol
     AR libnvme.a
ar: creating libnvme.a
 RANLIB libnvme.a
     CC nvme/linux.os
     CC nvme/ioctl.os
     CC nvme/filters.os
     CC nvme/fabrics.os
     CC nvme/util.os
     CC nvme/tree.os
nvme/tree.c: In function ‘nvme_ctrl_alloc’:
nvme/tree.c:1209: warning: ‘e’ may be used uninitialized in this function
     CC nvme/log.os
     CC nvme/cleanup.os
     CC ../ccan/ccan/list/list.os
     CC ../ccan/ccan/str/debug.os
     CC ../ccan/ccan/str/str.os
     CC libnvme.so.1.0.1
/usr/bin/ld: cannot find -lnvme
collect2: ld returned 1 exit status
make[2]: *** [libnvme.so.1.0.1] Error 1
make[2]: Leaving directory `/root/nvme-cli/libnvme/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/root/nvme-cli/libnvme'
make: *** [libnvme] Error 2
```

igaw, 2021-12-23:
- Would PyPI work for you?  `pip3 install --user meson`
- The search path for the library seems not working. One workaround would be to install libnvme as seperate step. This is what you would do anyway.

igaw, 2021-12-23:
BTW, the warning `nvme/tree.c:1209: warning: ‘e’ may be used uninitialized in this function` is a false positive. strtok_r() last argument ('e') is the state variable which doesn't need to be initialized according the documentation.

brendon, 2021-12-23:
Thanks @igaw :D Man have I gone down a rabbit hole of outdated dependencies! :D Right now I'm trying to get python 3.10 to compile with ssl so that I can install a newer meson. Will keep hacking at it!

brendon, 2021-12-23:
Some notes so far:

I compiled libnvme separately: entered `libnvme/`, ran `LDFLAGS=-lrt make`, then `make install`

Back in `/` I attempted to build again with `LDFLAGS=-lrt make` but it complained about a missing uuid library so:

`yum install libuuid libuuid-devel`

Now I think I've reached the end of the line with a dependency that Centos 6 doesn't provide:

```
LDFLAGS=-lrt make
make -C libnvme/
make[1]: Entering directory `/root/nvme-cli/libnvme'
make[2]: Entering directory `/root/nvme-cli/libnvme/src'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/root/nvme-cli/libnvme/src'
make[2]: Entering directory `/root/nvme-cli/libnvme/test'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/root/nvme-cli/libnvme/test'
make[2]: Entering directory `/root/nvme-cli/libnvme/examples'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/root/nvme-cli/libnvme/examples'
make[1]: Leaving directory `/root/nvme-cli/libnvme'
    CC nvme-rpmb.o
nvme-rpmb.c:26:26: error: linux/if_alg.h: No such file or directory
nvme-rpmb.c: In function ‘create_hash’:
nvme-rpmb.c:63: error: variable ‘provider_sa’ has initializer but incomplete type
nvme-rpmb.c:64: error: unknown field ‘salg_family’ specified in initializer
cc1: warnings being treated as errors
nvme-rpmb.c:64: error: excess elements in struct initializer
nvme-rpmb.c:64: error: (near initialization for ‘provider_sa’)
nvme-rpmb.c:65: error: unknown field ‘salg_type’ specified in initializer
nvme-rpmb.c:65: error: excess elements in struct initializer
nvme-rpmb.c:65: error: (near initialization for ‘provider_sa’)
nvme-rpmb.c:66: error: unknown field ‘salg_name’ specified in initializer
nvme-rpmb.c:66: error: extra brace group at end of initializer
nvme-rpmb.c:66: error: (near initialization for ‘provider_sa’)
nvme-rpmb.c:66: error: excess elements in struct initializer
nvme-rpmb.c:66: error: (near initialization for ‘provider_sa’)
nvme-rpmb.c:63: error: storage size of ‘provider_sa’ isn’t known
nvme-rpmb.c:86: error: ‘ALG_SET_KEY’ undeclared (first use in this function)
nvme-rpmb.c:86: error: (Each undeclared identifier is reported only once
nvme-rpmb.c:86: error: for each function it appears in.)
nvme-rpmb.c:63: error: unused variable ‘provider_sa’
make: *** [nvme-rpmb.o] Error 1
```

Will I likely encounter this same problem with `meson`? If so, I'll probably need to abandon ship :D

brendon, 2021-12-23:
I ended up rolling back to `1.13` and that compiled well :)

eli-schwartz, 2021-12-24:
I cannot find out from repology what version of python3 is available on Centos 6 -- it claimed the only available python version is python2. :/

meson 0.47 (which this project needs) requires python 3.5, meson 0.45 is the first version that bumped up from python 3.4 to python 3.5

If that doesn't help, what about switching tracks? https://sr.ht/~lattis/muon/ -- an implementation of meson that is written in c99 with no dependencies (in order to handle pkg-config, it requires building against libpkgconf though, but that is also fairly easy to get; it also has optional dependencies on curl and libarchive to do wrap dependency resolution).

I was able to build nvme-cli with minimal modifications using: `muon setup -Dlibnvme:python=false -Dopenssl=disabled builddir/`. It appears to choke on openssl's alphabetical versioning. ;)

There are two bugs in nmve-cli's meson.build which muon's stricter typechecking uncovered, and additionally muon doesn't support the `meson.project_license()` function but that would be easy to add. I temporarily hacked around it by moving the license string down into the relevant call sites.

eli-schwartz, 2021-12-24:
I submitted a PR at #1300 to fix those errors.

eli-schwartz, 2021-12-24:
> and additionally muon doesn't support the `meson.project_license()` function but that would be easy to add. I temporarily hacked around it by moving the license string down into the relevant call sites.

I submitted a patch to add this API to muon: https://lists.sr.ht/~lattis/muon/patches/27628

After that and the PR I linked in the previous comment, you should be able to build albeit without openssl support using muon.

brendon, 2021-12-24:
Amazing @eli-schwartz! Thank you so much for pursuing this. Way above and beyond! I'll give it a go as soon as I can :)

eli-schwartz, 2021-12-24:
Speaking with my Meson maintainer hat on, I want Meson to work out for people :)

And also I am a big fan of having multiple independent implementations for robustness reasons, which I think improves the ecosystem. I send patches for muon whenever I have time and my attention is pointed to a particular lacking feature. Sometimes the cross-pollination of build system development uncovers bugs in meson's implementation as well.

It is *very* nice to have a build system that doesn't depend on implementation details like "written in sh and m4", which has that option of reimplementing it.

igaw, 2021-12-27:
FWIW, I was able to build nvme-cli using muon. Though there were plenty warnings. muon seems to add '-Wpedantic' per default.

igaw, 2021-12-27:
> I ended up rolling back to `1.13` and that compiled well :)

The monolithic branch is the 1.x version which entered maintenance mode. That means no new feature, just bug fixes. Furthermore, don't expect that it lives for very long. The master branch is the upcoming 2.x version with meson dependency. So as long you are happy with the feature set from the 1.x version there is no need to deploy the 2.x version

Sure it's good to know that muon can be used alternative implementation.

annacrombie, 2021-12-27:
> FWIW, I was able to build nvme-cli using muon. Though there were plenty warnings. muon seems to add '-Wpedantic' per default.

muon maintainer here.  By default, muon sets `warning_level` to 3 (the highest).  This is why `-Wpedantic` is getting added.  You could adjust your default warning level with the `default_options` argument to `project()`.

eli-schwartz, 2021-12-27:
> FWIW, I was able to build nvme-cli using muon.

My patch to add project_license() was merged. :)

igaw, 2021-12-27:
> > FWIW, I was able to build nvme-cli using muon.
> 
> My patch to add project_license() was merged. :)

That is what I wanted to say. I was able to build nvme-cli with todays version of muon :)

igaw, 2021-12-27:
FWIW, I've just added two PRs with some more cleanups: #1301 and https://github.com/linux-nvme/libnvme/pull/159

igaw, 2021-12-27:
> muon maintainer here. By default, muon sets `warning_level` to 3 (the highest). This is why `-Wpedantic` is getting added. You could adjust your default warning level with the `default_options` [...]

Thanks for the tip. With the PRs mention above, the build is clean. Just one thing I noticed:

`cc1plus: warning: command-line option ‘-std=gnu99’ is valid for C/ObjC but not for C++`

Not a big deal, just wanted to let you know :)

annacrombie, 2021-12-27:
> `cc1plus: warning: command-line option ‘-std=gnu99’ is valid for C/ObjC but not for C++`
> Not a big deal, just wanted to let you know :)

Thanks, I will take a look at that.

brendon, 2021-12-29:
Thanks everyone :) Looks like this spurred a bunch of useful changes and cleanup :) I realised I was confused with `nvme-cli` and the actual `nvme` driver. I managed to find a compatible driver bundled with a `4.4` kernel (http://mirrors.coreix.net/elrepo-archive-archive/kernel/el6/x86_64/RPMS/) (the driver with the 2.x kernel booted but hung while trying to load `apache`). I'll probably stick with `1.3` of `nvme-cli` for now since things seem to be working well.

If you'd like me to pursue building with `meson` please let me know and I'll have a go :)

igaw, 2021-12-29:
Okay, in this case let's close this issue. Thanks for everyone involved!

closed 2021-12-29 by igaw