linux-nvme/nvme-cli #1295: error: ‘HMAC_CTX_free’ is deprecated: Since OpenSSL 3.0

johnmeneghini, 2021-12-21:
Looks like the recent OPENSSL changes were developed with a deprecated Open SSL library.

```
rhel-storage-08:nvme-cli(master) > make nvme
NVME_VERSION = 1.14.331.gf550
    CC ccan/tools/configurator/configurator
ccan/tools/configurator/configurator > ccan/config.h
    CC nvme.o
nvme.c: In function ‘gen_dhchap_key’:
nvme.c:6814:17: error: ‘HMAC_CTX_new’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
 6814 |                 HMAC_CTX *hmac_ctx = HMAC_CTX_new();
      |                 ^~~~~~~~
In file included from nvme.c:50:
/usr/include/openssl/hmac.h:33:33: note: declared here
   33 | OSSL_DEPRECATEDIN_3_0 HMAC_CTX *HMAC_CTX_new(void);
      |                                 ^~~~~~~~~~~~
nvme.c:6818:17: error: ‘ENGINE_load_builtin_engines’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
 6818 |                 ENGINE_load_builtin_engines();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from nvme.c:48:
/usr/include/openssl/engine.h:358:28: note: declared here
  358 | OSSL_DEPRECATEDIN_3_0 void ENGINE_load_builtin_engines(void);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
nvme.c:6819:17: error: ‘ENGINE_register_all_complete’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
 6819 |                 ENGINE_register_all_complete();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from nvme.c:48:
/usr/include/openssl/engine.h:415:27: note: declared here
  415 | OSSL_DEPRECATEDIN_3_0 int ENGINE_register_all_complete(void);
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvme.c:6821:17: error: ‘HMAC_Init_ex’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
 6821 |                 HMAC_Init_ex(hmac_ctx, raw_secret, cfg.key_len,md, NULL);
      |                 ^~~~~~~~~~~~
In file included from nvme.c:50:
/usr/include/openssl/hmac.h:43:27: note: declared here
   43 | OSSL_DEPRECATEDIN_3_0 int HMAC_Init_ex(HMAC_CTX *ctx, const void *key, int len,
      |                           ^~~~~~~~~~~~
nvme.c:6822:17: error: ‘HMAC_Update’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
 6822 |                 HMAC_Update(hmac_ctx, (unsigned char *)hostnqn,
      |                 ^~~~~~~~~~~
In file included from nvme.c:50:
/usr/include/openssl/hmac.h:45:27: note: declared here
   45 | OSSL_DEPRECATEDIN_3_0 int HMAC_Update(HMAC_CTX *ctx, const unsigned char *data,
      |                           ^~~~~~~~~~~
nvme.c:6824:17: error: ‘HMAC_Update’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
 6824 |                 HMAC_Update(hmac_ctx, (unsigned char *)hmac_seed,
      |                 ^~~~~~~~~~~
In file included from nvme.c:50:
/usr/include/openssl/hmac.h:45:27: note: declared here
   45 | OSSL_DEPRECATEDIN_3_0 int HMAC_Update(HMAC_CTX *ctx, const unsigned char *data,
      |                           ^~~~~~~~~~~
nvme.c:6826:17: error: ‘HMAC_Final’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
 6826 |                 HMAC_Final(hmac_ctx, key, &key_len);
      |                 ^~~~~~~~~~
In file included from nvme.c:50:
/usr/include/openssl/hmac.h:47:27: note: declared here
   47 | OSSL_DEPRECATEDIN_3_0 int HMAC_Final(HMAC_CTX *ctx, unsigned char *md,
      |                           ^~~~~~~~~~
nvme.c:6827:17: error: ‘HMAC_CTX_free’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
 6827 |                 HMAC_CTX_free(hmac_ctx);
      |                 ^~~~~~~~~~~~~
In file included from nvme.c:50:
/usr/include/openssl/hmac.h:35:28: note: declared here
   35 | OSSL_DEPRECATEDIN_3_0 void HMAC_CTX_free(HMAC_CTX *ctx);
      |                            ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:152: nvme.o] Error 1

rhel-storage-08:nvme-cli(master) > rpm -q --whatprovides /usr/include/openssl/hmac.h
openssl-devel-3.0.0-4.el9.x86_64

rhel-storage-08:nvme-cli(master) > git blame nvme.c | grep -C 2 HMAC_CTX_new
a05d4213 (Hannes Reinecke                  2020-12-17 09:43:50 +0100 6812)      if (md) {
a05d4213 (Hannes Reinecke                  2020-12-17 09:43:50 +0100 6813) #ifdef OPENSSL
a05d4213 (Hannes Reinecke                  2020-12-17 09:43:50 +0100 6814)              HMAC_CTX *hmac_ctx = HMAC_CTX_new();
a05d4213 (Hannes Reinecke                  2020-12-17 09:43:50 +0100 6815)              const char hmac_seed[] = "NVMe-over-Fabrics";
a05d4213 (Hannes Reinecke                  2020-12-17 09:43:50 +0100 6816)              unsigned int key_len;
```

igaw, 2021-12-22:
> Looks like the recent OPENSSL changes were developed with a deprecated Open SSL library.

Which distro is this, Fedora? Tumbleweed seems to be still shipping the 1.x version here. Ideally we find a way to use
either version.

johnmeneghini, 2021-12-22:
@igaw I've compiled your branch with both RHEL 8.5 and RHEL 9.0.

```
jmeneghi:> lsb_release -i -d -c -r
Distributor ID: RedHatEnterprise
Description:    Red Hat Enterprise Linux release 8.5 (Ootpa)
Release:        8.5
Codename:       Ootpa
jmeneghi: > pkg-config --modversion openssl
1.1.1k
jmeneghi:> rpm -q --whatprovides /usr/include/openssl/hmac.h
openssl-devel-1.1.1k-4.el8.x86_64
```

And on RHEL 9

```
 > cat /etc/os-release
NAME="CentOS Stream"
VERSION="9"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="9"
PLATFORM_ID="platform:el9"
PRETTY_NAME="CentOS Stream 9"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:9"
HOME_URL="https://centos.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux 9"
REDHAT_SUPPORT_PRODUCT_VERSION="CentOS Stream"

> pkg-config --modversion openssl
3.0.0
> rpm -q --whatprovides /usr/include/openssl/hmac.h
openssl-devel-3.0.0-5.el9.x86_64

```
It all looks good to me.

PR #1297 (2022-01-07): Add OpenSSL 3 support for dhchap-key generation
nvme: Add OpenSSL 3 support for DH-HMAC-CHAP key generation

OpenSSL 3 deprecated the HMAC_* API and introduced a new high level
API (EVP_*). Unfortunately, the EVP_* API is not available in OpenSSL
1.x. Add support for OpenSSL 3 as alternative implementation to
generate the DH-HMAC-CHAP key.

While at it also add error handling to the OpenSSL 1 version.

Fixes #1295
closed 2022-01-07 by igaw