linux-nvme/nvme-cli #1298: Question: IO Passthru metadata support

ahribeng, 2021-12-22:
Seems like write and read commands have the ability to read/write to a metadata file. The io passthru command does not have the same functionality. Is this purposeful, could it be updated to have the same functionality. Open to a pull request for it?

keithbusch, 2021-12-22:
I'm not sure I understand what you mean. Passthrough should be able to reach metadata regions for applicable commands. Could you provide an example command that is failing?

ahribeng, 2021-12-22:
Yea apologies for insufficient context.

**To be clear no command is failing**

When issuing a write via:

> nvme write ...

the user has the ability to fill the metadata buffer via a file using the -M switch

When issuing a read via:

> nvme read ...

the user has the ability to dump the metadata buffer to a file using the -M switch

Both write and read commands use the submit_io wrapper

When issuing the same write/read or other io command via:

> nvme io-passthru ...

the user does not have the ability to fill the metadata buffer using a file. The only option is to prefill using the -p switch
Additionally the metadata buffer cannot be dumped for commands that would issue data transfer from controller to host

Then io-passthru command uses the passthru wrapper

Was wondering if io-passthru could be changed to be have the same functionality?

keithbusch, 2021-12-22:
Gotcha, I understand now. The ability to direct the input/output for the metadata payload would indeed be appropriate to have in the generic io-passthru. Please send a pull request if you have a solution ready for consideration, otherwise I can add this item on my queue.

ahribeng, 2021-12-22:
Yea I don't have anything right now. Could you add to your queue? Ultimately I think this is a nice to have.

closed 2022-01-10 by igaw