linux-nvme/nvme-cli #1303: Ubuntu 20.04 http://ppa.launchpad.net/sbates/ppa/ubuntu focal Release Not Found

inside83, 2022-01-03:
Hi,
I just noticed that running apt-get update informs me that `http://ppa.launchpad.net/sbates/ppa/ubuntu focal Release` is not found

Any idea how to fix it?

keithbusch, 2022-01-03:
@sbates130272 , does this make any sense to you?

igaw, 2022-01-03:
As far I can tell, the repo doesn't package nvme-cli for the 'focal' release. The packaged version there is pretty old too. Ubuntu ships nvme-cli 1.9 https://packages.ubuntu.com/focal/nvme-cli

Why do you need the old version?  If you really need the old version from the ppa you have to build it against the 'focal' release.

inside83, 2022-01-03:
hey
thanks for a quick reply
and a happy new year to all :-)

I don't actually need older version.
hetrixtools.com needs nvme-cli to monitor nvme disks so I followed instructions here to install it.

igaw, 2022-01-03:
Thanks, also a happy new year.

Alright, in this case all is good. Let's close this issue in this case.

closed 2022-01-03 by igaw

inside83, 2022-01-03:
Wait, how is it all good?
I should remove the PPA, what?

igaw, 2022-01-03:
Yes, you don't need the PPA. Just use the nvme-cli package from the main distro.

inside83, 2022-01-05:
Ok, thank you
This may seam like a silly question but how to remove it?

igaw, 2022-01-05:
I am not an Ubuntu user, but found this here https://askubuntu.com/questions/307/how-can-ppas-be-removed