linux-nvme/nvme-cli #1309: build warning in wdc-nvme.c

igaw, 2022-01-07:
I get this build warning locally. This doesn't appear in CI build though. @jeffreyalien  could have a look?

```
In file included from ../plugins/wdc/wdc-nvme.c:34:
../plugins/wdc/wdc-nvme.c: In function ‘wdc_vs_smart_add_log’:
../plugins/wdc/wdc-nvme.c:4209:46: warning: array subscript ‘__u64 {aka long long unsigned int}[18]’ is partly outside array bounds of ‘unsigned char[156]’ [-Warray-bounds]
 4209 |                                 safe_div_fp((*raw & 0x00FFFFFFFFFFFFFF), 0xFFFF));
      |                                              ^~~~
../nvme.h:48:61: note: in definition of macro ‘json_object_add_value_float’
   48 |         json_object_object_add(o, k, json_object_new_double(v))
      |                                                             ^
../plugins/wdc/wdc-nvme.c:5300:45: note: referencing an object of size 156 allocated by ‘malloc’
 5300 |                         if ((data = (__u8*) malloc(sizeof (__u8) * WDC_BD_CA_LOG_BUF_LEN)) == NULL) {
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../plugins/wdc/wdc-nvme.c:4075:68: warning: array subscript ‘__u64 {aka long long unsigned int}[18]’ is partly outside array bounds of ‘unsigned char[156]’ [-Warray-bounds]
 4075 |                            bd_data->normalized_value, safe_div_fp((*raw & 0x00FFFFFFFFFFFFFF), 0xFFFF));
      |                                                                    ^~~~
../plugins/wdc/wdc-nvme.c:5300:45: note: referencing an object of size 156 allocated by ‘malloc’
 5300 |                         if ((data = (__u8*) 
```malloc(sizeof (__u8) * WDC_BD_CA_LOG_BUF_LEN)) == NULL) {
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

jeffreyalien, 2022-01-07:
@igaw  Yes, I'll take a look at this.

jeffreyalien, 2022-01-10:
@igaw What linux distro and compiler/tool chain version is this happening on?   It looks like a padding issue that could be easily fixed but it's I'm not seeing the warnings on my Redhat system.  I'd like to be able to recreate the warnings so I can verify it's been fixed.

igaw, 2022-01-11:
I am getting this on Tumbleweed. I am glad to verify your patch if you have something. You don't have to install a distro.

igaw, 2022-01-12:
Still the same warnings.

```
#define WDC_BD_CA_LOG_BUF_LEN                                   0x9C  

data = (__u8*) malloc(sizeof (__u8) * WDC_BD_CA_LOG_BUF_LEN))

struct  wdc_bd_ca_log_format {   
...
         __u8    raw_value[7]; 
...
};

raw = (__u64*)bd_data->raw_value;  
```

All these numbers do not match. The raw_value array is only 7 bytes but when accessing raw the compiler wants to access 8. I tried to change raw_value to 8 but then the buffer is too small (WDC_BD_CA_LOG_BUF_LEN).

jeffreyalien, 2022-01-12:
[compiler-warnings-2.txt](https://github.com/linux-nvme/nvme-cli/files/7855634/compiler-warnings-2.txt)
@igaw  Please try this version when you have a chance.  I remove the reserved byte and changed raw_value to an 8 byte array.  Hope that resolves the warnings.   Thanks.

igaw, 2022-01-12:
Yes, this patch works. No compiler warnings anymore.

jeffreyalien, 2022-01-12:
Ok, great.  I need to do some more verification of the changes.  I'll create a PR for this patch once done.  Thanks for the help.

igaw, 2022-01-12:
Glad I could help :)

jeffreyalien, 2022-01-14:
@igaw Could you try this version of wdc-nvme.txt to verify there are no build warnings?  Just rename it to wdc-nvme.c and replace the existing version.    I found a problem with the way I initially fixed.  Thanks. 
[wdc-nvme.txt](https://github.com/linux-nvme/nvme-cli/files/7872918/wdc-nvme.txt)

igaw, 2022-01-17:
Still the same warning with [from](https://github.com/linux-nvme/nvme-cli/files/7872918/wdc-nvme.txt) 
```
../plugins/wdc/wdc-nvme.c: In function ‘wdc_vs_smart_add_log’:
../plugins/wdc/wdc-nvme.c:4281:46: warning: array subscript ‘__u64 {aka long long unsigned int}[18]’ is partly outside array bounds of ‘unsigned char[156]’ [-Warray-bounds]
 4281 |                                 safe_div_fp((*raw & 0x00FFFFFFFFFFFFFF), 0xFFFF));
      |                                              ^~~~
../nvme.h:48:61: note: in definition of macro ‘json_object_add_value_float’
   48 |         json_object_object_add(o, k, json_object_new_double(v))
      |                                                             ^
../plugins/wdc/wdc-nvme.c:5405:45: note: referencing an object of size 156 allocated by ‘malloc’
 5405 |                         if ((data = (__u8*) malloc(sizeof (__u8) * WDC_BD_CA_LOG_BUF_LEN)) == NULL) {
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../plugins/wdc/wdc-nvme.c:4128:68: warning: array subscript ‘__u64 {aka long long unsigned int}[18]’ is partly outside array bounds of ‘unsigned char[156]’ [-Warray-bounds]
 4128 |                            bd_data->normalized_value, safe_div_fp((*raw & 0x00FFFFFFFFFFFFFF), 0xFFFF));
      |                                                                    ^~~~
../plugins/wdc/wdc-nvme.c:5405:45: note: referencing an object of size 156 allocated by ‘malloc’
 5405 |                         if ((data = (__u8*) malloc(sizeof (__u8) * WDC_BD_CA_LOG_BUF_LEN)) == NULL) {
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[65/65] Linking target nvme
```

jeffreyalien, 2022-01-18:
@igaw Could you try this version?  I think it should have all the warnings fixed.  Thanks
[wdc-nvme.txt](https://github.com/linux-nvme/nvme-cli/files/7890918/wdc-nvme.txt)

igaw, 2022-01-19:
The last version is fine, no gcc warnings. Nice!

jeffreyalien, 2022-01-19:
Created https://github.com/linux-nvme/nvme-cli/pull/1325 to resolve the build warnings in the wdc plugin parts.

igaw, 2022-01-19:
Fix applied. Closing this bug report.

closed 2022-01-19 by igaw