linux-nvme/nvme-cli #1317: Build warning in micron-nvme.c plugin

jeffreyalien, 2022-01-13:
I'm hitting these warning messages from the latest level of nvme cli code on my redhat 8.4 system:

../plugins/micron/micron-nvme.c: In function \u2018micron_internal_logs\u2019:
../plugins/micron/micron-nvme.c:73:29: warning: \u2018%s\u2019 directive writing up to 4095 bytes into a region of size between 3072 and 4095 [-Wformat-overflow=]
     sprintf(tempFolder, "%s/%s", dir, file);
                             ^~
../plugins/micron/micron-nvme.c:1699:48:
         WriteData((__u8*)&ns, sizeof(ns), dir, file, "id-ns");
                                                ~~~~
../plugins/micron/micron-nvme.c:73:5: note: \u2018sprintf\u2019 output between 2 and 5120 bytes into a destination of size 4096
     sprintf(tempFolder, "%s/%s", dir, file);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[13/13] Linking target nvme

It appears the tempFolder array in the WriteData function is too small.  I increased it to 8192 and that resolved the warning messages but not sure that's the proper/desired fix.

jeffreyalien, 2022-01-14:
@igaw Could you forward this onto the micron plugin owner?   Thanks

igaw, 2022-01-17:
I think this might be something for @hanumanthuh

hanumanthuh, 2022-01-17:
Jeffrey,  the change should be fine.

igaw, 2022-01-19:
@jeffreyalien are you going to send a PR?

jeffreyalien, 2022-01-19:
@igaw @hanumanthuh  Just submitted https://github.com/linux-nvme/nvme-cli/pull/1328 for this issue.  Sorry for the delay.

igaw, 2022-01-20:
PR merged. Thanks!

closed 2022-01-20 by igaw