linux-nvme/nvme-cli #1320: Secure erase returns success when run with partition mounted, but can still access files

springl0aded, 2022-01-17:
I have a Kingston A2000 SSD. And what I'm seeing suggests to me that secure erase isn't working as I would expect.
I'm testing this on an Unraid server, which is a slackware base. The package appears to be nvme-cli 1.11

I use fdisk to create a 60GB partition
Then I forrmat it with xfs
Then I mount it to a mountpoint, copy some files to it.
Then (neglecting to umount) I run: 
nvme format /dev/nvme0n1 -s 2
After the bail out counter, this completes basically instantly:
> Sending format operation ... 
> Success formatting namespace:1

But I can still see the files in the mount point. I can copy and write files to the mounted partition. 
> "echo 1 > /proc/sys/vm/drop_caches"
> "sync"

...do not seem to change this.

Once I umount, I can no longer mount the partition again. But when I run fdisk to recreate the partition again this warning question appears:

> Partition #1 contains a xfs signature.
> Do you want to remove the signature? [Y]es/[N]o:

I don't see why fdisk can tell anything about previous partitions if the namespace was secure erased properly.

If I run the exact same operations _except_ that I DO umount prior to nvme format, when I run fdisk I do NOT get the info about xfs signature.

This makes me think the secure erase did not happen property but still reported success. Is there a way to confirm a secure erase was successful?

igaw, 2022-01-17:
Does this also happen with latest the version of nvme-cli (v1.16)?

springl0aded, 2022-01-17:
I just realized the package included with linux mint is actually even older. It appears to be 1.09-1ubuntu0.1.

I repeated the initial process with a mint 20.2 live CD. The only differences were the lack of a bail out timer and when attempting to secure erase with ses=2 the following text appears after success:
> failed to re-read partition table

This did not appear if the partition was unmounted.

I'll see if I can install 1.16 somehow.

igaw, 2022-01-18:
I've missed this yesterday "neglecting to umount".

nvme-cli does not care if the disk is mounted or not. The operation will be blindly executed. That means `nvme format` will reformat the disk but the block layer and file system will not be told. Eventually the file system will write data and thus write on the freshly formatted disk.

This boils down to: this order of operation is not supported. You need to unmount before issuing a `nvme format`.

closed 2022-01-18 by igaw