linux-nvme/nvme-cli #1321: intel plugin does not work with Intel SSDPEKKF512G8

paulmenzel, 2022-01-18:
With the Intel SSDPEKKF512G8, the intel plugin commands like `internal-log` and `market-name` do not work.

    $ ./nvme version
    nvme version 1.14.68.g24dd
    $ sudo ./nvme list
    Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev  
    --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
    nvme0n1               BTHH82250YQK512D     SSDPEKKF512G8 NVMe INTEL 512GB           1         512.11  GB / 512.11  GB    512   B +  0 B   D03N
    $ sudo ./nvme fw-log /dev/nvme0n1 # works
    Firmware Log for device:nvme0n1
    afi  : 0x11
    frs1 : 0x007ffff10a57a8 (D03N....)
    frs2 : 0x007ffff10a57b0 (D03N....)
    frs3 : 0x007ffff10a57b8 (........)
    frs4 : 0x007ffff10a57c0 (........)
    frs5 : 0x007ffff10a57c8 (........)
    frs6 : 0x007ffff10a57d0 (........)
    frs7 : 0x007ffff10a57d8 (........)

    $ sudo ./nvme intel internal-log /dev/nvme0n1
    failed on cmd.data_len 4096 cmd.cdw13 0 cmd.cdw12 0 cmd.cdw10 1024 err 286 remaining size 1024
    NVMe status: Access Denied: Access to the namespace and/or LBA range is denied due to lack of access rights(0x286)
    $ sudo ./nvme intel market-name /dev/nvme0n1
    NVMe status: Invalid Log Page: The log page indicated is invalid(0x109)

keithbusch, 2022-01-18:
It doesn't appear that Intel supports these vendor specific log pages for this model. I believe this one uses 3rd party controller and firmware from SMI.

paulmenzel, 2022-01-18:
Thank you for the prompt response. Can this be somehow checked, and a user friendlier error be shown?

igaw, 2022-01-19:
Some vendor plugins do have checks if the plugin is supported by the device (see wdc for example). It seems the intel plugin doesn't have such a allow/deny list.

What 'userfriendly' means depends on the context. As this is a low level tool I do prefer a very technical error message.

igaw, 2022-02-15:
Closing due to inactivity. If it's still relevant please reopen.

closed 2022-02-15 by igaw