linux-nvme/nvme-cli #1322: How to update firmware of Intel SSDPEKKF512G8?

paulmenzel, 2022-01-18:
The Dell OptiPlex 5055 shipped with the Intel SSDPEKKF512G8 with firmware D03N.

    $ ./nvme version
    nvme version 1.14.68.g24dd
    $ sudo ./nvme list
    Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev  
    --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
    nvme0n1               BTHH82250YQK512D     SSDPEKKF512G8 NVMe INTEL 512GB           1         512.11  GB / 512.11  GB    512   B +  0 B   D03N

I would like to update it to version D06N [released by Dell](https://www.dell.com/support/home/de-de/drivers/driversdetails?driverid=5929g&oscode=wt64a&productcode=optiplex-5055-ryzen-cpu). Unfortunately, the provided archive seems only be suitable for Microsoft Windows, and contains a file `FirmwareModules/HH_D06N.dll`.

Is it possible to update the device from within GNU/Linux?

paulmenzel, 2022-01-18:
$ lspci -nn -s 01:00.0
    01:00.0 Non-Volatile memory controller [0108]: Intel Corporation SSD Pro 7600p/760p/E 6100p Series [8086:f1a6] (rev 03)

igaw, 2022-01-18:
Yes, it's possible to update firmware. See the [fwup](https://fwupd.org/) project. Though the vendor has to provide the firmware to the upstream project. It doesn't look Dell has provided anything for your device.

paulmenzel, 2022-01-18:
Yes, I checked that beforehand. ;-) Unfortunately, it’s not there despite Dell selling the device with Ubuntu.

igaw, 2022-01-18:
As far I can tell, I was able to update a nvme device (Toshiba) via fwup. I suggest you try to bother Dell. Even though I wouldn't expect anything out of it but if no one is asking for such a service they will not provide it.

igaw, 2022-01-19:
nvme-cli supports firmware updates via 'fw-download' and 'fw-commit'.

For this the correct input file needs to be provided which is beyond this project. 

As mentioned before, the canonical way is to get the vendor convinced to provide firmware updates via fwup. Dell is contributing to fwup and there is even a big statement on the website of fwup how strategic important fwup is for them. So you really should contact them about this issue.

closed 2022-01-19 by igaw