linux-nvme/nvme-cli #1324: build warnings on 32bit architectures

igaw, 2022-01-19:
```
[44/65] Compiling C object nvme.p/nvme.c.o
../nvme.c: In function ‘get_effects_log’:
../nvme.c:693:64: warning: left shift count >= width of type [-Wshift-count-overflow]
  693 |                 nvme_command_set_supported = (cap_value & (1UL << 37)) != 0;
      |                                                                ^~
../nvme.c:694:66: warning: left shift count >= width of type [-Wshift-count-overflow]
  694 |                 other_command_sets_supported = (cap_value & (1UL << (37+6))) != 0;
      |                                                                  ^~
[45/65] Compiling C object nvme.p/nvme-print.c.o
../nvme-print.c: In function ‘json_fw_log’:
../nvme-print.c:521:33: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  521 |                                 (uint64_t)fw_log->frs[i],
      |                                 ^
../nvme-print.c: In function ‘nvme_show_fw_log’:
../nvme-print.c:5374:33: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
 5374 |                                 (uint64_t)fw_log->frs[i],
      |                                 ^
ninja: build stopped: subcommand failed.
```

sc108-lee, 2022-01-21:
My system currently unavailable for building with 32bit arch
Anyway I modified some, plz check it out how it works

1. UL (unsigned long) is 32bit on 32bit, 64bit on 64bit.
    That's why shifting that occur overflow.
    -> using libnvme macro which that shift down first of cap value (not occur overflow)

2. fixed via type cast of char[] addr, and using le64_to_cpu to get right values

hreinecke, 2022-01-21:
Ouch. The cast (uint64_t)fw_log->frs[i] is definitely wrong.
The definition is:

struct nvme_firmware_slot {
        __u8    afi;
        __u8    resv[7];
        char    frs[7][8];
        __u8    resv2[448];
};

So you would need to convert the 'char[8]' thingie into a proper 64-bit value first (keeping in mind endianness) and then print it out using the standard PRI macros.

igaw, 2022-01-21:
FWIW, you can tell meson to cross compile using the '--cross-file' argument. I did the above with the following settings

```
$ meson .build --cross-file x86-linux 
$ ninja -C .build
```

~/.local/share/meson/cross/x86-linux 
```ini
[binaries]
c               = 'gcc'
cpp             = 'g++'
ld              = 'ld'
ar              = 'ar'
as              = 'as'
size            = 'size'
objdump         = 'objdump'
objcopy         = 'objcopy'
strip           = 'strip'
gdb             = 'gdb'
pkgconfig       = 'pkg-config'

[properties]

[built-in options]
c_args          = ['-m32']
c_link_args     = ['-m32']
cpp_args        = ['-m32']
cpp_link_args   = ['-m32']

[target_machine]
system          = 'linux'
cpu_family      = 'x86'
cpu             = 'i686'
endian          = 'little'

```

sc108-lee, 2022-01-21:
@hreinecke 
I'm seeing that on spec as a just 8bytes.
And original writer might consider the revision exist as a number.
That's why print as a number I guess.
And also if the value is number, then it should comes as little endian
That's why I bring that as __le64

@igaw Thanks for sharing the config :)

commit 6c55edf9 (2022-01-21): fix #1324 using explicit casting, using libnvme macro for cap
commit e8b8423c (2022-01-21): Merge pull request #1333 from sc108-lee/fix/#1324

fix #1324 using explicit casting, using libnvme macro for cap
closed 2022-01-21 by igaw