linux-nvme/nvme-cli #1331: crash in nvme_ns_get_name()

igaw, 2022-01-20:
nvme-cli: c7eb65086896 ("Merge pull request #1328 from jeffreyalien/compiler-warnings2")
libnvme: 609e6faef0ca ("Merge pull request #185 from martin-belanger/ccan-static-lib")

with `s/NVME_NS_FLBAS_LBA_MASK/NVME_NS_FLBAS_LOWER_MASK/` local fixup.

Not sure if we just need to check if the pointer is `NULL` here:

```c
6706                                    nvme_ctrl_for_each_path(c, p) {
6707                                            printf("%s%s", first ? "": ", ",
6708                                                   nvme_ns_get_name(nvme_path_get_ns(p)));
6709                                            first = false;
6710                                    }
```

```
0  0x000000000048ff2a in nvme_ns_get_name (n=0x0) at ../subprojects/libnvme/src/nvme/tree.c:1391
No locals.
#1  0x0000000000438ba5 in nvme_show_detailed_list (r=0x70e730) at ../nvme-print.c:6707
        first = true
        h = 0x70e880
        s = 0x711b70
        c = 0x7120d0
        p = 0x7122d0
        n = 0x0
#2  0x0000000000439c58 in nvme_show_list_items (r=0x70e730, flags=VERBOSE) at ../nvme-print.c:6908
No locals.
#3  0x000000000040ef54 in list (argc=2, argv=0x7fffffffe360, cmd=0x6eb580 <list_cmd>, plugin=0x6ec360 <builtin>) at ../nvme.c:2349
        desc = 0x4a1fa0 "Retrieve basic information for all NVMe namespaces"
        verbose = 0x4a0525 "Increase output verbosity"
        flags = VERBOSE
        r = 0x70e730
        err = 0
        cfg = {output_format = 0x49ff40 "normal", verbose = 1}
        opts = {{option = 0x4a002d "output-format", short_option = 111 'o', meta = 0x4a003b "FMT", config_type = CFG_STRING, 
            default_value = 0x7fffffffe0b0, argument_type = 1, help = 0x49fe92 "Output format: normal|json"}, {
            option = 0x4a053f "verbose", short_option = 118 'v', meta = 0x0, config_type = CFG_NONE, default_value = 0x7fffffffe0b8, 
            argument_type = 0, help = 0x4a0525 "Increase output verbosity"}, {option = 0x0, short_option = 0 '\000', meta = 0x0, 
            config_type = CFG_NONE, default_value = 0x0, argument_type = 0, help = 0x0}}
#4  0x000000000043cd51 in handle_plugin (argc=2, argv=0x7fffffffe360, plugin=0x6ec360 <builtin>) at ../plugin.c:155
        str = 0x7fffffffe6e9 "list"
        use = "nvme list <device> [OPTIONS]\000\000\000\000\320\346p\000\000\000\000\000\220\337p\000\000\000\000\000\320\332p\000\000\000\000\000\340\322p\000\000\000\000\000\300\315p\000\000\000\000\000\300\307p", '\000' <repeats 13 times>, "\020\303p\000\000\000\000\000\200\276p\000\000\000\000\000йp\000\000\000\000\000p\246p\000\000\000\000\000\260\241p\000\000\000\000\000\060\235p", '\000' <repeats 13 times>, "\300\343p\000\000\000\000\000\020\337p\000\000\000\000\000\020\327p\000\000\000\000\000\000\322p\000\000\000\000\000\000\314p\000\000\000\000\000P\307p", '\000' <repeats 13 times>, "\260"...
        extension = 0x7fffffffe350
        prog = 0x6ec3a0 <nvme>
        cmd = 0x6ec080 <commands>
        cr = 0x0
        cr_valid = false
#5  0x000000000041e88a in main (argc=3, argv=0x7fffffffe358) at ../nvme.c:7294
        err = 0
```

igaw, 2022-01-20:
A NULL check doesn't seem to be enough:

```
#0  0x000000000048ff35 in nvme_ns_get_name (n=0x0) at ../subprojects/libnvme/src/nvme/tree.c:1391
#1  0x0000000000438bad in nvme_show_detailed_list (r=0x70e730) at ../nvme-print.c:6709
#2  0x0000000000439c63 in nvme_show_list_items (r=0x70e730, flags=VERBOSE) at ../nvme-print.c:6910
#3  0x000000000040ef54 in list (argc=2, argv=0x7fffffffe360, cmd=0x6eb580 <list_cmd>, plugin=0x6ec360 <builtin>) at ../nvme.c:2349
#4  0x000000000043cd5c in handle_plugin (argc=2, argv=0x7fffffffe360, plugin=0x6ec360 <builtin>) at ../plugin.c:155
#5  0x000000000041e88a in main (argc=3, argv=0x7fffffffe358) at ../nvme.c:7294

(gdb) p p
$1 = (nvme_path_t) 0x7122d0

(gdb) p *p
$2 = {entry = {next = 0x7120e0, prev = 0x7120e0}, nentry = {next = 0x7122e0, prev = 0x7122e0}, c = 0x7120d0, n = 0x0, 
  name = 0x712330 "nvme17c18n1", sysfs_dir = 0x70e850 "/sys/class/nvme/nvme18/nvme17c18n1", ana_state = 0x712350 "optimized", 
  grpid = 1}
```

hreinecke, 2022-01-21:
Should be fixed by PR #1332 . Can you check?

igaw, 2022-01-21:
Yes it works. PR merged.

closed 2022-01-21 by igaw