linux-nvme/nvme-cli #1334: meson: fallback failure for subproject of subproject

sc108-lee, 2022-01-21:
openssl(subproject of libnvme) wraping is failed.
  ( it's success with libnvme meson build )

Determining dependency 'openssl' with CMake executable '/usr/local/bin/cmake'
Try CMake generator: auto
None of 'FC' are defined in the environment, not changing global flags.
None of 'FFLAGS' are defined in the environment, not changing global flags.
None of 'LDFLAGS' are defined in the environment, not changing global flags.
None of 'FC_LD' are defined in the environment, not changing global flags.
None of 'F_LD' are defined in the environment, not changing global flags.
Called `/usr/local/bin/cmake -DNAME=openssl -DARCHS= -DCOMPS= --trace --trace-expand --no-warn-unused-cli .` in /root/Desktop/repo/nvme-cli/.build/meson-private/cmake_openssl -> 0
Run-time dependency openssl found: NO (tried pkgconfig and cmake)
Looking for a fallback subproject for the dependency openssl
subprojects/libnvme/meson.build:50: WARNING: Dependency openssl not found but it is available in a sub-subproject.
To use it in the current project, promote it by going in the project source
root and issuing the following command: 
meson wrap promote subprojects/libnvme/subprojects/openssl.wrap
Subproject directory not found and openssl.wrap file not found
Subproject  subprojects/openssl is buildable: NO (disabling)
Dependency openssl from subproject subprojects/openssl found: NO

...
nvme-cli meson build result
nvme-cli 2.0

  Subprojects
     json-c: YES
    libnvme: YES 1 warnings
    openssl: NO Subproject directory not found and openssl.wrap file not found

libnvme meson build result below
libnvme 1.0

  Subprojects
     json-c: YES
    openssl: YES

sc108-lee, 2022-01-21:
I've checked that is problem of old version of meson

meson >= 0.56.0 support the feature.
https://github.com/mesonbuild/meson/blob/master/docs/markdown/Release-notes-for-0.56.0.md#wraps-from-subprojects-are-automatically-promoted

closed 2022-01-21 by sc108-lee

igaw, 2022-01-21:
That's interesting. I thought the version check in project() should handle this. I remember meson complained if I did used a feature which was not part of =0.47.0.

igaw, 2022-01-21:
Ah got it, the nested subproject is not working.

igaw, 2022-01-21:
But it even tells you what to do in such a case: "meson wrap promote subprojects/libnvme/subprojects/openssl.wrap"

Nice!

sc108-lee, 2022-01-21:
@igaw yes! that's it. 
I was worried about normal user can't see that promote.
But it's support from 0.56, so no problem :)