linux-nvme/nvme-cli #1347: Inreasing log level doesn't work

igaw, 2022-02-01:
Increasing the log level is supported only a few commands but this is not working. The code handling OPT_INCR is not working as expected.

`argconfig_parse` is considering OPT_INCR as flag and never goes into the CFG_INCREMENT code snipped. But even with this simple patch here it doesn't work as expected. 

```patch
@@ -219,7 +219,8 @@ int argconfig_parse(int argc, char *argv[], const char *program_desc,
                        }
                        if (option_index == options_count)
                                continue;
-                       if (long_opts[option_index].flag) {
+                       if (long_opts[option_index].flag &&
+                           options[option_index].config_type != CFG_INCREMENT) {
                                *(uint8_t *)(long_opts[option_index].flag) = 1;
                                continue;
                        }

```

More debugging needed.

ikegami-t, 2022-02-03:
Hi,

It seems that the current implementation and behavior work as expected as below.
1. The defult log level is set to `notice` if not defined but possible to be not set to `notice` if defined already.
```
#ifndef DEFAULT_LOGLEVEL
#  define DEFAULT_LOGLEVEL LOG_NOTICE
#endif
```
2. The verbose argument is set to `1` at first.
```
			if (s->argument_type == no_argument
			    && s->default_value != NULL) {
				value_addr = (void *)(char *)s->default_value;

				long_opts[option_index].flag = value_addr;
				long_opts[option_index].val = 1;
			} else {
				long_opts[option_index].flag = NULL;
				long_opts[option_index].val = 0;
			}
```
3. Then incremented to `2` but the comment description seems not correct.
```
		} else if (s->config_type == CFG_INCREMENT) {
			/*
			 * Extreme getopt_long fiddling.
			 *
			 * As per man page:
			 * If flag is non-NULL, getopt_long() returns 0,
			 * and flag  points to a variable which is set to
			 * val if the option is found.
			 *
			 * So we need to increase 'val', not 'value_addr'.
			 */
			long_opts[option_index].val++;
```
4. At last the log level chnaged to `info` from `notice` but seems the log level changed to `warning` if the option not set and also seems not changed the log level to `debug` actually.
```
	switch (verbose) {
	case 0:
		nvme_log_level = LOG_WARNING;
		break;
	case 1:
		nvme_log_level = LOG_NOTICE;
		break;
	case 2:
		nvme_log_level = LOG_INFO;
		break;
	default:
		nvme_log_level = LOG_DEBUG;
		break;
	}
	if (quiet)
		nvme_log_level = LOG_ERR;
```

By the way now I am thinking to change the option as below so do you have any comment or advice about this?
1. Change OPT_INCR argument_type to optional_argument from no_argument.
```diff
 #define OPT_INCR(l, s, v, d) \
-	{l, s, "NUM", CFG_INCREMENT, v, no_argument, d}
+	{l, s, "NUM", CFG_INCREMENT, v, optional_argument, d}
```
2. Change the log level only if the option set without the argument and increment the log level as `1` level from the default log level `notice` etc.
```
nvme *** ... --verbose (Only the option set with the argument then the log level incremented to `info` etc.)
``` 
3. If the option argument is set just chnagte the log level to the argument value without the increment.
```
nvme *** ... --verbose 7 (Change the log level to 7: debug)
``` 

Regards,
Ikegami

igaw, 2022-02-04:
Let me look at it again. I played with 'nvme-cli $CMD -v -v -v' and printed out the log level set. It never changed. And I was looking with gdb into it and we always left `argconfig_parse()` early. It never got to the point where the `} else if (s->config_type == CFG_INCREMENT) {` came into play

ikegami-t, 2022-02-04:
Yes as you mentioned it looks `-v` does not work but only `--verbose` works so seems the short version option has any issue.
By the way seems `--verbose --verbose` multiple options only handled as same with the `--verbose` single option.