linux-nvme/nvme-cli #1362: nvme provides no simple method of obtaining the total size of the device

bezlovesnode, 2022-02-08:
As someone who codes a buildroot style project, I need device details such as Model, Serial and Size. Each of these can be obtained directly from /dev/nvme0 via id-ctrl. Except for Size. This needs to be obtained from nvme list, which spews forth a namespace list of everything the system can see. So, instead of my focus being on the actual device, which is found via a for nvme in /dev/nvme[0-9] loop. I then have to structure my loop around the namespaces found in the list?

What if a device has multiple namespaces? What if a namespace doesn't encompass the entire device? Such uncertainty. All I need is the total size of data a device has for itself, delivered inside the /dev/nvme[0-9]. I think it would only be fair to bring this inline with a similar hdparm function. Already, similar details are obtained by a combination of hdparm -i and hdparm -I in combination with some grep and cut commands.

keithbusch, 2022-02-08:
I am not sure I understand. You can get the total size via id-ctrl from the 'tnvmcap' field.

bezlovesnode, 2022-02-08:
This is also what I initially assumed, until recently, a device registered 0 in this field.

keithbusch, 2022-02-08:
I see. It is optional for devices that don't support multi-namespace capabilities. So in the case for those kinds of controllers, your request is for this tool to display the capacity accessible to a controller?

I'm not sure how there is an equivalent from hdparm. It looks like that only operates on block devices, but nvme controllers are not block devices. They just provide a parent handle to the namespaces that provide the block devices, and there doesn't seem to be an equivalent export in the scsi subsystem.

bezlovesnode, 2022-02-08:
My understanding is the controller currently has the ModelNumber and SerialNumber of the NVMe device itself. But, it's missing a capacity.

I'm making a loose comparison to hdparm, in that for example /dev/sda is a block device for the entire hard drive, but /dev/sda1 is the block device for the first partition. I guess people would expect to find the device details in /dev/nvme0, rather then the namespaces, which are found globally via nvme list -o json (ironically the normal output isn't formatted in a way which someone could easily grep / cut.)

keithbusch, 2022-02-08:
True, the tabular output isn't easily programmatically parsable. It's just a simple summary for human consumption. If we were doing this over again, json may have been the default output.

The nvme equivalent to scsi's /dev/sda is /dev/nvme0n1. The equivalent to partitions like /dev/sda1 is /dev/nvme0n1p1. In that sense, I think we have feature parity with hdparm. You will need to issue 2 commands, though: `nvme id-ctrl /dev/nvme0n1` for model+serial, and `nvme id-ns /dev/nvme0n1` for capacity.

If you consider a multi-LUN scsi target, each LUN gets their own scsi handle: /dev/sda, /dev/sdb, etc... There is no hdparm option, as far as I can see, to get the entire capacity of all the LUNs on a single scsi array. You have to query each LUN individually. That's essentially the same as what we're providing with NVMe.

bezlovesnode, 2022-02-08:
okay, I'm starting to see what you mean, so for example, a for namespace in /dev/nvme[0-9]n[0-9] should pick up every single namespace?

bezlovesnode, 2022-02-08:
However, for me nvme id-ns /dev/nvme0n1 is not picking up capacity.

keithbusch, 2022-02-08:
> However, for me nvme id-ns /dev/nvme0n1 is not picking up capacity.

It does provide the capacity. It's just a little obfuscarted. An example from my test machine

```
# nvme id-ns /dev/nvme0n1
NVME Identify Namespace 1:
nsze    : 0x1d1c5970
...
lbaf  0 : ms:0   lbads:9  rp:0x2 (in use)
```

You have to multiply nsze by the "in use" lbdas shift. Something like this little script:

```
# NSZE=$(nvme id-ns /dev/nvme0n1 | grep -e "nsze *: *0x.*$" | cut -d":" -f2
# SHIFT=$(nvme id-ns /dev/nvme0n1 | grep "in use" | grep -o "lbads:[0-9]*" | cut -d":" -f2)
# echo $((NSZE << SHIFT))
250059350016
```

keithbusch, 2022-02-08:
> okay, I'm starting to see what you mean, so for example, a for namespace in /dev/nvme[0-9]n[0-9] should pick up every single namespace?

That should mostly work, as long as you don't have more than 10 controllers or 10 namespaces per controller.

igaw, 2022-02-09:
The initial question was if there was a simple way to figure this out. Is this is something we should consider to add. I referring to the calculation part?

closed 2022-07-08 by igaw

oren0e, 2024-12-22:
I think that instead of this (btw I see that this calculation was added to "nvmcap" in the output from the `id-ns` command)

> # NSZE=$(nvme id-ns /dev/nvme0n1 | grep -e "nsze *: *0x.*$" | cut -d":" -f2)
# SHIFT=$(nvme id-ns /dev/nvme0n1 | grep "in use" | grep -o "lbads:[0-9]*" | cut -d":" -f2)
# echo $((NSZE << SHIFT))
250059350016

a capacity column in the table output from `nvme list` should be added. The capacity of the drive is basic info.

igaw, 2025-01-03:
I don't want to extend the output of `nvme list`. There is already too much information displayed. Maybe it could be added to the verbose output. Anyway, care to send a patch?