linux-nvme/nvme-cli #1372: ../nvme.h:26:10: fatal error: json.h: No such file or directory

Bobarshad, 2022-02-11:
$ ninja -C .build

ninja: Entering directory `.build'
[1/29] Compiling C object nvme.p/fabrics.c.o
FAILED: nvme.p/fabrics.c.o 
cc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/include/json-c -I/usr/include/uuid -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O3 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -MD -MQ nvme.p/fabrics.c.o -MF nvme.p/fabrics.c.o.d -o nvme.p/fabrics.c.o -c ../fabrics.c
In file included from ../fabrics.c:41:0:
../nvme.h:26:10: fatal error: json.h: No such file or directory
 #include <json.h>
          ^~~~~~~~

sc108-lee, 2022-02-11:
you need to check your json-c existence.
meson detect json-c that's why Added this "-I/usr/include/json-c " on your log.
but it seems there no file "json.h" on that location.

Bobarshad, 2022-02-11:
There is no "json.h" in "\usr\include\json-c" directory. Is there an application that I have to install? I have installed "sudo apt-get install libjsoncpp-dev," but it is not working.
```
 ls -l  /usr/include/json-c
total 108
-rw-r--r-- 1 root root  2498 Jul 26  2020 arraylist.h
-rw-r--r-- 1 root root  2666 Jul 26  2020 debug.h
-rw-r--r-- 1 root root   143 Feb 10 19:49 json_config.h
-rw-r--r-- 1 root root  1348 Jul 26  2020 json_c_version.h
-rw-r--r-- 1 root root   372 Jul 26  2020 json_inttypes.h
-rw-r--r-- 1 root root 42860 Jul 26  2020 json_object.h
-rw-r--r-- 1 root root  5007 Jul 26  2020 json_pointer.h
-rw-r--r-- 1 root root 10494 Jul 26  2020 json_tokener.h
-rw-r--r-- 1 root root  3611 Jul 26  2020 json_util.h
-rw-r--r-- 1 root root 10311 Jul 26  2020 linkhash.h
-rw-r--r-- 1 root root  4314 Jul 26  2020 printbuf.h
```

I

Bobarshad, 2022-02-11:
I have installed libjson-c-dev package, and it solved the problem.

    sudo apt install libjson-c-dev

eli-schwartz, 2022-02-13:
jsoncpp and json-c are two entirely different packages, but the original list of files you posted were legitimately from the correct package. It is just that you seemed to be missing half of the files from that directory.

I'm guessing that `sudo apt install libjson-c-dev` repaired your installation, or something.

igaw, 2022-02-14:
Looks like we should update the README, adding these install steps.

closed 2022-02-14 by igaw