linux-nvme/nvme-cli #1374: long double does not fit with json-c

sc108-lee, 2022-02-11:
[json_object_add_value_float](https://github.com/linux-nvme/nvme-cli/blob/master/nvme-print.c#L188

https://github.com/linux-nvme/nvme-cli/blob/master/nvme.h#L47

128 bits long double is not fit in 64 bits double
https://github.com/json-c/json-c/blob/master/json_object.c#L1066)

hreinecke, 2022-03-30:
Any particular reason why this needs to be a 'long double' in the first place?
To my understanding a 'long double' will just increase the accuracy of the value, but not change the value itself.

IE it should make a difference to the actual value if we were using 'long double' or 'double', no?

hreinecke, 2022-03-30:
Ah. GCC magic. Oh well.

But nevertheless; the json format is just for output, so we might as well have gcc print the 'long double' into a string and make it a string object.

igaw, 2022-04-01:
Well, the thing with `json_object_add_value_float()` is that is *not* a function from json-c. Instead all the nvme-print.c code is using the implementation in `utils/json.c`. This code handles the double longs correctly.

We might want to get rid of our own JSON implementation in favor of using json-c for nvme-print.c. Though I'd rather defer this work after the 2.0 release. This is something we can change under the hood.

igaw, 2022-07-08:
> Well, the thing with `json_object_add_value_float()` is that is _not_ a function from json-c. Instead all the nvme-print.c code is using the implementation in `utils/json.c`. This code handles the double longs correctly.

I got confused. We don't use `utils/json.c` anymore. I was able to rip it out. So I think we need to do the thing Hannes was proposing

closed 2022-07-11 by igaw