linux-nvme/nvme-cli #1394: NVMe/FC devices mounted through fstab file fail to mount at boot.

maurizio-lombardi, 2022-02-22:
I received a number of bug reports claiming that logical volumes created on NVMe/FC devices and mounted through fstab fail to mount at boot and system goes to emergency mode.
Commenting the fstab entries makes the system boot normally.

I think the problem is in the nvmefc-boot-connections.service file:
nvmefc-boot-connections.service must be run in order for the devices to appear, so it must be run before the mounts are attempted.
IMO this means that it must be ordered before local-fs-pre.target

igaw, 2022-02-22:
Sounds reasonable from your description. Though I don't have a lot of experience in this area.

@mwilck do you happen to have some insights on this one?

igaw, 2022-03-25:
@maurizio-lombardi any updates on this one?

igaw, 2022-05-18:
Any feedback?

maurizio-lombardi, 2022-05-18:
Hello @igaw , unfortunately no, the customer that reported this bug fell silent and we aren't getting any feedback on this issue

igaw, 2022-05-18:
Alright, let's close it. We can reopen if we can reproduce it somehow.

maurizio-lombardi, 2023-02-13:
@igaw I have a reproducer finally

maurizio-lombardi, 2023-02-13:
@mwilck I can boot with:

`Before=local-fs-pre.target
Conflicts=shutdown.target`

but if I add `DefaultDependencies=no` it doesn't boot anymore

igaw, 2023-02-21:
I think I see a possible race. When the system unit file is running and it writes to `/sys/class/fc/fc_udev_device/nvme_discovery`, it  is implicitly assumed that the transport driver has already loaded and was able to discover all remote ports (see 1nvme_fc_nvme_discovery_store()`). But I thought if this is the case, the nvme subsystem will still issue a udev event which we would then handle in user space via the udev rules. Obviously, this only works if udev is also up and running (which I asssume this is the case).

igaw, 2023-02-21:
Looking at the logs more closely, the qla2xxx driver has found all ports long before systemd comes into play. So my race theory is wrong.

igaw, 2023-02-21:
The write to `/sys/class/fc/fc_udev_device/nvme_discovery` will just trigger the kernel to iterate over the list of ports and issue an new udev event  (see `nvme_fc_signal_discovery_scan()`). If we assume this part works fine (the qla2xxx driver has found those ports long before systemd start running) then user space is not ready to consume this event. And looking at the udev startup log entries, I see a different ordering between the good and the bad case. Any idea how we could verify if the udev daemon is not ready?

igaw, 2023-02-21:
@maurizio-lombardi could you enable the `pr_debug()` in `kobject_uevent_env()` in early boot? Then we should see if the kernel is actually signalling user space or not. This would prove or disprove my theory.

mwilck, 2023-02-21:
@maurizio-lombardi , could you try adding
```
DefaultDependencies: no
After=systemd-udevd.service
```
If that doesn't help, perhaps try `After=systemd-udev-trigger.service`

wxiong38, 2023-02-21:
I tried all of above suggestions. none of them works for me.

I added the following line in nvmefc-boot-connections.service, it works in my environment.
Before=local-fs.target

mwilck, 2023-02-21:
This makes no sense to me. In https://github.com/systemd/systemd/blob/01e4f03f344856c103626e309d79321aa3e81641/src/core/service.c#L639, `DefaultDependencies=yes` (the default) would order a service **after** `sysinit.target` and `basic.target`, which are both in turn ordered **after** `local-fs.target`. I fail to see how it would be possible to revert this ordering just by specifying a contradictory `Before=local-fs.target` or `Before=local-fs-pre.target` dependency. 

With `DefaultDependencies=no`, we'd be able to cleanly order the service before `local-fs-pre.target`. I have no idea why that doesn't work for you guys. If this is set, additional dependencies must be set in order to make sure that `nvmefc-boot-connections.service` isn't started prematurely. This was the rationale behind `After=systemd-udevd.service`. No idea why that doesn't help, but it's hard to tell without detailed boot logs.

wxiong38, 2023-02-21:
I can take boot log for you. What options you want to add in nvmefc**.service?

maurizio-lombardi, 2023-02-22:
> @maurizio-lombardi , could you try adding
> 
> ```
> DefaultDependencies=no
> After=systemd-udevd.service
> ```
> 
> If that doesn't help, perhaps try `After=systemd-udev-trigger.service`

Doesn't work for me.

@igaw tomorrow I will enable the pr_debug() in kobject_uevent_env() and send the logs

igaw, 2023-02-22:
I've setup a similar setup and it looks like I am able to reproduce it:

```shell
localhost:~ # echo add > /sys/class/fc/fc_udev_device/nvme_discovery 
KERNEL[671.465514] change   /devices/virtual/fc/fc_udev_device (fc)
ACTION=change
DEVPATH=/devices/virtual/fc/fc_udev_device
SUBSYSTEM=fc
FC_EVENT=nvmediscovery
NVMEFC_HOST_TRADDR=nn-0x200000109b579ef5:pn-0x100000109b579ef5
NVMEFC_TRADDR=nn-0x201700a09890f5bf:pn-0x201900a09890f5bf
SEQNUM=4864

KERNEL[671.466391] change   /devices/virtual/fc/fc_udev_device (fc)
ACTION=change
DEVPATH=/devices/virtual/fc/fc_udev_device
SUBSYSTEM=fc
FC_EVENT=nvmediscovery
NVMEFC_HOST_TRADDR=nn-0x200000109b579ef5:pn-0x100000109b579ef5
NVMEFC_TRADDR=nn-0x201700a09890f5bf:pn-0x201b00a09890f5bf
SEQNUM=4865

[...]

localhost:~ # nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------

localhost:~ # nvme connect-all -t fc -w nn-0x200000109b579ef6:pn-0x100000109b579ef6 -a nn-0x201700a09890f5bf:pn-0x201800a09890f5bf
[ 1041.614122][ T3455] nvme nvme0: NVME-FC{0}: create association : host wwpn 0x100000109b579ef6  rport wwpn 0x201800a09890f5bf: NQN "nqn.2014-08.org.nvmexpress.discovery"
[ 1042.328174][ T3455] nvme nvme0: queue_size 128 > ctrl maxcmd 32, reducing to maxcmd
[ 1042.332357][ T3455] nvme nvme0: NVME-FC{0}: controller connect complete
[...]
localhost:~ # nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme5n1          /dev/ng5n1            80-AABMxwn3xAAAAAAAF NetApp ONTAP Controller                  1          34.36  GB /  34.36  GB      4 KiB +  0 B   FFFFFFFF
/dev/nvme1n2          /dev/ng1n2            80-AABMxwn3xAAAAAAAB NetApp ONTAP Controller                  2          34.36  GB /  34.36  GB      4 KiB +  0 B   FFFFFFFF
/dev/nvme1n1          /dev/ng1n1            80-AABMxwn3xAAAAAAAB NetApp ONTAP Controller                  1          34.36  GB /  34.36  GB      4 KiB +  0 B   FFFFFFFF
```

As I expected the udev events are fired by the kernel but user space doesn't react on it.  udevadmin only show the kernel events but not the userspace events.

```shell
# ps -elf | grep udev
4 S root      1734     1  0  80   0 - 21580 do_epo 20:28 ?        00:00:02 /usr/lib/systemd/systemd-udevd
```

The udev daemon is running.

And then, when I comment out the entry in fstab and exit the emergency shell, I see discovery events on the shell.

igaw, 2023-02-23:
After playing for a while I found a combination which seems to work for my test setup. No idea if this is right or make sense in terms of dependencies etc. I suppose systemd expert should review it.

`nvmf-connect.target`
```
[Unit]
Description=All instances of nvmf-autoconnect daemon
After=basic.target
Before=local-fs-pre.target

```

`nvmf-connect@.service`
```
#
# Unit file used by 70-nvmf-autoconnect.rules.
#

[Unit]
Description=NVMf auto-connect scan upon nvme discovery controller Events
PartOf=nvmf-connect.target
Requires=nvmf-connect.target

[Service]
Type=simple
Environment="CONNECT_ARGS=%i"
ExecStart=/bin/sh -c "/usr/sbin/nvme connect-all --quiet `/bin/echo -e '${CONNECT_ARGS}'`"
```

`nvmefc-boot-connections.service`
```
[Unit]
Description=Auto-connect to subsystems on FC-NVME devices found during boot
ConditionPathExists=/sys/class/fc/fc_udev_device/nvme_discovery
DefaultDependencies=no
Before=local-fs-pre.target
After=systemd-udev-settle.service

[Service]
Type=oneshot
ExecStart=/bin/sh -c "echo add > /sys/class/fc/fc_udev_device/nvme_discovery"

[Install]
WantedBy=local-fs.target
```

maurizio-lombardi, 2023-02-23:
> After playing for a while I found a combination which seems to work for my test setup. No idea if this is right or make sense in terms of dependencies etc. I suppose systemd expert should review it.

Thanks! I'm going to test it on my setup right now

igaw, 2023-02-23:
Note, you systemd expert told me he still sees a cycle dependency. Also he suggest to replace `After=systemd-udev-settle.service` with `After=sys-devices-virtual-fc-fc_udev_device.device`. Anyway the thing I was told is that with `DefaultDependencies=no` there is no guarantee that anything is there.

I think the target needs to be placed after basic (which I think includes the udev daemon being there and it was able to load the default rules living in `/usr/lib/udev/rules.d`) and before we try to start mounting stuff from `fstab`. 

Just to make sure I am pretty much a noob with this stuff. So take everything I say with same grain of salt.

mwilck, 2023-02-23:
We should *not* use `After=systemd-udev-settle.service`.  That service has been deprecated for a while, and won't be in the boot transaction normally.

mwilck, 2023-02-23:
>  I was told is that with DefaultDependencies=no there is no guarantee that anything is there.

That's true. We need to find appropriate `Requires=` and `After=`  directives that make sure what we need for nvmf is present. That was the rationale of my previous suggestions, which unfortunately weren't sufficient.

It's important to understand that `Before=local-fs-pre.target` **implies** `DefaultDependencies=no`. Without it, we get a dependency cycle and the outcome is essentially random. Even if we have heard "it works" about this configuration multiple times now, it is not the right thing to do.

`After=sys-devices-virtual-fc-fc_udev_device.device` is a very interesting suggestion, but I am unsure what systemd would do if `fc_udev_device` doesn't exist (not unlikely during boot).  AFAICT it would just ignore the `After=` dependency in this case; We'd need a `Requires=` dependency as well.

I think we need to pull in a separate unit that simply runs a "modprobe nvme-fc". This unit should be required by `nvmefc-boot-connections.service with both `Requires=` and `After=`, and should have `DefaultDependencies=no` set. The unit will be executed early, and will make sure `fc_udev_device` is available. 

We also need to set the `systemd` udev tag on `fc_udev_device` if we want the dependency on the device unit to work.

igaw, 2023-02-23:
As far I can tell, this stuff just work due to pure timing luck. There is nothing which guarantees the implicit assumed ordering of events. The modules are loaded very early in the boot process and when the systemd finally gets time to run it will find the drivers and `nvme-fc` already finished with all their initialization. 

Anyway  it's also worth to read up on the original commit on the auto connect systemd files. I'm pretty sure I understood the meaning of the `nvmf-connect.target` wrong. 391d3b12248a ("nvme-cli: nvmf auto-connect scripts")

maurizio-lombardi, 2023-02-23:
> After playing for a while I found a combination which seems to work for my test setup. No idea if this is right or make sense in terms of dependencies etc. I suppose systemd expert should review it.

FYI. this doesn't work for me, the boot process gets stuck;
same result with `After=sys-devices-virtual-fc-fc_udev_device.device`

igaw, 2023-02-23:
When you enabled the systemd debug logs, you can see the reason why `nvmf-connect@.servce` is not executed.

igaw, 2023-03-03:
Thanks Martin's help, this here is the solution:

` nvmf-connect@.service`
```
#
# Unit file used by 70-nvmf-autoconnect.rules.
#

[Unit]
Description=NVMf auto-connect scan upon nvme discovery controller Events
DefaultDependencies=no
After=systemd-udevd.service
Before=local-fs-pre.target
PartOf=nvmf-connect.target
Requires=nvmf-connect.target
```

`nvmefc-boot-connections.service`
```
[Unit]
Description=Auto-connect to subsystems on FC-NVME devices found during boot
ConditionPathExists=/sys/class/fc/fc_udev_device/nvme_discovery
DefaultDependencies=no
After=systemd-udevd.service
Before=local-fs-pre.target
```

igaw, 2023-03-03:
@wxiong38 in cause you have some time to verify the above configuration, it would really help.

maurizio-lombardi, 2023-03-03:
@igaw I will test this next Monday

wxiong38, 2023-03-06:
Hi Daniel, It works in sles15sp5.

maurizio-lombardi, 2023-03-06:
@igaw tested on RHEL9, it works for me too

igaw, 2023-03-07:
Thanks for testing. I'll prepare a PR.

PR #1842 (2023-03-07): fabrics: Fix ordering for auto connect services
In order to be able to mount file systems via /etc/fstab we have to make sure that the corresponding auto connect services have been executed. Because the mounting of the local filesystem happens very early in the boot we have to carefully sort these service file into the boot process.

First, we have to disable the DefaultDependency as this will automatically add dependency on sysinit.target which is too late (after local mounts). Though without the default dependency we have to provide a Before and After conditions.

The Before is simple as we  [... cut]
closed 2023-03-07 by igaw