linux-nvme/nvme-cli #1421: Create-ns using Admin-Passthru

njwilder, 2022-03-11:
Hello!

Problem: I must send a namespace management "create" command through admin-passthru as opposed to just using create-ns for reasons. Say there is no way of avoiding this and this command must be issued by admin-passthru. I saw a similar-ish issue (#142) from a while ago, but it pretty much just goes "use the sub-command" [which I agree with!].

The confusion comes from when I need to send other information not covered by DW10 and DW11 in the NVMe specification. So for example: I try sending a command: 
"sudo nvme admin-passthru /dev/nvme0 -o 0x0d --cdw10 0 --cdw11 0"
but this command does not account for things that you need to give to create-ns like nsze, ncap, and blockSize. I see that the spec mentions the Data Pointer and I feel like this would be where you would specify these fields, but I am just not seeing it.

One thought was that the fields need to be written to memory, then "pointed" to by datapointer, but even if I knew how to set datapointer to a written spot in memory, how do I find out what the values that need to be written into memory are?

keithbusch, 2022-03-11:
if you want to use the generic passthrough command, you will need to generate the data payload you want to attach to this command. The easiest way to do that might be to write a special program, allocate a 'struct nvme_id_ns' structure, then fill in the parts that satisfy the namespace you wish to create. Write that structure in to a file, then attach that file to the data-out payload for the passthrough command.

njwilder, 2022-03-11:
Genius! I ended up getting confused because I am using the nvme-cli with a python abstraction over it, and I did not fully realize that the "NM - Data Structure for Create" was actually referring to something that needs to be passed to the command. Thank you!

closed 2022-03-11 by njwilder

keithbusch, 2022-03-11:
Thanks! That's awesome you were able to make sense from that high-level hint! :)