linux-nvme/nvme-cli #1422: nvme list no longer shows full device path

mjekdi, 2022-03-11:
The nvme list command used to display the full device path, e.g. /dev/nvme0n1 instead of nvme0n1. This change breaks some of our scripts. Is this a bug or intentional change?

$ nvme version
nvme version 1.15
$ sudo nvme list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          50026B728253677F     KINGSTON SEDC1000M960G                   1         960.20  GB / 960.20  GB    512   B +  0 B   S68F0121

$ nvme version
nvme version 2.0
$ sudo nvme list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
nvme0n1               201728135452         Micron_7300_MTFDHBE3T8TDF                1          10.74  GB /  10.74  GB    512   B +  0 B   9542010

keithbusch, 2022-03-11:
I believe it was intentional, due to the fact /dev/ is repetitive and could be implicitly assumed.

I do understand the problems that changing output has on scripts, so let me see if we can easily change this back to the previous behavior.

hreinecke, 2022-03-16:
Might, but still I find the '/dev/nvme0n1' notation being the better choice. And we should modify the 2.x version to print it the same way.

igaw, 2022-03-16:
I am on it.

igaw, 2022-03-16:
TBH, I am a bit lost. With libnvme we are using the sysfs paths and I haven't found a way to map the found device to '/dev'. There is the major and minor available in `dev`. Though there is no simple way to go from this info to '/dev'. Parsing `/proc/mounts` or `/proc/self/mountinfo` just feels wrong. And ("add path to nvme list") assumed also everything lives in `/dev` or the path provided.

keithbusch, 2022-03-16:
Yeah, I know... I was thinking just assume /dev/ and prepend it to the output. I think that's a little silly, but I can't imagine it would be anything different.

igaw, 2022-03-17:
I was curious what udevadm is doing in the case of `udevadm info -rq name  /sys/class/nvme/nvme0`

So the main part starts here: 

https://github.com/systemd/systemd/blob/e8aba093b405bb53cd26d82d26436aa4836b2e83/src/udev/udevadm-info.c#L349

From there the uevent file is opened and DEVNAME is parsed. Eventually we end up in `device_set_devname()` which does:

https://github.com/systemd/systemd/blob/e8aba093b405bb53cd26d82d26436aa4836b2e83/src/libsystemd/sd-device/sd-device.c#L463-L482

So yep `/dev` is pretty much hard code, at least for any systemd based distro. 

And after reading up on https://github.com/linux-nvme/nvme-cli/commit/73a49f3e9933a359f5e77a20bcc733ce7bfcdaf9 ("add path to nvme list") I think there is nothing to do. We are using `/dev/class/nvme` as starting point, so these devices will be picked up.

Though for the SPDK users, in this case `/dev` will be wrong. One thing we could do is, checking if the device path is exists and if not just drop the prefix.

PR #1442 (2022-03-18): nvme: Print full device path
Restore nvme-cli 1.x behavior to print the full path to the device
file. We assume that the devtmpfs is mounted under /dev which is the
case for any systemd based distro and a de-facto standard.

But SPDK users see their device populated under /dev/spdk have their
device file appear. So instead blindly prefix /dev do a stat to check
if the file exist.

If our educated guesses fail, don't add the full path. This should be
really a corner case and as long no one complains don't overengineer.

Fixes: #1422
closed 2022-03-18 by igaw