linux-nvme/nvme-cli #1426: controller in bad state can lead to segv

mgerdts, 2022-03-14:
With a misbehaving controller, I found that `nvme list` can have a null pointer deref.

This is on Rocky Linux 8.5 with a fresh build of v1.16.  I first encountered it with v1.14.

```
$ ./nvme list
Failed to open /sys/class/nvme-subsystem/nvme-subsys0/nvme0: No such file or directory
Segmentation fault (core dumped)
```

Under gdb:

```
(gdb) run list
Starting program: /home/mgerdts/nvme-cli/nvme list
Missing separate debuginfos, use: yum debuginfo-install glibc-2.28-164.el8.x86_64
Failed to open /sys/class/nvme-subsystem/nvme-subsys0/nvme0: No such file or directory

Program received signal SIGSEGV, Segmentation fault.
0x000000000042c75a in scan_subsystem (nsid=0, ns_instance=0, s=<optimized out>) at nvme-topology.c:387
warning: Source file is more recent than executable.
387					if (n->ctrl->state &&
Missing separate debuginfos, use: yum debuginfo-install libhugetlbfs-2.21-17.el8.x86_64 libuuid-2.32.1-28.el8.x86_64
(gdb) list
382			for (i = 0; i < s->nr_namespaces; i++) {
383				n = &s->namespaces[i];
384				n->name = strdup(ns[i]->d_name);
385				for (j = 0; j < s->nr_ctrls; j++) {
386					n->ctrl = &s->ctrls[j];
387					if (n->ctrl->state &&
388							!strcmp(n->ctrl->state, "live") &&
389							!scan_namespace(n))
390						break;
391				}
(gdb) print n->ctrl
$1 = (struct nvme_ctrl *) 0x6a28f0
(gdb) print n->ctrl->state
$2 = 0x0
```

In this case `nvme_get_ctrl_attr` returned NULL because the `state` file didn't exist:

```
Breakpoint 1, nvme_get_ctrl_attr (path=0x6a4a30 "/sys/class/nvme-subsystem/nvme-subsys0/nvme0", attr=attr@entry=0x464e1d "state") at nvme-topology.c:60
60		ret = asprintf(&attrpath, "%s/%s", path, attr);
(gdb) print attr
$3 = 0x464e1d "state"
(gdb) print path
$4 = 0x6a4a30 "/sys/class/nvme-subsystem/nvme-subsys0/nvme0"
(gdb) next
61		if (ret < 0)
(gdb)
64		value = calloc(1, 1024);
(gdb)
65		if (!value)
(gdb)
68		fd = open(attrpath, O_RDONLY);
(gdb)
69		if (fd < 0)
(gdb) print fd
$5 = -1
(gdb) print attrpath
$6 = 0x6a5300 "/sys/class/nvme-subsystem/nvme-subsys0/nvme0/state"
```

The subsystem was in a weird state, with `nvme0` as a dangling link.

```
[root@ws2 nvme-cli]# ls -l /sys/class/nvme-subsystem/nvme-subsys0/
total 0
-r--r--r--. 1 root root 4096 Mar 14 17:01 firmware_rev
-rw-r--r--. 1 root root 4096 Mar 14 17:34 iopolicy
-r--r--r--. 1 root root 4096 Mar 14 16:37 model
drwxr-xr-x. 3 root root    0 Mar 14 16:55 ng0n1
lrwxrwxrwx. 1 root root    0 Mar 14 16:55 nvme0 -> ../../../pci0000:60/0000:60:01.1/0000:61:00.2/nvme/nvme0
drwxr-xr-x. 8 root root    0 Mar 14 16:55 nvme0n1
lrwxrwxrwx. 1 root root    0 Mar 14 16:55 nvme7 -> ../../../pci0000:60/0000:60:01.1/0000:61:00.2/nvme/nvme7
drwxr-xr-x. 2 root root    0 Mar 14 17:32 power
-r--r--r--. 1 root root 4096 Mar 14 16:39 serial
-r--r--r--. 1 root root 4096 Mar 14 16:55 subsysnqn
lrwxrwxrwx. 1 root root    0 Mar 14 16:39 subsystem -> ../../../../class/nvme-subsystem
-rw-r--r--. 1 root root 4096 Mar 14 16:37 uevent
```

The latest 2.0 RC code (tested at commit 7ff90b5f5ff718f760b266c06eecdad359d86a28) does not have this problem, despite no changes in `nvme-topology.c` between the broken and working versions.  If there will be future 1.x releases, this may need some attention.

igaw, 2022-03-15:
The scan code is completely new and lives in libnvme, so this could explain why 2.0-rc is not affected.

But what is really strange `if (n->ctrl->state && ...)` is there to prevent the crashing strcmp() call? 

Are you sure the binary is matching with the debug source? There is `warning: Source file is more recent than executable.` 

The binary might miss this check. It was added in bace574bbe55 ("nvme-topology: fix controller check in scan_subsystem()") in v1.15.

mgerdts, 2022-03-16:
Oops, the code I pasted was a trivial fix, which I'm not sure is the right fix.

I hit the problem again and reproduced with a fresh build.

```
 gdb ./nvme
GNU gdb (GDB) Red Hat Enterprise Linux 8.2-16.el8
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./nvme...done.
(gdb) run list
Starting program: /home/mgerdts/nvme-cli/nvme list
Missing separate debuginfos, use: yum debuginfo-install glibc-2.28-164.el8.x86_64
Failed to open /sys/class/nvme-subsystem/nvme-subsys0/nvme0: No such file or directory

Program received signal SIGSEGV, Segmentation fault.
0x000000000042c75a in scan_subsystem (nsid=0, ns_instance=0, s=<optimized out>) at nvme-topology.c:387
387					if (!strcmp(n->ctrl->state, "live") &&
Missing separate debuginfos, use: yum debuginfo-install libhugetlbfs-2.21-17.el8.x86_64 libuuid-2.32.1-28.el8.x86_64
(gdb) where
#0  0x000000000042c75a in scan_subsystem (nsid=0, ns_instance=0, s=<optimized out>) at nvme-topology.c:387
#1  scan_subsystems (t=t@entry=0x7fffffffdbf0, subsysnqn=subsysnqn@entry=0x0, ns_instance=ns_instance@entry=0, nsid=nsid@entry=0, dev_dir=0x0)
    at nvme-topology.c:598
#2  0x0000000000410443 in list (argc=<optimized out>, argv=<optimized out>, cmd=<optimized out>, plugin=<optimized out>) at nvme.c:2292
#3  0x000000000042af02 in handle_plugin (argc=argc@entry=1, argv=argv@entry=0x7fffffffe090, plugin=0x69b440 <builtin>) at plugin.c:158
#4  0x0000000000402871 in main (argc=2, argv=0x7fffffffe088) at nvme.c:6616
(gdb) list
382			for (i = 0; i < s->nr_namespaces; i++) {
383				n = &s->namespaces[i];
384				n->name = strdup(ns[i]->d_name);
385				for (j = 0; j < s->nr_ctrls; j++) {
386					n->ctrl = &s->ctrls[j];
387					if (!strcmp(n->ctrl->state, "live") &&
388							!scan_namespace(n))
389						break;
390				}
391			}
```

That is with:

```
$ git log -n 1 --oneline
deee9ca (HEAD, tag: v1.16) release v1.16

$ git status
HEAD detached at v1.16
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	.build/
	.gdb_history
	subprojects/

nothing added to commit but untracked files present (use "git add" to track)
```

igaw, 2022-03-17:
Ah, got it. I was confused as well. We need to add the `if (n->ctrl->state` as you already have figured out. Let me spin a patch.

igaw, 2022-03-17:
Fix merged. Thanks for the bug report!

closed 2022-03-17 by igaw

mgerdts, 2022-03-24:
Fixed in commit f6eb85fc5d3ea7e32f5c4bae7de3e2cc80579fa6.  Thanks!

PR #1439 (2022-03-17): nvme-topology: Check if state variable is a valid string
The state variable might not be initialized at all, that is it can be
NULL. strcmp is not able to handle NULL gracefully.

Fixes: bace574bbe55 ("nvme-topology: fix controller check in scan_subsystem()")
Reported-by: mgerdts (github)

Fixes: #1426