linux-nvme/nvme-cli #1430: nvme-list - undefined attributes show as (null)

martin-belanger, 2022-03-15:
When an attribute is undefined (NULL), the output of `nvme list` displays `(null)` entries. I wasn't sure if that was by design or whether we should instead display either an empty string or maybe the string `N/A`?

Here's an example where the firmware version is undefined and shows as `(null)`.
```
$ sudo nvme list -v
...
Device   SN               MN                 FR       ...
-------- ---------------- ------------------ -------- ...
nvme18   3YW2643          dellemc-powerstore 2.1.1.0  ...
nvme9    20220217011349e8 DELL-SFSS          (null)   ...
nvme7    20220217011349e8 DELL-SFSS          (null)   ...
```

martin-belanger, 2022-03-15:
Personally, I don't really care about the `(null)`. Unfortunately, we have test engineers that don't agree with me and want me to fix it. They don't understand that `nvme-cli` is an open-source project (and I'm not the one deciding what goes into it :stuck_out_tongue_winking_eye:). 

Maybe we could define a wrapper function like this to replace a NULL by a default string to be printed.
```
static inline const char * string_or_default(const char * s, const char * default)
{
    return s ? s : default;
}
```

And it could be used in `nvme-cli` as follows:
```
printf("%-8s %-20s %-40s %-8s %-6s %-14s %-12s ",
       string_or_default(nvme_ctrl_get_name(c), "unknown"),
       string_or_default(nvme_ctrl_get_serial(c), "n/a"),
       string_or_default(nvme_ctrl_get_model(c), "n/a"),
       string_or_default(nvme_ctrl_get_firmware(c), "n/a"),
       string_or_default(nvme_ctrl_get_transport(c), "unknown"),
       string_or_default(nvme_ctrl_get_address(c), "unknown"),
       string_or_default(nvme_subsystem_get_name(s), "unknown"));

```

martin-belanger, 2022-03-15:
I vote to leave it as-is. The risk of breaking something is too great.

closed 2022-03-15 by martin-belanger

igaw, 2022-03-15:
I'd say major version updates are there for breaking a few things...