linux-nvme/nvme-cli #1438: fused operations: eg.compare and write

coderaskanoter, 2022-03-17:
can it send a fused operation(eg.compare and write fused operation) by nvme cli. If it can, can you give me a example of caw command?

keithbusch, 2022-03-17:
No, the kernel doesn't support it. We would be depending on the low level driver to dispatch fused commands in consecutive slots of the same queue, and it doesn't provide any such mechanism.

igaw, 2022-03-18:
Closing. There is not much we can do unless we have support from the kernel for this.  There were some attempts to get this feature working but it's not that simple see the discussion on the nvme mailing list:

  https://lore.kernel.org/linux-nvme/20210105224939.1336-2-clay.mayers@kioxia.com/T/

closed 2022-03-18 by igaw