linux-nvme/nvme-cli #1446: Building nvme in-tree fails to produce output

hreinecke, 2022-03-18:
When building nvme in-tree it fails to produce output:

~~~
# /tmp/nvme discover -t tcp -a 127.0.0.1
#
~~~

This is a regression introduced by f356ab0 ("fabrics: Allow to change sysconfdir for hostnqn and hostid file").

hreinecke, 2022-03-18:
In particular this change:
~~~
diff --git a/src/nvme/fabrics.c b/src/nvme/fabrics.c
index 01fd4ce..e70512e 100644
--- a/src/nvme/fabrics.c
+++ b/src/nvme/fabrics.c
@@ -41,9 +41,10 @@
 #define NVMF_HOSTID_SIZE       37
 #define UUID_SIZE              37  /* 1b4e28ba-2fa1-11d2-883f-0016d3cca427 + \0 */
 
+#define NVMF_HOSTNQN_FILE      "SYSCONFDIR/nvme/hostnqn"
+#define NVMF_HOSTID_FILE       "SYSCONFDIR/nvme/hostid"
+
 const char *nvmf_dev = "/dev/nvme-fabrics";
-const char *nvmf_hostnqn_file = "/etc/nvme/hostnqn";
-const char *nvmf_hostid_file = "/etc/nvme/hostid";
 
 const char *arg_str(const char * const *strings,
                size_t array_size, size_t idx)
~~~
looks really dodgy. What happened to the good old defines?

hreinecke, 2022-03-18:
Please, make SYSCONFDIR is compile-time define, and provide defaults in the code.

igaw, 2022-03-18:
Sorry, I didn't realize the pre compiler is not exchanging the SYSCONFDIR in the string. SYSCONFDIR is a normal define.

hreinecke, 2022-03-18:
The usual way would be to use
~~~
SYSCONFDIR ## "<any string>"
~~~
IIRC. Avoids using __stringify. But anyway.

igaw, 2022-03-18:
The problem is that SYSCONFDIR itself is not a string and I couldn't figure out how to get ## working except with the __stringify macros.

igaw, 2022-03-18:
So it's just a simple:

  `SYSCONFDIR "string"`

closed 2022-03-18 by igaw