linux-nvme/nvme-cli #1456: Just a question: Assembled json nodes are not printed properly in the latest snapshot.

franciscomunoz, 2022-03-25:
I noticed a that I'm not able to see the last  json node via json_print_object when compiling the latest version of the intel plugin

With SHA 53086d6e0c9792935e4, my tree looks like the one below (22 Sep 2021)

{
  "latstats" : {
    "type" : "read",
    "values" : {
      "bucket" : {
        "id" : 0,
        "start" : "0.00us",
        "end" : "9.00us",
        "value" : 0
      },
      "bucket" : {
        "id" : 1,
        "start" : "10.00us",
        "end" : "19.00us",
        "value" : 0
      },
      "bucket" : {
        "id" : 2,
        "start" : "20.00us",
        "end" : "39.00us",
        "value" : 0
      },
      "bucket" : {
        "id" : 3,
        "start" : "40.00us",
        "end" : "79.00us",
        "value" : 0
      },
      "bucket" : {
        "id" : 4,
        "start" : "80.00us",
        "end" : "159.00us",
        "value" : 0
      },
      "bucket" : {
        "id" : 5,
        "start" : "160.00us",
        "end" : "319.00us",
        "value" : 0
      },
      "bucket" : {
        "id" : 6,
        "start" : "320.00us",
        "end" : "639.00us",
        "value" : 0
      },
      "bucket" : {
        "id" : 7,
 "start" : "640.00us",
        "end" : "+INF",
        "value" : 0
      }
    }
  },
  "Average latency since last reset" : {
    "value in us" : 0
  }

, but i'm  just seeing

{
  "latstats":{
    "type":"read",
    "values":{
      "bucket":{
        "id":7,
        "start":"0.00us",
        "end":"+INF",
        "value":0
      }
    }
  },
  "Average latency since last reset":{
    "value in us":0
  }
}

Not sure if this is an issue. I already tried in an Ubuntu/RHEL hosts.

igaw, 2022-03-28:
Let me double check if d10feb0b1b95 ("nvme: Adapt to API rework") didn't introduce this regression.

igaw, 2022-03-30:
Unfortunately, I was not able to figure out where we broke things and I can't really test it (no hardware). From you bug report it's a Optane device, correct? 

There was no change in the JSON bits, so it's something with either the port from the monolithic branch to master branch or the API libnvme changes. Can't really say.

@franciscomunoz would it possible for you to figure out what's going on?

igaw, 2022-03-30:
BTW, do you get the correct output when using the terminal output (that is without --json)? If so, at least we know it's in the JSON  code path and not in the data collecting path.

franciscomunoz, 2022-04-01:
Let me do some experiments tomorrow,

hreinecke, 2022-04-01:
There's always the chance that the latency statistics had been reset in the HW itself; after all, these are statistics, and as you have the setting 'Average latency since last reset' it would imply that the statistics _can_ be reset. And if they are reset I would expect the same output which you've seen.

franciscomunoz, 2022-04-01:
Hi, @igaw @hreinecke 
Ok, I did a quick check in the intel plugin code and found that change below helps to print all the buckets.  In short, function 
json_object_add_value_object doesn't append objects with the same string value. In my case it sees "bucket" several times and values are not appended. 

diff --git a/plugins/intel/intel-nvme.c b/plugins/intel/intel-nvme.c
index 3a73ff9..6bf4c52 100644
--- a/plugins/intel/intel-nvme.c
+++ b/plugins/intel/intel-nvme.c
@@ -729,14 +729,15 @@ static void json_add_bucket_optane(struct json_object *bucket_list, __u32 id,
                                   __u32 lower_us, enum inf_bound_type start_type,
                                   __u32 upper_us, enum inf_bound_type end_type, __u32 val)
 {
-       char buffer[BUFSIZE];
+       char buffer[14];
        struct json_object *bucket = json_create_object();

        init_buffer(buffer, BUFSIZE);

+       snprintf(buffer, sizeof(buffer) -1 , "bucket id: %u", id);
+
        json_object_add_value_object(bucket_list,
-               "bucket", bucket);
-       json_object_add_value_int(bucket, "id", id);
+               buffer, bucket);

I further  debugged this thing and it looks like the json function (json_object_object_add_ex)  hash the keys and if a given key is already there, then the value is refreshed. See
[image]

This is a new behavior. Please let me know your thoughs.
--Francisco.

hreinecke, 2022-04-04:
Well, I guess libjson-c is actually correct; there can only be one json object with a specific name (at any given level). If you have more objects with the same name it really needs to be an array.
So the original json representation was actually wrong, and it should have been an array of buckets to begin with.

igaw, 2022-04-06:
The PR #1474 fixed the problem. Thanks everyone!

closed 2022-04-06 by igaw