linux-nvme/nvme-cli #1465: Is it possible to verify that the data is encrypted after running --ses=2?

PtrStruct, 2022-03-30:
So reading about the various format options and their parameters, there's one that sticks out as more interesting than the others. And it's the `--ses=2`

**_2: Crypto Erase, this completes much faster (under 1 second in most cases) by swapping out the cryptographic key so that all data is rendered unreadable. Like all three cases, this will deallocate the LBAs and some drives may support deterministic read zero after TRIM for subsequent reads._**

Calling `sudo nvme format _device_ -ses=2 --force` shouldn't remove any data, only swap out the existing DEK that's generated within the SED if I understand things correctly. Is there a way to verify this? After running that command is the a GUI tool that would visually show me that the data is encrypted? 
I'm pretty sure that once this command has been invoked, the drive is un-mountable, meaning that I won't see it in the File Explorer, forcing me to re-allocate the space for it which comes with a quick format. If I do that I don't get the chance to see the encrypted data.

hreinecke, 2022-03-30:
That understanding is correct. Data will be unavailable after that operation, and depending on the implementation you might be able to figure out the result by looking at the 'unallocated capacity' (UNVMCAP) field in the identify controller output.
But in general I wouldn't bet on this information to be visible.
And yes, you would need to re-allocate data on the device. I would even think that all namespaces are gone, and that you would need to create new namespaces.
Additionally I'd be really surprised if Windows would be able to do that on it's own.

So use with _extreme_ care.

PtrStruct, 2022-03-30:
Ah! Thanks for confirming that.

Could you please elaborate on this part?
_by looking at the 'unallocated capacity' (UNVMCAP) field in the identify controller output_

How would I go about doing that?

igaw, 2022-04-01:
```shell
$ nvme id-ctrl /dev/nvme0n1 -H

...
unvmcap   : 0
[127:0] : 0     Unallocated NVM Capacity (UNVMCAP)
...
```

closed 2022-04-01 by igaw