linux-nvme/nvme-cli #1472: Build Failure when nvme cli and libnvme are out of sync

jeffreyalien, 2022-04-01:
@igaw Could you explain on the libnvme.wrap is supposed to work?  In particular, how does it insure that you have the right commit level of libnvme code when building nvme cli?  When cloning a new version of both nvme cli along with libnvme, is there additional command or clone option needed to automatically get required commit level of the libnvme code?

igaw, 2022-04-04:
I've tried to summarize it here: https://github.com/linux-nvme/nvme-cli/blob/master/README.md#meson-tips

The wrap feature of meson is essentially the same thing as 'git submodules'. Thus the same workflow applies. If the section in the README is not explaining it in a understandable way, we should ask someone with better English skills than me to provide a better text.

closed 2022-04-04 by igaw

igaw, 2022-04-04:
In this case you have libnvme installed on your system. meson will use the installed library. Check also the excellent meson documentation: https://mesonbuild.com/Wrap-dependency-system-manual.html

jeffreyalien, 2022-04-05:
@igaw Thanks for the documentation links.  Everything worked as expected once I removed/uninstalled the version of libnvme I had on my system.  I had installed it manually before I really understood how meson and the dependency checking worked.

igaw, 2022-04-06:
Glad you could resolve the issue. The wrap feature is really great, though old time Makefile and autohell users have to learn something new :)

eli-schwartz, 2022-04-06:
A couple of notes:
- you can have subprojects/libnvme/ be both a meson wrap file, and a git submodule -- in this case, Meson will use `git submodule update --init` to fetch it, if nvme-cli is a git checkout, and `git clone` the `*.wrap` url if nvme-cli is a tarball without git metadata
- If you want to ignore the system installed libnvme library, use `meson setup builddir --wrap-mode=forcefallback` or `meson setup builddir --force-fallback-for=libnvme`

jeffreyalien, 2022-04-06:
@igaw @eli-schwartz  Here's my attempt at updating the meson tips section of the readme file.  Let me know if you have any comments or suggestions.   Or if it looks good to you, I'll create a PR for it.  

In case meson doesn't find libnvme header files (via pkg-config) it will fallback using subprojects. 
meson checks out libnvme in subprojects directory as git tree once and then will checkout to the 
commit level specified in the libnvme.wrap file revision parm.  The libnvme code level will not 
change unless explicitly told by a change to the revision specified in libnvme.wrap. That means 
the current branch is updated via git, the subprojects/libnvme branch will only be updated if the 
wrap file is changed.  

The submodule code and be updated manually via normal git operations or with the following
command:

$ meson subprojects update

jeffreyalien, 2022-04-06:
@eli-schwartz I added that comment.  Maybe it should state "the subprojects/libnvme branch will only be updated if the revision value in wrap file is updated." ??  At least that's what I noticed when experimenting with different changes.

igaw, 2022-04-06:
I think the point is, that subprojects wont be automatically updated. You need to explicitly do something like 'meson subprojects update' for example.

eli-schwartz, 2022-04-06:
Right.

Whether the wrap file is changed or not is irrelevant and won't be considered.

The first time, and every other time, that you ask it to update, by:
- trying to build the project, and needing that subproject
- running `meson subprojects download`
- running `meson subprojects update`

Meson will do one of three things:
- "the subproject is not available so I will update it now"
- "the subproject is available, what version is irrelevant, skip it as already made available"
- in the specific case of `update`, it will try to fetch and reset to the version specified in the wrap, even if the wrap file wasn't changed

In that third case, you can point to a branch and that branch should be updated even without editing the .wrap file at all.

If you specify a particular revision, then meson will still update it to that revision if you have gone (outside of meson) and `git checkout`'ed a different revision. That's because meson will reset to that revision either way, but resetting to the revision you are already on is a no-op.

jeffreyalien, 2022-04-06:
Ok, that makes sense.  I thought I saw it automatically update the libnvme code after I changed the wrap revision yesterday.  I just redid that experiment and it did not update so I must have done a subprojects update to get it updated.  Here's an updated meson tips section based on the above comments - not much different then the original text, but I added some new text to help clarify :

####  Meson Tips  ####
In case meson doesn't find libnvme header files (via pkg-config) it will fallback using subprojects. 
meson checks out libnvme in subprojects directory as git tree once to the commit level specified in
the libnvme.wrap file revision parm.  After this initial checkout, the libnvme code level will not 
change unless explicitly told.  That means if the current branch is updated via git, the 
subprojects/libnvme branch will not updated accordingly.  To update it, either use the normal git 
operations or the command: 

$ meson subprojects update

eli-schwartz, 2022-04-06:
Yeah, the newly tweaked wording looks good. 👍