linux-nvme/nvme-cli #1475: nvme/tcp uses port #4420 (instead of port #8009) by default for discovery

martin-gpy, 2022-04-04:
I noticed nvme/tcp in libnvme still uses port #4420 (instead of port #8009) by default for discovery. For e.g, if one runs 'nvme discover -t tcp -w 192.168.1.114 -a 192.168.1.113', it ends up wrongly using port #4420 instead of port #8009, for creating the discovery controller here.

martin-gpy, 2022-04-04:
In this code snippet from src/nvme/tree.c : set_default_trsvcid():

if (!strcmp(c->transport, "tcp")) {
       if (c->discovery_ctrl) {
                /* Default port for NVMe/TCP discovery controllers */
                 c->trsvcid = strdup(__stringify(NVME_DISC_IP_PORT));
                } else {
                        /* Default port for NVMe/TCP io controllers */
                        c->trsvcid = strdup(__stringify(NVME_RDMA_IP_PORT));
                }

So though the above code sets the trsvcid to port #8009 by default for nvme/tcp discovery, the discovery_ctrl is still not set here when invoked from nvme_create_ctrl() -> set_default_trsvcid(). So it doesn't enter the above 'if (c->discovery_ctrl)' condition, and hence fails.

martin-gpy, 2022-04-04:
Unfortunately, one also can't identify whether it is a discovery controller by checking its well-known subsystem nqn in nvme_create_ctrl(), since with TP 8013, discovery controllers can also have their own unique subsystem nqn.

igaw, 2022-04-05:
@martin-gpy is the patch #1480 working for you?

martin-gpy, 2022-04-06:
> @martin-gpy is the patch #1480 working for you?

Yes, gave this try and the changes look good. I now see NVMe/TCP discovery controllers properly using port #8009 whereas I/O controllers use port #4420.

igaw, 2022-04-06:
Thanks for testing!

PR #1480 (2022-04-06): fabrics: Set default trsvcid ports for TCP and RDMA
libnvme doesn't know at controller creation if the new controller is
going to be discovery controller. But nvme-cli knows it, so just set
defaults on the caller level.

Fixes: #1475