linux-nvme/nvme-cli #1476: Wrong KATO value assigned to I/O controllers during nvme connect-all with -p option

martin-gpy, 2022-04-04:
Noticed that whenever nvme connect-all is run with the -p option, it assigns the same 30s KATO value to both the persistent discovery controller as well as the I/O controller. Again this problem is seen only with the -p option, and not otherwise to the nvme connect-all:

$ nvme list-subsys
$

$ nvme connect-all -t fc -w nn-0x20000024ff19ab70:pn-0x21000024ff19ab70 -a nn-0x203400a098fd4ca6:pn-0x203500a098fd4ca6
$

$ nvme list-subsys
nvme-subsys1 - NQN=nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119
\
 +- nvme1 fc traddr=nn-0x203400a098fd4ca6:pn-0x203500a098fd4ca6,host_traddr=nn-0x20000024ff19ab70:pn-0x21000024ff19ab70 live

$ cat /sys/class/nvme-subsystem/nvme-subsys1/nvme1/kato
5

$ nvme disconnect-all

$ nvme list-subsys
$

$ nvme connect-all -t fc -w nn-0x20000024ff19ab70:pn-0x21000024ff19ab70 -a nn-0x203400a098fd4ca6:pn-0x203500a098fd4ca6 -p

$ nvme list-subsys
nvme-subsys1 - NQN=nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119
\
 +- nvme1 fc traddr=nn-0x203400a098fd4ca6:pn-0x203500a098fd4ca6,host_traddr=nn-0x20000024ff19ab70:pn-0x21000024ff19ab70 live
nvme-subsys0 - NQN=nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery
\
 +- nvme0 fc traddr=nn-0x203400a098fd4ca6:pn-0x203500a098fd4ca6,host_traddr=nn-0x20000024ff19ab70:pn-0x21000024ff19ab70 live

$ cat /sys/class/nvme-subsystem/nvme-subsys0/nvme0/kato
30

$ cat /sys/class/nvme-subsystem/nvme-subsys1/nvme1/kato
30

martin-gpy, 2022-04-04:
Problem comes from set_discovery_kato() invoked from nvmf_discover(), which is executed for both discovery as well as I/O controllers. This should be called only for discovery controllers alone.

igaw, 2022-04-06:
As @hreinecke pointed, this behavior is intentional. It's questionable to have different keep alive settings for different controller types. As you pointed out this a deviation from the previous behavior. I agree with Hannes there is nothing gained on having different keep alive timeout values for the discovery controllers and  I/O controllers. The only argument I see is to keep the 'load' for discovery controller lower. But this argument is only valid for persistent discovery controllers. And if you really want a different setting you can always create a persistent discovery controller as a separate step. 

@keithbusch do you have any opinion on this topic?

martin-gpy, 2022-04-06:
> As @hreinecke pointed, this behavior is intentional. It's questionable to have different keep alive settings for different controller types. As you pointed out this a deviation from the previous [...]

All that the spec mandates is the following:

- Non-persistent regular discovery controllers should have a zero KATO value.
- Persistent discovery controllers should have a non-zero KATO value. I/O controllers too are expected to have a non-zero KATO.

In the current Linux implementation, the following default values were assigned to the various controllers:

Non-persistent discovery controllers  - KATO default set to zero 
Persistent discovery controllers - KATO default set to NVMF_DEV_DISC_TMO (i.e. 30s)
I/O controllers - KATO default set to NVME_DEFAULT_KATO (i.e. 5s)

So are you planning to retain the above default values itself for legacy reasons? Or are you suggesting new default values here? Please clarify.

martin-gpy, 2022-04-06:
Also, the new libnvme/nvme-cli v2.x was assigning different default values for the I/O controllers for different nvme commands, as explained in the above description:

For e.g. during a 'nvme connect-all', the I/O controllers had a default KATO of 5s. But when calling 'nvme connect-all -p', the I/O controllers ended up having a default value of 30s. See above for details.

That's a bug, and needs to be corrected - you obviously need the same default KATO value assigned irrespective of whatever nvme command is called. Hence raised this issue.

igaw, 2022-04-08:
As far I understood the situation we have two cases to consider:
- a) the user provides a KATO value on the command line: use this value for all type controllers
- b) the user doesn't provide the a KATO: use the current defaults which are different for I/O and discovery controllers.

I need to check what we currently do and how a) and b) could be supported. The issue is that `--persistent`  argument is making everything really complex. I wonder if it wouldn't be better to ban the `--persistent` from `connect-all`

igaw, 2022-04-08:
I've tested #1489 

- `nvme connect-all ...` will set kato to 5 seconds for the discovered I/O controller.
- `nvme connect-all ... -p` will results in I/O controller set to 5 seconds and the discovery controller to 30 seconds (backwards compatible value). 
- `nvme connect-all ... -p -k 23` will set the kato value to 23 seconds for all controllers.

So I think this should address a) and b).

igaw, 2022-04-08:
#1489 is merged, closing this issue.

closed 2022-04-08 by igaw