linux-nvme/nvme-cli #1496: WDC: unsupported device for this command

Gigas002, 2022-04-11:
Probably I'm doing something wrong, but calling `wdc` commands on my western digital drives returns `ERROR : WDC: unsupported device for this command`. Here's output of my `sudo nvme list` (with an exception for serial number):

```text
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme1n1          /dev/ng1n1            xxxxxxxxxxxx         WDS100T1X0E-00AFY0                       1           1,00  TB /   1,00  TB    512   B +  0 B   614600WD
/dev/nvme0n1          /dev/ng0n1            xxxxxxxxxxxxxxxx     KINGSTON OM8PDP3512B-AA1                 1         512,11  GB / 512,11  GB    512   B +  0 B   EDFK0S03
```

And what I'm trying to do:

```text
sudo nvme wdc vs-temperature-stats /dev/nvme1n1
ERROR : WDC: unsupported device for this command
sudo nvme wdc drive-log /dev/nvme1n1
ERROR : WDC: unsupported device for this command
```

igaw, 2022-04-14:
The wdc plugin filters out devices it does not support. So this might be here the case. Let me add @jeffreyalien he might be able to tell more about your device.

jeffreyalien, 2022-04-14:
I believe a WDS100T1X0E-00AFY0 model drive has a pci device id of 0x5011 and a product # of SN810.  Looks like the only WDC plugin command supported for this drive is the vs-internal-log command.

Gigas002, 2022-04-15:
Ah, I understand. Thank you for explanation

igaw, 2022-04-18:
Closing bug.

closed 2022-04-18 by igaw