linux-nvme/nvme-cli #1505: nvme connect-all reports report 'already connected' anymore

igaw, 2022-04-14:
If `nvme connect-all ` is executed more than once we get this error:

```shell
$ .build/nvme connect-all -t tcp --traddr=10.161.8.24  --trsvcid=4421
Failed to write to /dev/nvme-fabrics: Operation already in progress
failed to connect controller, error 1006
```

The problem is that libnvme doesn't return` errno=EALREADY` anymore and the check 
for `EALREADY` in __discover() fails:

```c
			} else if (errno == EALREADY && !quiet) {
				char *traddr = log->entries[i].traddr;

				space_strip_len(NVMF_TRADDR_SIZE, traddr);
				fprintf(stderr,
					"traddr=%s is already connected\n",
					traddr);
			}
```

libnvme returns ENVME_CONNECT_WRITE instead.

igaw, 2022-04-14:
Though there is no change in behavior as we will ignore this errno completely.

GIven this, we need to make the libnvme logging less noisy and catch the in nvme-cli correct errno.

martin-gpy, 2022-04-27:
> If `nvme connect-all ` is executed more than once we get this error message:
> 
> ```shell
> $ .build/nvme connect-all -t tcp --traddr=10.161.8.24  --trsvcid=4421
> Failed to write to /dev/nvme-fabrics: Operation already in progress
> failed to connect controller, error 1006
> ```

The above error seen when 'nvme connect-all' is repeatedly invoked is actually consistent with the behavior in the nvme-cli-monolithic branch.

Problem now is after you have lowered the logging level with your commit at https://github.com/linux-nvme/libnvme/pull/351/commits/9282931a5b92892b4d3445c906062abf16899d7f, I don't see any errors at all now if 'nvme connect-all' is repeatedly invoked as shown below:

$ nvme connect-all -t fc -w nn-0x20000024ff19ab70:pn-0x21000024ff19ab70 -a nn-0x203400a098fd4ca6:pn-0x203500a098fd4ca6
$

Even the return value shows as 0 for the above (indicating no errors):

$ echo $?
0

And nothing specific logged in the messages file too for the above invocation:

...
kernel: [923480.309807][T30325] nvme nvme0: queue_size 128 > ctrl maxcmd 32, reducing to maxcmd
kernel: [923480.309838][T30325] nvme nvme0: NVME-FC{0}: controller connect complete
kernel: [923480.309885][T30843] nvme nvme0: NVME-FC{0}: new ctrl: NQN "nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery"
kernel: [923480.700415][T30843] nvme nvme0: Removing ctrl: NQN "nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery"
...

martin-gpy, 2022-04-27:
So lowering the log level is not at all intuitive to the end user for the above scenario since nothing specific gets logged here on the console nor in the messages file, and hence only ends up confusing the user.

PR #1550 (2022-06-03): Log connect errors
fabrics: Already connected uses a different error code

The libnvme library reports connection errors with it's own error
codes, e.g. ENVME_CONNECT_ALREADY stands for already connected. Update
the connect-all call so that it prints 'already connected' again.

Fixes: #1505
closed 2022-06-03 by igaw