linux-nvme/nvme-cli #1508: Please clarify licensing

leahneukirchen, 2022-04-18:
LICENSE says GPL2, CONTRIBUTING.md says GPL2, nvme.spec says GPL, meson.build says LGPLv2+.

nvme.c says GPL2+, nvme.h says GPL2 only...

eli-schwartz, 2022-04-18:
I think the meson.build calling it LGPL is a copy/paste bug, perhaps from libnvme which actually does use the Lesser variant of the GPL in its `COPYING` file.

The initial commit 90546431517be78d79448ac411bffa3a67df786a has the original header GPL2-only, C source GPL2+ split. In commit d7026f0bc23ad0de47eedd75639d6a3c155ccc8f the LICENSE file was first added... referring to GPL3+; after an attempt to switch to Apache it then reverted to GPL2+.

Quite confusing...

igaw, 2022-04-18:
As far I know it supposed to be GPL2-only.  @keithbusch could help to clarify this?

subnut, 2022-04-19:
Is this correct?
<table>
	<tr>
		<td>nvme-cli</td>
		<td>GPL-2.0-only</td>
	</tr>
	<tr>
		<td>libnvme</td>
		<td>LGPL-2.1-or-later</td>
	</tr>
</table>

igaw, 2022-04-19:
My understanding is that libnvme is LGPL-2.1-only.

FWIW, I plan to get the SPDX identifier added to each file, which should make it more clearer what license type is used more explicit. But this will take a while.

subnut, 2022-04-19:
> My understanding is that libnvme is LGPL-2.1-only.

I don't think so -

<samp>
<pre>$ grep -IFe LGPL -R .
./libnvme/meson.build:# SPDX-License-Identifier: LGPL-2.1-or-later
./libnvme/nvme.i:// SPDX-License-Identifier: LGPL-2.1-or-later
./pyproject.toml:    "License :: OSI Approved :: GNU Lesser General Public License v2 or later (LGPLv2+)",
./pyproject.toml:license="LGPL-2.1-or-later"
./examples/display-tree.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./examples/meson.build:# SPDX-License-Identifier: LGPL-2.1-or-later
./examples/display-columnar.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./examples/telemetry-listen.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./examples/discover-loop.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./doc/meson.build:# SPDX-License-Identifier: LGPL-2.1-or-later
./README.md:is currently licensed LGPL, see COPYING for more information.
./build/libnvme.spec:License: LGPLv2+
./test/meson.build:# SPDX-License-Identifier: LGPL-2.1-or-later
./test/register.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./test/zns.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./test/cpp.cc:// SPDX-License-Identifier: LGPL-2.1-or-later
./test/tree.py:SPDX-License-Identifier: LGPL-3.1-or-later
./test/test.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./Makefile:# SPDX-License-Identifier: LGPL-2.1-or-later
./ccan/meson.build:# SPDX-License-Identifier: LGPL-2.1-or-later
./internal/meson.build:# SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/types.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/filters.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/tree.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/ioctl.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/linux.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/fabrics.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/util.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/log.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/linux.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/tree.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/private.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/json.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/ioctl.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/filters.c:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/util.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/nvme/fabrics.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./src/meson.build:# SPDX-License-Identifier: LGPL-2.1-or-later
./src/libnvme.h:// SPDX-License-Identifier: LGPL-2.1-or-later
./meson.build:# SPDX-License-Identifier: LGPL-2.1-or-later
./meson.build:    license: 'LGPLv2+',
</pre>
</samp>

igaw, 2022-04-21:
I agree, looks more like LGPL-2.1-or-later.

igaw, 2022-05-13:
So I am using this https://github.com/spdx/spdx-license-matcher to figure out what the LICENSE/COPYING file.

```
nvme-cli: The given license text matches 94.14% with that of GPL-2.0+ based on Levenstein distance.
libnvme: The given license text matches 96.7% with that of LGPL-2.1-only based on Levenstein distance.
```

The not 100% match is due to the formatting, e.g.

```diff
The given license text matches 94.14% with that of GPL-2.0+ based on Levenstein distance.
--- 

+++ 

@@ -1,122 +1,340 @@

 GNU GENERAL PUBLIC LICENSE
-
-Version 2, June 1991
-
-Copyright (C) 1989, 1991 Free Software Foundation, Inc.
-
-51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA
-
-Everyone is permitted to copy and distribute verbatim copies of this license document, but changing it is not allowed.
-
-Preamble
-
-The licenses for most software are designed to take away your freedom to share and change it. By contrast, the GNU General Public License is intended to guarantee your freedom to share and change free software--to make sure the software is free for all its users. This General Public License applies to most of the Free Software Foundation's software and to any other program whose authors commit to using it. (Some other Free Software Foundation software is covered by the GNU Lesser General Public License instead.) You can apply it to your programs, too.
-
-When we speak of free software, we are referring to freedom, not price. Our General Public Licenses are designed to make sure that you have the freedom to distribute copies of free software (and charge for this service if you wish), that you receive source code or can get it if you want it, that you can change the software or use pieces of it in new free programs; and that you know you can do these things.
-
-To protect your rights, we need to make restrictions that forbid anyone to deny you these rights or to ask you to surrender the rights. These restrictions translate to certain responsibilities for you if you distribute copies of the software, or if you modify it.
-
-For example, if you distribute copies of such a program, whether gratis or for a fee, you must give the recipients all the rights that you have. You must make sure that they, too, receive or can get the source code. And you must show them these terms so they know their rights.
-
-We protect your rights with two steps: (1) copyright the software, and (2) offer you this license which gives you legal permission to copy, distribute and/or modify the software.
-
-Also, for each author's protection and ours, we want to make certain that everyone understands that there is no warranty for this free software. If the software is modified by someone else and passed on, we want its recipients to know that what they have is not the original, so that any problems introduced by others will not reflect on the original authors' reputations.
-
-Finally, any free program is threatened constantly by software patents. We wish to avoid the danger that redistributors of a free program will individually obtain patent licenses, in effect making the program proprietary. To prevent this, we have made it clear that any patent must be licensed for everyone's free use or not licensed at all.
-
-The precise terms and conditions for copying, distribution and modification follow.
-
-TERMS AND CONDITIONS FOR COPYING, DISTRIBUTION AND MODIFICATION
-
-   0. This License applies to any program or other work which contains a notice placed by the copyright holder saying it may be distributed under the terms of this General Public License. The "Program", below, refers to any such program or work, and a "work based on the Program" means either the Program or any derivative work under copyright law: that is to say, a work containing the Program or a portion of it, either verbatim or with modifications and/or translated into another language. (Hereinafter, translation is included without limitation in the term "modification".) Each licensee is addressed as "you".
-
-   Activities other than copying, distribution and modification are not covered by this License; they are outside its scope. The act of running the Program is not restricted, and the output from the Program is covered only if its contents constitute a work based on the Program (independent of having been made by running the Program). Whether that is true depends on what the Program does.
-
-   1. You may copy and distribute verbatim copies of the Program's source code as you receive it, in any medium, provided that you conspicuously and appropriately publish on each copy an appropriate copyright notice and disclaimer of warranty; keep intact all the notices that refer to this License and to the absence of any warranty; and give any other recipients of the Program a copy of this License along with the Program.
-
-   You may charge a fee for the physical act of transferring a copy, and you may at your option offer warranty protection in exchange for a fee.
-
-   2. You may modify your copy or copies of the Program or any portion of it, thus forming a work based on the Program, and copy and distribute such modifications or work under the terms of Section 1 above, provided that you also meet all of these conditions:
-
-      a) You must cause the modified files to carry prominent notices stating that you changed the files and the date of any change.
-
[... cut ...]
+  Yoyodyne, Inc., hereby disclaims all copyright interest in the program
+  `Gnomovision' (which makes passes at compilers) written by James Hacker.
+
+  {signature of Ty Coon}, 1 April 1989
+  Ty Coon, President of Vice
+
+This General Public License does not permit incorporating your program into
+proprietary programs.  If your program is a subroutine library, you may
+consider it more useful to permit linking proprietary applications with the
+library.  If this is what you want to do, use the GNU Lesser General
+Public License instead of this License.
+
```

igaw, 2022-05-13:
Yep, we need to get this sorted out... we need to figure out what the original indention was and fixup everything up accordingly.

keithbusch, 2022-05-13:
The intention was LGPL for libnvme, and GPL2 for nvme-cli.

eli-schwartz, 2022-05-13:
"-only" or "-or-later"?

keithbusch, 2022-05-13:
or-later

igaw, 2022-05-16:
nvme-cli status:

I tried to figure out what the difference between the GPL2-only and GPL2-or-later in LICENSE text is and how the spdx-license-matcher tool figures this out. It turns out that the license text is identically and the differentiation is via the formatting of the text and some small details in the license template. For example

```diff
--- a/LICENSE
+++ b/LICENSE
@@ -1,7 +1,7 @@
-GNU GENERAL PUBLIC LICENSE
+                    GNU GENERAL PUBLIC LICENSE
                        Version 2, June 1991

- Copyright (C) 1989, 1991 Free Software Foundation, Inc., <http://fsf.org/>
+ Copyright (C) 1989, 1991 Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA
  Everyone is permitted to copy and distribute verbatim copies
  of this license document, but changing it is not allowed.
@@ -290,8 +290,8 @@ to attach them to the start of each source file to most effectively
 convey the exclusion of warranty; and each file should have at least
 the "copyright" line and a pointer to where the full notice is found.

-    {description}
-    Copyright (C) {year}  {fullname}
+    <one line to give the program's name and a brief idea of what it does.>
+    Copyright (C) <year>  <name of author>

     This program is free software; you can redistribute it and/or modify
     it under the terms of the GNU General Public License as published by
@@ -329,7 +329,7 @@ school, if any, to sign a "copyright disclaimer" for the program, if
   Yoyodyne, Inc., hereby disclaims all copyright interest in the program
   `Gnomovision' (which makes passes at compilers) written by James Hacker.

-  {signature of Ty Coon}, 1 April 1989
+  <signature of Ty Coon>, 1 April 1989
   Ty Coon, President of Vice

 This General Public License does not permit incorporating your program into
@@ -337,4 +337,3 @@ proprietary programs.  If your program is a subroutine library, you may
 consider it more useful to permit linking proprietary applications with the
 library.  If this is what you want to do, use the GNU Lesser General
 Public License instead of this License.
-
```

will make the tool report GPL2-only. I had to search for awhile to find a GPL2-or-later
formatted version which could be trusted: https://git.yoctoproject.org/poky/plain/meta/files/common-licenses/GPL-2.0-or-later

This version is then recognized as 

  `The following license ID(s) match: GPL-2.0-only, GPL-2.0-or-later, GPL-2.0, GPL-2.0+`

The current version of the LICENSE text is recognized as GPL-2.0+. So I guess we could just leave it
as it is or copy the version from yocto to make the tools happy. I don't think there something really
broken for the LICENSE file.

We have to review all files now and see if the files include all the  'or later version' part. As far I can
say there are few files which don't have it, e.g. fabrics.c. Some files like nvme.c do have it.

eli-schwartz, 2022-05-16:
> I tried to figure out what the difference between the GPL2-only and GPL2-or-later is and how the spdx-license-matcher tool figures this out. It turns out that the license text is identically and the differentiation is via the formatting of the text and some small details in the license template.

This license tool seems a bit inefficient, since the formatting of a license should never matter, only the wording. It should ideally break it into a stream of words and check whether the wording is identical.

Either way, there's no such thing as a "GPL-2.0-or-later license text", there's just license text which is either version 2 or version 3 of the GPL (or LGPL), plus a ***License Header*** in each file stating which license it is under and optionally saying you're permitted to upgrade the COPYING / LICENSE file to a newer version at your pleasure.

igaw, 2022-05-16:
Yes, I came to same conclusion. I was just curious how the spdx-license-matcher is working and why it said GPLv2+. FWIW, the tool ignores the formatting. There are small things which makes to tool to differentiate between GPLv2-only GPLv2-or-later:

```diff
--- GPL-2.0-only.xml    2022-05-16 13:01:47.427641719 +0200
+++ GPL-2.0-or-later.xml        2022-05-16 13:01:47.427641719 +0200
@@ -1,37 +1,18 @@
 <?xml version="1.0" encoding="UTF-8"?>
 <SPDXLicenseCollection xmlns="http://www.spdx.org/license">
-  <license licenseId="GPL-2.0-only" isOsiApproved="true"
-  name="GNU General Public License v2.0 only">
+  <license licenseId="GPL-2.0-or-later" isOsiApproved="true"
+  name="GNU General Public License v2.0 or later">
     <crossRefs>
       <crossRef>https://www.gnu.org/licenses/old-licenses/gpl-2.0-standalone.html</crossRef>
       <crossRef>https://opensource.org/licenses/GPL-2.0</crossRef>
     </crossRefs>
-    <standardLicenseHeader>
-      Copyright (C) <alt name="copyright" match=".+">yyyy name of author</alt>
-      <p>
-        This program is free software; you can redistribute it and/or
-        modify it under the terms of the GNU General Public License
-        as published by the Free Software Foundation; version 2.
-      </p>
-      <p>
-        This program is distributed in the hope that it will be
-        useful, but WITHOUT ANY WARRANTY; without even the implied
-        warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR
-        PURPOSE. See the GNU General Public License for more details.
-      </p>
-      <p>
-        You should have received a copy of the GNU General Public License
-        along with this program; if not, write to the Free Software Foundation,
-        Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301<optional spacing="after">,</optional>
-        USA.
-      </p>
-    </standardLicenseHeader>
     <notes>
-      This license was released: June 1991. This license identifier refers to the choice
-      to use the code under GPL-2.0-only, as distinguished from use of code under GPL-2.0-or-later
-      (i.e., GPL-2.0 or some later version). The license notice (as seen in the Standard License
-      Header field below)  states which of these applies to the code in the file. The example
-      in the exhibit to the license shows the license notice for the "or later" approach.
+      This license was released: June 1991. This license identifier refers to the
+      choice to use code under GPL-2.0-or-later (i.e., GPL-2.0 or some later version),
+      as distinguished from use of code under GPL-2.0-only. The license notice (as seen
+      in the Standard License Header field below)  states which of these applies
+      the code in the file. The example in the exhibit to the license shows the license
+      notice for the "or later" approach.
     </notes>
     <text>
     <titleText>
@@ -403,12 +384,10 @@
         convey the exclusion of warranty; and each file should have at least
         the "copyright" line and a pointer to where the full notice is found.
       </p>
+      <standardLicenseHeader>
       <p>
-        <optional spacing="none">&lt;</optional>one line to give the program's name and <alt name="ideaArticle" match="a brief|an">an</alt> idea of what it does.<optional spacing="none">&gt;</optional>
-        <br></br>
-        Copyright (C)
-        <optional spacing="none">&lt;</optional><alt name="templateYear" match="yyyy|year">yyyy</alt><optional spacing="after">&gt;</optional>
-        <optional spacing="before">&lt;</optional>name of author<optional spacing="none">&gt;</optional>
+        <alt name="description" match=".+">&lt;one line to give the program's name and an idea of what it does.&gt;</alt><br/>
+        Copyright (C) <alt name="copyright" match=".+">&lt;yyyy&gt; &lt;name of author&gt;</alt>
       </p>
       <p>
         This program is free software; you can redistribute it and/or
@@ -428,6 +407,7 @@
         Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301<optional spacing="after">,</optional>
         USA.
       </p>
+      </standardLicenseHeader>
       <p>
         Also add information on how to
         contact you by electronic and paper mail.
@@ -458,7 +438,7 @@
         `Gnomovision' (which makes passes at compilers) written by James Hacker.
       </p>
       <p>
-       <optional spacing="none">&lt;</optional>signature of Ty Coon<optional spacing="before">&gt;</optional>,
+       <optional spacing="none">&lt;</optional>signature of Ty Coon<optional spacing="none">&gt;</optional>,
        1 April 1989 Ty Coon, President of Vice
       </p>
     </optional>
```
see  https://github.com/spdx/license-list-XML.git

closed 2022-05-18 by igaw

igaw, 2022-05-18:
> Is this correct?
> nvme-cli 	GPL-2.0-only
> libnvme 	LGPL-2.1-or-later

Yes. I reviewed the projects and added the missing SPDX identifiers. For libnvme 
there were no big surprises.

Though for nvme-cli it turns out fabrics.c is currently licensed as GPL-2.0-only which 
tags the whole project as GPL-2.0-only. Almost all other files are licensed under
 GPL-2.0-or-later. So if anyone wants to have nvme-cli available under GPL-2.0-or-later needs to get the fabrics.c license sorted out. 

I didn't get a lot of response on my request to relax the license: https://lore.kernel.org/linux-nvme/20220516161608.of4tklsysuvgbtie@carbon.lan/

PR #1526 (2022-05-18): treewide: Add SPDX identifier
Add the default project wide license SPDX identifier GPL-2.0-or-later
to all files which do not have any copyright statement.

Add the GPL-2.0-or-later or GPL-2.0-only according the copyright
statement listed in the file.

Fixes: #1508 

After adding to most of the files a SPDX-License-Identifier it's possible
to identify where we have not GPL-2.0-or-later files. Most of the newer 
files are from my wrong request to the authors to provide these files
under GPL-2.0-only.

`LGPL-2.1-or-later`, compatible with GPL-2.0-or-later:
- Makefile
- meson-vcs-tag.sh
- tests/r [... cut]