linux-nvme/nvme-cli #1516: NVMe List fails with "Failed to scan topoplogy: No such file or directory" error.

jeffreyalien, 2022-04-29:
I'm getting error on a system running the Ubuntu 4.4 kernel.  The list command works fine on the old version of nvme cli from the monolithic branch.  I believe the error is happening because the there's nothing in the /sys/class/nvme-subsystem dir.  In the older version of code, there's a function called legacy_list that is setup to handle case for "For pre-subsystem enabled kernels".

My question is what's considered a pre subsystem enable kernel?  Does the Ubuntu the 4.4 kernel version fall into that category?  And should the new (nvme cli 2.0 version) be able to handle the older kernels?   Or is version 4.4 just not supported with nvme cli 2.0?

igaw, 2022-05-13:
`/sys/class/nvme-subsystem` was introduce with v4.15. As nvme-cli 2.x relies on this sysfs
structure, I'd say the minimum kernel version is the next stable version, which is v4.19. I am not 
so keen to add the pre-subsystem stuff from nvme-cli 1.x into nvme-cli 2.x. I'd rather keep nvme-cli 1.x maintained a bit longer for this.

igaw, 2022-05-18:
So should just add a note to libnvme/nvme-cli?

jeffreyalien, 2022-05-18:
I'm fine with doing that.  I mainly wanted to know where the cutoff was - when the nvme-subsystem was added.

hreinecke, 2022-06-10:
Yes, we should be adding a warning about older kernels. We still have the monolithic branch which should be used there.

PR #1566 (2022-06-13): README: add note about min kernel version
The min kernel version is v4.15 kernel because libnvme depends on the
/sys/class/nvme-subsystem interface.

Fixes #1516
closed 2022-06-13 by igaw