linux-nvme/nvme-cli #1519: NVMe List command not listing the NVMe devices on Ubuntu 20.04

harshitap26, 2022-05-12:
The NVMe list command is not listing the NVMe devices that are mapped to our host. We can see the NVMe devices created on our array and also see the host mapping done, but it is not getting reflected on the array. 
The NVMe connection is properly done and it list the nvme controllers as well.

We also tried using the older version of NVMe CLI, but that doesn't work as well. Facing this issue with our Ubuntu Setups

igaw, 2022-05-17:
Can you provide any logs, e.g. kernel logs and also the output from 'nvme list-subsys -vvvvv' with the current 2.0 version? 

Maybe also  `tar -c -J -f /tmp/sysfs-nvme.tar.xz  /sys/devices/virtual/nvme-fabrics/ctl /sys/devices/virtual/nvme-subsystem /sys/class/nvme /sys/class/nvme-subsystem`

igaw, 2022-05-18:
`nvme list-subsys` doesn't require a device name, it's optional. 

I don't  understand what you are doing, e.g. I fail to understand what you mean with 'if we create the duplicate connection'. Could you share some logs and the commands you are issuing?

igaw, 2022-07-08:
No feedback. Closing bug report.

harshitap26, 2022-07-08:
Sorry missed the previous comment. So we do the below steps -

1. We make an nvme connection from our host (Ubuntu VM) to our storage array. The command `nvme list-subsys` works fine and shows all the connections.
2. We create an NVMe volume on our storage array. On the array we can see that the volume is mapped on our ubuntu host.

Expected Results - 
On the ubuntu host the command `nvme list` should list the nvme volume that we created on the storage array

Actual Results -
The `nvme list` command prints nothing.

This scenario is only observed on our Ubuntu Setups. It works fine on our other like RHEL.

harshitap26, 2022-07-08:
Also by duplicate connection we mean the below scenario
1. We make an nvme connection from our host (Ubuntu VM) to our storage array (Controller). The command nvme list-subsys works fine and shows all the connections.
2. We create an NVMe volume on our storage array. On the array we can see that the volume is mapped on our ubuntu host.
3. On the ubuntu host the command `nvme list' is not listing the nvme volume that we created on the storage array
4. If we again make a duplicate NVMe connection between the ubuntu host and the array (Controller), then the `nvme list` works fine again.

igaw, 2022-07-08:
No worries. Happens.

Could you post kernel logs (if there are any) and the output from `nvme list-subsys -vvvv`?

closed 2022-09-30 by igaw

MatiasBjorling, 2022-10-11:
Did you try with sudo?

ganou2007, 2024-07-02:
I have a system on Oracle Linux 9.X, When commanded "nvme list", the output is blank as below:
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------

But if the same NVMe SSD installed in a Ubuntu, the output was full of content for each item. 

What would be missed ?
Thanks.

ganou2007, 2024-07-02:
[image]

igaw, 2024-07-03:
First, please do not hijack old threads. Second, we can't support downstream versions nvme-cli. This means if you don't run a fairly updated version of nvme-cli, it's hard to help. You didn't provide any information on the kernel version nor nvme-cli version. Furthermore, I asked for debug logs in this report, so it's clear in order to help we, logs are necessary. But again, don't hijack this thread.

ganou2007, 2024-07-03:
What did you mean "hijack old threads"? Does it require to create a new question?