linux-nvme/nvme-cli #1520: nvme connect -J option is broken

martin-gpy, 2022-05-13:
The nvme connect -J option for reading the config JSON file and applying the respective values during connect is broken. It currently ignores the values specified in the JSON file for the connect, and continues to rely on the default values itself for the connect. For e.g.

$ cat /etc/nvme/config.json
[
  {
    "hostnqn":"nqn.2014-08.org.nvmexpress:uuid:12372496-59c4-4d1b-be09-74362c0c1afc",
    "hostid":"59de25be-7383-48f0-8a79-df4bce9573f3",
    "subsystems":[
      {
        "nqn":"nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116",
        "ports":[
          {
            "trsvcid":"4420",
            "keep_alive_tmo": "45",
            "hdr_digest": true
          }
        ]
      }
    ]
  }
]

Now running nvme connect specifying the above config JSON file:

$ nvme connect -n nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116 -t tcp -a 192.168.1.116 -w 192.168.1.16 -J /etc/nvme/config.json

The connect is successful:

$ nvme list-subsys /dev/nvme0n1
nvme-subsys0 - NQN=nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116
\
 +- nvme0 tcp traddr=192.168.1.116,trsvcid=4420,host_traddr=192.168.1.16 live optimized

But nvme0 still retains the default KATO value of 5s itself (and hdr_digest remains false too):

$ cat /sys/class/nvme-subsystem/nvme-subsys0/nvme0/kato
5

Ideally, the nvme connect should have picked up the above KATO value of 45s as well as enabled hdr_digest for the above nvme connect, as specified in the config JSON file, and not the default values.

igaw, 2022-05-18:
One thing which seems necessary to have in the "ports" dictionary is the transport. Without it the config will not be considered at all.

With `"transport": "tcp"`: It all looks like the config settings are applied to a different `nvme_ctrl_t` when we read the config file compared to the one then actually used `nvmf_add_ctrl()`. And thus `merge_config()` function does not apply the kato value. 

`json_parse_port()` looks up  via `nvme_lookup_ctrl()` a  `nvme_fabrics_config` but it's not the one `nvmf_connect_disc_entry()` uses in `nvmf_add_ctrl()`.

martin-gpy, 2022-05-23:
I included the 3 commits described in https://github.com/linux-nvme/libnvme/pull/363, but that still didn't help apply the values from the config JSON during the nvme connect:

$ cat /etc/nvme/config.json
[
  {
    "hostnqn":"nqn.2014-08.org.nvmexpress:uuid:12372496-59c4-4d1b-be09-74362c0c1afc",
    "hostid":"59de25be-7383-48f0-8a79-df4bce9573f3",
    "subsystems":[
      {
        "nqn":"nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116",
        "ports":[
          {
            "trsvcid":"4420",
            "keep_alive_tmo": "60",
            "hdr_digest": true
          }
        ]
      }
    ]
  }
]

$ nvme connect -t tcp -w 192.168.1.16 -a 192.168.1.116 -n nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116 -J /etc/nvme/config.json

$ nvme list-subsys /dev/nvme0n1
nvme-subsys0 - NQN=nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116
\
 +- nvme0 tcp traddr=192.168.1.116,trsvcid=4420,host_traddr=192.168.1.16 live optimized

$ cat /sys/class/nvme-subsystem/nvme-subsys0/nvme0/kato
5

So it looks like it is still taking the default values itself here.  Any pointers on what could be wrong?

igaw, 2022-05-23:
Please add `"transport": "tcp"` to your JSON file. It's mandatory.

martin-gpy, 2022-05-23:
> Please add `"transport": "tcp"` to your JSON file. It's mandatory.

But just adding "transport doesn't seem enough:

$ nvme connect -t tcp -w 192.168.1.16 -a 192.168.1.116 -n nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116 -J /etc/nvme/config.json
No transport address for 'tcp'

So looks like one needs to add the traddr & host-traddr as well here to the config JSON file. But again, I don't find that logical at all because one has already provided the transport type, traddr & host-traddr (along with the nqn) in the command line while invoking nvme connect - otherwise the nvme connect command doesn't even run. So why can't it just apply those params from the command line itself?

martin-gpy, 2022-05-23:
Also btw, the above config JSON file was generated using the -O dump option. So if these params are mandatory in the config JSON file, the -O dump option should have ensured these mandatory params were included while generating the config JSON file.

igaw, 2022-05-23:
> So looks like one needs to add the traddr & host-traddr as well here to the config JSON file. But again, I don't find that logical at all because one has already provided the transport type, traddr [...]

Because when reading the JSON file, the command line options are not considered. The merging of the config option happens after the parsing of the file.

I suppose the `-O dump` should include these settings. Doesn't really make sense if the parser is not able to read self generated config files.

martin-gpy, 2022-05-23:
I'm afraid these changes still don't work for me:

$ cat /etc/nvme/config.json
[
  {
    "hostnqn":"nqn.2014-08.org.nvmexpress:uuid:12372496-59c4-4d1b-be09-74362c0c1afc",
    "subsystems":[
      {
        "nqn":"nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116",
        "ports":[
          {
            "transport": "tcp",
            "traddr": "192.168.1.116",
            "host_traddr": "192.168.1.16",
            "trsvcid": "4420",
            "keep_alive_tmo": "60",
            "hdr_digest": true
          }
        ]
      }
    ]
  }
]

$ nvme connect -t tcp -w 192.168.1.16 -a 192.168.1.116 -n nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116 -J /etc/nvme/config.json

$ nvme list-subsys /dev/nvme0n1
nvme-subsys0 - NQN=nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116
\
 +- nvme0 tcp traddr=192.168.1.116,trsvcid=4420,host_traddr=192.168.1.16 live optimized

$ cat /sys/class/nvme-subsystem/nvme-subsys0/nvme0/kato
5

igaw, 2022-05-24:
I am fine with keeping just this issue for the moment. We could change the title to something more generic, e.g. import/export of configuration broken

igaw, 2022-05-24:
Here the output of `nvme connect-all -O dump > config.json` and I added the `keep_alive_tmo` part:

```json
[
  {
    "hostnqn":"nqn.2014-08.org.nvmexpress:uuid:1a9e23dd-466e-45ca-9f43-a29aaf47cb21",
    "hostid":"1a9e23dd-466e-45ca-9f43-a29aaf47cb21",
    "subsystems":[
      {
        "nqn":"nqn.2014-08.org.nvmexpress:NVMf:uuid:7a7aa300-7545-11ec-93c5-40a8f0343914",
        "ports":[
          {
            "transport":"tcp",
            "traddr":"10.161.8.24",
            "trsvcid":"4423",
            "disable_sqflow":true,
            "keep_alive_tmo":"45"
          },
          {
            "transport":"tcp",
            "traddr":"10.161.8.24",
            "trsvcid":"4422",
            "disable_sqflow":true
          }
        ]
      },
      {
        "nqn":"nqn.2014-08.org.nvmexpress:NVMf:uuid:de63429f-50a4-4e03-ade6-0be27b75be77",
        "ports":[
          {
            "transport":"tcp",
            "traddr":"10.161.8.24",
            "trsvcid":"4421",
            "disable_sqflow":true
          },
          {
            "transport":"tcp",
            "traddr":"10.161.8.24",
            "trsvcid":"4420",
            "disable_sqflow":true
          }
        ]
      }
    ]
  }
]
```

and here the connect attempt. Note the additional  `keep_alive_tmo=45` argument given to the kernel:

```bash
# ./nvme connect-all -J ../nvme-anantara.json -vvvv
connect ctrl, 'nqn=nqn.2014-08.org.nvmexpress.discovery,transport=tcp,traddr=10.161.8.24,trsvcid=4420,hostnqn=nqn.2014-08.org.nvmexpress:uuid:1a9e23dd-466e-45ca-9f43-a29aaf47cb21,hostid=1a9e23dd-466e-45ca-9f43-a29aaf47cb21,ctrl_loss_tmo=600'
connect ctrl, response 'instance=0,cntlid=1'
nvme0: ctrl connected
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme0
nvme0: discover length 256
nvme0: discover length 2048
nvme0: discover genctr 6, retry
lookup ctrl (transport: tcp, traddr: 10.161.8.24, trsvcid 4420)
connect ctrl, 'nqn=nqn.2014-08.org.nvmexpress:NVMf:uuid:de63429f-50a4-4e03-ade6-0be27b75be77,transport=tcp,traddr=10.161.8.24,trsvcid=4420,hostnqn=nqn.2014-08.org.nvmexpress:uuid:1a9e23dd-466e-45ca-9f43-a29aaf47cb21,hostid=1a9e23dd-466e-45ca-9f43-a29aaf47cb21,ctrl_loss_tmo=600,disable_sqflow'
connect ctrl, response 'instance=1,cntlid=2'
nvme1: ctrl connected
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme1
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme1
nvme0: disconnected
connect ctrl, 'nqn=nqn.2014-08.org.nvmexpress.discovery,transport=tcp,traddr=10.161.8.24,trsvcid=4421,hostnqn=nqn.2014-08.org.nvmexpress:uuid:1a9e23dd-466e-45ca-9f43-a29aaf47cb21,hostid=1a9e23dd-466e-45ca-9f43-a29aaf47cb21,ctrl_loss_tmo=600'
connect ctrl, response 'instance=0,cntlid=3'
nvme0: ctrl connected
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme0
[... 38 lines cut ...]
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys3/nvme4
nvme0: disconnected
```

and the double checks says all good

```bash
# cat /sys/class/nvme-subsystem/nvme-subsys3/nvme3/kato 
5
# cat /sys/class/nvme-subsystem/nvme-subsys3/nvme4/kato 
45
```

The discover.conf has only these entries here:

```
--transport=tcp --traddr=10.161.8.24 --trsvcid=4420
--transport=tcp --traddr=10.161.8.24 --trsvcid=4421
--transport=tcp --traddr=10.161.8.24 --trsvcid=4422
--transport=tcp --traddr=10.161.8.24 --trsvcid=4423
```

So two things I noticed compared to your config file. First there is the `hostid` in my output and the `transport` is there as well.

Can you make sure that you run really the latest version of nvme-cli and libnvme with the mentioned patches?

martin-gpy, 2022-05-24:
Agree. I updated to the latest libnvme & nvme-cli versions and applied your commits. And it now works fine for me.

$ cat /etc/nvme/config.json
[
  {
    "hostnqn":"nqn.2014-08.org.nvmexpress:uuid:12372496-59c4-4d1b-be09-74362c0c1afc",
    "subsystems":[
      {
        "nqn":"nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116",
        "ports":[
          {
            "transport": "tcp",
            "traddr": "192.168.1.116",
            "host_traddr": "192.168.1.16",
            "trsvcid": "4420",
            "keep_alive_tmo": "60",
            "hdr_digest": true
          }
        ]
      }
    ]
  }
]

$ nvme connect -t tcp -w 192.168.1.16 -a 192.168.1.116 -n nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116 -J /etc/nvme/config.json

$ nvme list-subsys /dev/nvme0n1
nvme-subsys0 - NQN=nqn.1992-08.com.netapp:sn.48391d66c0a611ecaaa5d039ea165514:subsystem.subsys_CLIENT116
\
 +- nvme0 tcp traddr=192.168.1.116,trsvcid=4420,host_traddr=192.168.1.16 live optimized

$ cat /sys/class/nvme-subsystem/nvme-subsys0/nvme0/kato
60

So this looks good. Thanks.

igaw, 2022-05-24:
Okay, so let's go with the fix for libnvme.

closed 2022-05-24 by igaw