linux-nvme/nvme-cli #1533: Can the nvme wdc namespace-resize command be used only for the data center product line of Western Digital?

daiaji, 2022-05-22:
https://community.wd.com/t/wd-blue-sn550-purge-sanitize-namespace-resize-poor-write-performance
This link reports that this command is not available for consumer SSD SN550.
https://www.thomas-krenn.com/de/wiki/Drive-Resize_von_Western_Digital_NVME_SSDs
This link reports that the data center product line SN640 supports this command.
It looks like consumer grade SSDs don't support namespace tweaking at all.
Maybe I should just forgo buying Western Digital's consumer-grade SSD?

igaw, 2022-05-23:
Maybe @jeffreyalien can answer this question.

jeffreyalien, 2022-05-23:
@daiaji In general, the WDC plugin commands are written for the data center product line.  The drive-resize command is supported by the following drives:  SN630, SN640, SN840, SN540, and ZN540.  These drives are all part of the data center product line; none of our consumer grade drives support the drive-resize command.  Does that answer your question.

daiaji, 2022-05-24:
@jeffreyalien  In fact, I want to buy a consumer-grade SSD that is large enough and create two namespaces for placing it in the mobile hard disk box for full-disk backup of the PC, but if the consumer-grade SSD does not support namespace resizing, then  Western Digital's SSD is the same as other SSDs, and there is nothing that appeals to me.

closed 2022-05-24 by daiaji