linux-nvme/nvme-cli #1542: 'nvme list-subsys' displays ANA state too

martin-gpy, 2022-05-27:
It seems the 'nvme list-subsys' command output from the latest nvme-cli v2.x displays the ANA state too, unlike the previous nvme-cli-monolithic version. Basically the ANA state was displayed only when 'nvme list-subsys /dev/nvmeXnY' was run, and not just 'nvme list-subsys'. For e.g.

**nvme-cli-monolithic**:
$ nvme list-subsys
nvme-subsys0 - NQN=nqn.1992-08.com.netapp:sn.1a3ca52d792611ec802600a098fd4ca6:subsystem.subsys_145
\
 +- nvme0 fc traddr=nn-0x202400a098fd4ca6:pn-0x202500a098fd4ca6 host_traddr=nn-0x200000109b579d5e:pn-0x100000109b579d5e live
nvme-subsys1 - NQN=nqn.1992-08.com.netapp:sn.1a3ca52d792611ec802600a098fd4ca6:subsystem.subsys_145_test
\
 +- nvme1 fc traddr=nn-0x202400a098fd4ca6:pn-0x202500a098fd4ca6 host_traddr=nn-0x200000109b579d5e:pn-0x100000109b579d5e live

**Current nvme-cli v2.x**:
$ nvme list-subsys
nvme-subsys28 - NQN=nqn.1992-08.com.netapp:sn.b81bb6bdc1d911ec81a400a098fd4ca6:subsystem.nvme_chat-16-117_3
\
 +- nvme28 fc traddr=nn-0x204800a098fd4ca6:pn-0x204a00a098fd4ca6,host_traddr=nn-0x20000090fa94298d:pn-0x10000090fa94298d live **_optimized_**
nvme-subsys12 - NQN=nqn.1992-08.com.netapp:sn.b81bb6bdc1d911ec81a400a098fd4ca6:subsystem.nvme_chat-16-117_1
\
 +- nvme12 fc traddr=nn-0x204800a098fd4ca6:pn-0x204a00a098fd4ca6,host_traddr=nn-0x20000090fa94298d:pn-0x10000090fa94298d live **_optimized_**

So is this intentional or a bug?

igaw, 2022-05-27:
Not a bug, intentional feature.

martin-gpy, 2022-05-30:
> Not a bug, intentional feature.

But the point is that the ANA state for a controller may be optimized for one ns in a subsystem, and non-optimized for another ns in the same subsystem i.e. the same controller can have different ANA states for different ns in the same subsystem. 

So it doesn't make sense to display the ANA state in the 'nvme list-subsys' output, and should only be displayed in the 'nvme list-subsys /dev/nvmeXnY' output.

martin-gpy, 2022-05-30:
So I think this remains a bug. Could you please reopen this issue?

igaw, 2022-05-31:
Sure thing, after our offline discussion, the best strategy here is to make `list-subsys` behave as we have in the monolithic version and introduce a new command which shows all the states in one go (#1544)

PR #1549 (2022-06-03): nvme-print: Show ANA state only for one namespace
'nvme list-subsys' shows the state of all controllers belonging to a
subsystem. The ANA state is a per namespace attribute hence it only
makes sense to show it if the user lists the subsystem for a
namespace.

Fixes: 7435ed9ae6a6 ("nvme-print: Show paths from the first namespace only")

Fixes: #1542
closed 2022-06-03 by igaw