linux-nvme/nvme-cli #1555: "nvme discover" is failing with nvme cli 2.0 in RHEL 8.5

sarabjotsoit, 2022-06-06:
Steps:
1. Install nvme cli 2.0.
2. Load the modules and discover the drives.

**Observation:**
nvme discover is failing with nvme cli 2.0
with NVME cli 2.0
[root ~]# nvme discover -t rdma -a 192.168.0.24 -q host
**failed to get discovery log: Invalid slot**
[root ~]# nvme version
nvme version 2.0

sarabjotsoit, 2022-06-06:
The same is working with nvme-cli 1.13.  We have tested with RHEL 8.5 and Ubuntu 22.04.

jeffreyalien, 2022-06-07:
I see a few mask values used for the nvme_get_log function are incorrect:

NVME_LOG_CDW10_LSP_MASK = 0xf - and should be 0x7f
NVME_LOG_CDW11_LSI_MASK = 0xff  - and should be 0xffff

I don't think they are a factor in this case though since these fields are not used when reading the discovery log.

sarabjotsoit, 2022-06-10:
Hi, igaw

Please find the o/p with verbosity:

[root@dhcp-10-206-137-70 ~]# nvme discover -t rdma -a 192.168.1.25 -q host -i 16 -s 4420 -vvv
scan controller nvme0
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme0
scan controller nvme0 path nvme0c0n1
scan controller nvme1
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme1
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme1
scan controller nvme1 path nvme1c1n1
scan controller nvme10
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme10
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme10
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme10
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys2/nvme10
scan controller nvme10 path nvme2c10n1
scan controller nvme11
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme11
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme11
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme11
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys2/nvme11
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys3/nvme11
scan controller nvme11 path nvme3c11n1
scan controller nvme12
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme12
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme12
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme12
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys2/nvme12
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys3/nvme12
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys4/nvme12
scan controller nvme12 path nvme4c12n1
scan controller nvme13
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme13
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme13
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme13
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys2/nvme13
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys3/nvme13
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys4/nvme13
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys5/nvme13
scan controller nvme13 path nvme5c13n1
scan controller nvme14
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme14
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme14
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme14
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys2/nvme14
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys3/nvme14
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys4/nvme14
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys5/nvme14
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys6/nvme14
scan controller nvme14 path nvme6c14n1
scan controller nvme15
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme15
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme15
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme15
scan controller nvme15 namespace nvme15n1
scan controller nvme2
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme2
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme2
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme2
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys2/nvme2
scan controller nvme2 path nvme2c2n1
scan controller nvme3
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme3
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme3
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme3
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys2/nvme3
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys3/nvme3
scan controller nvme3 path nvme3c3n1
[... 50 lines cut ...]
scan subsystem nvme-subsys2
scan subsystem nvme-subsys2 namespace nvme2n1
scan subsystem nvme-subsys3
scan subsystem nvme-subsys3 namespace nvme3n1
scan subsystem nvme-subsys4
scan subsystem nvme-subsys4 namespace nvme4n1
scan subsystem nvme-subsys5
scan subsystem nvme-subsys5 namespace nvme5n1
scan subsystem nvme-subsys6
scan subsystem nvme-subsys6 namespace nvme6n1
scan subsystem nvme-subsys7
connect ctrl, 'nqn=nqn.2014-08.org.nvmexpress.discovery,transport=rdma,traddr=192.168.1.25,trsvcid=4420,hostnqn=host,hostid=76e66935-8e8b-4efb-abad-111678e786ed,ctrl_loss_tmo=600'
connect ctrl, response 'instance=16,cntlid=32777'
nvme16: ctrl connected
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme16
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme16
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme16
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys16/nvme16
nvme16: discover length 256
nvme16: discover length 9216
nvme16: discover genctr 0, retry
nvme16: could only fetch 8 of 1 records
failed to get discovery log: Invalid slot
nvme16: disconnected
[root@dhcp-10-206-137-70 ~]#

sarabjotsoit, 2022-06-10:
Hi, Team

Can we get a early RC on top of nvme-cli 2.0 code. This is currently blocking us to upgrade the interop.

Thank you in advance.

Thanks... Sarabjot

hreinecke, 2022-06-10:
Ah. Problem is in libnvme/src/nvme/fabrics.c, where we issue a 'get log page' for the entire discovery log, but then issue a _second_ 'get log page' to just get the header of the discovery log page to figure out if the generation counter changed.
And that second 'get log page' is overwriting the 'numrec' value, causing the code to assume that only 1 entry is present.
Hence this strange message 'could only fetch 8 of 1 records'.

hreinecke, 2022-06-10:
@sarabjotsoit , can you please test with the above patch?

sarabjotsoit, 2022-06-10:
Hi, @hreinecke 

Thank you for the patch. I have tried patching the code with the new changes, but failed to complile it.

{noformat}
ccan           common.h         COPYING       Documentation  fabrics.c  libnvme          LICENSE      meson_options.txt  nvme.c           nvme-models.c  nvme-print.h  nvmf-autoconnect  plugins         regress  subprojects  update-docs.sh
cmd.h          completions      define_cmd.h  etc            fabrics.h  libnvme.pc.in    Makefile     meson-vcs-tag.sh   nvme.control.in  nvme-models.h  nvme-rpmb.c   plugin.c          pyproject.toml  scripts  test         util
cmd_handler.h  CONTRIBUTING.md  doc           examples       internal   libnvme.spec.in  meson.build  nvme-builtin.h     nvme.h           nvme-print.c   nvme.spec.in  plugin.h          README.md       src      tests
[root@dhcp-10-206-137-70 nvme-cli-2.0]# make clean
ninja -C .build -t clean
Cleaning... 81 files.
[root@dhcp-10-206-137-70 nvme-cli-2.0]# make
ninja -C .build
ninja: Entering directory `.build'
[0/1] Regenerating build files.
The Meson build system
Version: 0.61.4
Source dir: /root/NVME/nvme-cli-2.0
Build dir: /root/NVME/nvme-cli-2.0/.build
Build type: native build
Project name: libnvme
Project version: 1.0
C compiler for the host machine: cc (gcc 8.5.0 "cc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-4)")
C linker for the host machine: cc ld.bfd 2.30-108
Host machine cpu family: x86_64
Host machine cpu: x86_64
C++ compiler for the host machine: c++ (gcc 8.5.0 "c++ (GCC) 8.5.0 20210514 (Red Hat 8.5.0-4)")
C++ linker for the host machine: c++ ld.bfd 2.30-108
Dependency uuid found: YES 2.32.1 (cached)
Found pkg-config: /usr/bin/pkg-config (1.4.2)
Run-time dependency json-c found: YES 0.15
Dependency openssl found: YES 1.1.1k (cached)
Checking if "__builtin_type_compatible_p" compiles: YES (cached)
Checking if "typeof" compiles: YES (cached)
Checking if "byteswap.h" compiles: YES (cached)
Checking if "bswap64" : links: YES (cached)
Checking if "statement-expr" compiles: YES (cached)
Checking if "isblank" : links: YES (cached)
Configuring libnvme.spec using configuration
Configuring config.h using configuration
Program python3 found: YES (/usr/bin/python3)
Program swig found: NO
Has header "Python.h" with dependency python: NO (cached)
Configuring conf.py using configuration
Configuring api.rst using configuration
Configuring index.rst using configuration
Configuring quickstart.rst using configuration
Configuring installation.rst using configuration
Configuring config-schema.json using configuration
Build targets in project: 10

libnvme 1.0

  User defined options
    backend: ninja

Found ninja-1.10.2.git.kitware.jobserver-1 at /usr/local/bin/ninja
Cleaning... 0 files.
NOTICE: You are using Python 3.6 which is EOL. Starting with v0.62.0, Meson will require Python 3.7 or newer
[1/31] Compiling C object src/libnvme.so.1.0.0.p/nvme_json.c.o
FAILED: src/libnvme.so.1.0.0.p/nvme_json.c.o 
cc -Isrc/libnvme.so.1.0.0.p -Isrc -I../src -I. -I.. -Iccan -I../ccan -Iinternal -I../internal -I/usr/include/uuid -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O3 -fomit-frame-pointer -D_GNU_SOURCE -include internal/config.h -fPIC -MD -MQ src/libnvme.so.1.0.0.p/nvme_json.c.o -MF src/libnvme.so.1.0.0.p/nvme_json.c.o.d -o src/libnvme.so.1.0.0.p/nvme_json.c.o -c ../src/nvme/json.c
../src/nvme/json.c:15:10: fatal error: json.h: No such file or directory
 #include <json.h>
          ^~~~~~~~
compilation terminated.
[20/31] Compiling C object src/libnvme.so.1.0.0.p/nvme_tree.c.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:19: libnvme] Error 1

{noformat}

Is it possible to get a complete file, including the nvme files.?

hreinecke, 2022-06-10:
You are missing the 'libjson-c' library.

sarabjotsoit, 2022-06-10:
Hi, @hreinecke 

Can you please provide that?

hreinecke, 2022-06-10:
Please check with your distro on how to install it.

sarabjotsoit, 2022-06-10:
Hi, @hreinecke 

I am using RHEL 8.5. Checked and found that the json-c is already installed 

**Package json-c-0.13.1-2.el8.x86_64 is already installed.
Package json-glib-1.4.4-1.el8.x86_64 is already installed.**
Dependencies resolved.
Nothing to do.
Complete!
[root@dhcp-10-206-137-70 nvme-cli-2.0]# make
ninja -C .build
ninja: Entering directory `.build'
[1/11] Compiling C object src/libnvme.so.1.0.0.p/nvme_json.c.o
FAILED: src/libnvme.so.1.0.0.p/nvme_json.c.o 
cc -Isrc/libnvme.so.1.0.0.p -Isrc -I../src -I. -I.. -Iccan -I../ccan -Iinternal -I../internal -I/usr/include/uuid -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O3 -fomit-frame-pointer -D_GNU_SOURCE -include internal/config.h -fPIC -MD -MQ src/libnvme.so.1.0.0.p/nvme_json.c.o -MF src/libnvme.so.1.0.0.p/nvme_json.c.o.d -o src/libnvme.so.1.0.0.p/nvme_json.c.o -c ../src/nvme/json.c
../src/nvme/json.c:15:10: fatal error: json.h: No such file or directory
 #include <json.h>
          ^~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
make: *** [Makefile:19: libnvme] Error 1

jeffreyalien, 2022-06-10:
@sarabjotsoit I think you need to install json-c-devel lib to fix this error.  At least that fixed the same failure for me recently.  
yum install json-c-devel

igaw, 2022-06-13:
```
nvme16: discover length 256
nvme16: discover length 9216
nvme16: could only fetch 8 of 4 records
failed to get discovery log: Invalid slot
```

@hreinecke's fix is not updating `numrec` alongside `numrec` hence the `8 of 4` error.

igaw, 2022-06-13:
@sarabjotsoit could give https://github.com/linux-nvme/libnvme/pull/378 a test? I think this should address all the issues you see.

igaw, 2022-07-07:
I've merged https://github.com/linux-nvme/libnvme/pull/378. Closing this bug report.

closed 2022-07-07 by igaw