linux-nvme/nvme-cli #1558: Get SMART attributes by id

b10s, 2022-06-07:
Hi,

I'm trying to get SMART attributes 247 and 248 on my `Samsung SSD 970 EVO Plus 1TB` using nvme ctl but can't find proper way.

How to get SMART attributes using nvme ctl (or maybe some other way)?

reference:
https://media-www.micron.com/-/media/client/global/documents/products/technical-note/solid-state-storage/tnfd23_m500_smart_attributes_calc_waf.pdf

igaw, 2022-06-09:
I don't really see where these attributes are coming from. The spec doesn't know about these. Maybe @hanumanthuh might know it.

igaw, 2022-07-08:
[TN-FD-21 ](https://media-www.micron.com/-/media/client/global/documents/products/technical-note/solid-state-storage/tnfd21_m500-mu02_smart_attributes.pdf?rev=d5b5514d6bba4e9dbaf44a7b5d28ca46) explains how to retrieve the data. 

```diff
--- a/plugins/micron/micron-nvme.c
+++ b/plugins/micron/micron-nvme.c
@@ -3355,6 +3355,8 @@ static int micron_logpage_dir(int argc, char **argv, struct command *cmd,
         {0xC0, 0, "SMART Cloud Health Log"},
         {0xC2, 0, "Firmware Activation History"},
         {0xC3, 0, "Latency Monitor Log"},
+        {0xF7, 0, "Host program page count"},
+        {0xF8, 0, "Background program page count"}
     };
 
     printf("Supported log page list\nLog ID : Description\n");
```

In theory `nvme micron log-page-directory /dev/nvme0n1` should at least list the two entries. Obviously, this is just a starting point. Someone needs to sit down and add the missing pieces. I don't have the hardware to test it.

ikegami-t, 2022-07-24:
It seems that it is difficult to get the SMART attributes on the drive mentioned since probably OCP specification is not supported by the drive.
**Note**:  It can be gottne the SMART attributes by the vendor specific command also if there but not sure actually.

ID: 247 (0xF7): Host program page count seems to be gotten by the get log page command smart / health information log data unit written below.
https://nvmexpress.org/wp-content/uploads/NVM-Express-Base-Specification-2.0b-2021.12.18-Ratified.pdf
5.16.1.3 SMART / Health Information (Log Identifier 02h)
Page 181: Figure 207: Get Log Page – SMART / Health Information Log
<pre>
Bytes: 63:48
Desctiption: Data Units Written: Contains the number of 512 byte data units the host has written to the controller as part of processing a User Data Out Command; this value does not include metadata. This value is reported in thousands (i.e., a value of 1 corresponds to 1,000 units of 512 bytes written) and is rounded up (e.g., one indicates that the number of 512 byte data units written is from 1 to 1,000, three indicates that the number of 512 byte data units written is from 2,001 to 3,000).
Refer to the specific I/O Command Set specification for the list of User Data Out Commands that affect this field.
A value of 0h in this field indicates that the number of Data Units Written is not reported.
</pre>
**Note**: The value is reported in thousands and the unit is 512 bytes.

ID: 248 (0xf8): Background program page count seems to be gotten by the get log page command smart / health information extended / smart cloud attributes log page physical media units written below.
https://www.opencompute.org/documents/datacenter-nvme-ssd-specification-v2-0r21-pdf
Page 15: 4.8.5 SMART / Health Information Extended (Log Identifier C0h)
<pre>
Requirement ID: SMART-1
Byte Address: 15:0
Field: Physical Media Units Written
# of Bytes: 16
Field Description: Shall contain the number of bytes written to the media; this value includes both user and metadata written to the user and system areas. It shall be possible to use this attribute to calculate the Write Amplification Factor (WAF).
</pre>
https://www.opencompute.org/documents/nvme-cloud-ssd-specification-v1-0-3-pdf
Page 9: 4.8.4 SMART Cloud Attributes Log Page
<pre>
Req ID: SMART-1
Byte Address: 15:0
Field: Physical Media Units Written
# of Bytes: 16
Field description: Contains the number of bytes written to the media; this value includes both user and metadata written to the user and system areas. It shall be possible to use this attribute to calculate the Write Amplification Factor (WAF).
</pre>
**Note**: It is needed to be supported the OCP speficication on the drive but those are for the datacenter NVMe SSD and the NVMe cloud SSD so it is possible to be not supported by the drive mentioned by this issue.

igaw, 2022-07-25:
Thanks for digging into it @ikegami-t! This is something Micron needs to implement from your research. Best chances is to contact Micron directly and ask for supporting this feature.

ikegami-t, 2023-03-23:
The nvme-cli features are implemented as the following commands. Seems the drive not supported the OCP feature and any vendor specific feature as a client and cosumer SSD but not an enterprise and data center SSD. Also not sure about other way so the issue may be needed to be closed as unsupported without any supporting.
- ID: 247 (0xF7): Host program page count
1. nvme smart-log
- ID: 248 (0xf8): Background program page count
1. nvme ocp smart-add-log
2. nvme micron vs-smart-add-log
3. nvme micron vs-smart-ext-log

igaw, 2023-03-24:
I agree, if it's vendor extension, this needs to be implemented in the plugins. I don't plan to work on this and it look no one else either. I am going to close this.

closed 2023-03-24 by igaw