linux-nvme/nvme-cli #1559: nvme-sanitize documentation conflicting statements

tuxrenci, 2022-06-08:
The documentation for **nvme-sanitize** provides conflicting statements. This problem exists in both the text and HTML versions, and also in the **man** page.

Specifically, the documentation says (highlighting added):
> The <device> parameter is mandatory NVMe character device (ex: **/dev/nvme0**)

However, in the **EXAMPLES** section, the syntax shown references the Block device (highlighting added):
> nvme sanitize **/dev/nvme0n1** --sanact=0x01

This is confusing. Is the first statement accurate, or is the usage example correct? The conflict needs to be resolved.

igaw, 2022-06-17:
I think the first example is correct. The spec (section `5.24 Sanitize command`) doesn't list any namespace id which could be provided. Also section `8.21 Sanitize Operations` says

> A sanitize operation alters all user data in the NVM subsystem such that recovery of any previous user data
> from any cache, the non-volatile media, or any Controller Memory Buffer is not possible. It is implementation
> specific whether Submission Queues and Completion Queues within a Controller Memory Buffer are altered
> by a sanitize operation; all other data stored in all Controller Memory Buffers is altered by a sanitize
> operation.

tuxrenci, 2022-07-23:
At the core of this problem is the phrase **_I think_**.

The purpose of documentation is to remove any ambiguity and turn **_I think_** into **_I know_**.

I suggest that specifications don't help; plenty of software deviates (intentionally or not) from underlying specifications (otherwise every TCP/IP stack implementation would perform identically, to the dismay of OS fingerprinters). Moreover, documentation exists not to recite the spec, but to tell the user what the software **actually does**.

igaw, 2022-07-23:
> The purpose of documentation is to remove any ambiguity and turn I think into I know.

To make it short, I don't know the answer, hence I am just guessing.

ikegami-t, 2022-11-11:
It is described by the specification as below.
> 3.3.3.1 Submission Queue Entry
> Figure 87: Common Command Format
> Namespace Identifier (NSID): ... If the namespace identifier is not used for the command, then this field shall be cleared to 0h. ...

So it seems thta if the block device is specified instead of the character device it is cleared to zero by the device controller software but actually it is depended on the device controller software behavior. I think that the device driver sets the name space ID value to submit the commnad to device if the block device is specified as same with other commands.
By the way is this really required to describe this by the document? (Sorry I am not sure actual the command behavior since on my local environment if the command executed the device data is erased.)

tuxrenci, 2022-11-12:
As I indicated originally,
> The conflict needs to be resolved.

If the solution is changing the documentation so the example is consistent with the rest of the document, well, that **does** resolve the conflict.

I suggest that it is worth noting (in the **man** page) that the results of specifying by block _vs._ character device may vary depending on the behavior of the device controller (and how the hardware adheres to/deviates from the actual NVMe specifications). I'm not suggesting that **nvme-cli** is responsible for the actions of the device controller - all the tool can do is provide a well-documented, NVMe-spec-valid method for interacting with the hardware.

ikegami-t, 2022-11-13:
Thanks for your comment and suggetion. Yes I think that the documentation conflict can be resolved by the examples fix only. About the devices behavior difference is described by the other command documention as below as probably you know.
> Documentation/nvme-format.txt
> ...
> DESCRIPTION
> ...
> If the character device is given, and the controller does not support
> formatting of particular namespaces (ID_CTRL.FNA bit 0 enabled), then all
> namespaces will be formatted. If FNA is disabled, then the namespace
> identifier must be specified with the 'namespace-id' option; specify a
> value of 0xffffffff to send the format to all namespaces. If the block
> device is given, the namespace identifier will default to the namespace
> ID of the block device given, but can be overridden with the same option.

igaw, 2022-11-16:
Updated the commit message with a little bit of information and dropped the generated doc files from the commit.

igaw, 2022-11-16:
Thanks everyone. I've applied the doc update patch.

closed 2022-11-16 by igaw