linux-nvme/nvme-cli #1560: nvme-format documentation missing information

tuxrenci, 2022-06-08:
The documentation for **nvme-format** is missing information on the process exit codes/output.

In the **DESCRIPTION** section, the first statement is highlighting added):
> For the NVMe device given, send an nvme Format Namespace admin command and **provides the results**.

However, there is no **RETURNS** section, or other information on exit codes or error messages. While various execution scenarios are mentioned, there is no information on how an invoking process might determine success/failure.

igaw, 2022-06-17:
Not a single command has a RETURN section. Please could someone provide a patch which addresses this?

All commands will behave the same, they will return 0 on success and 1 on failure:

```c
int main(int argc, char **argv)                                                                               
{                                                                                                             
        [...]
        return err ? 1 : 0;                                                                                   
}                                                                                                             
```

ikegami-t, 2022-07-04:
It seems that it is better to be described about the RETURN section by the nvme command help description only but not described by the sub commands since all sub commands return 0 on seccess and 1 on failure as same as mentioned.

igaw, 2022-07-08:
@ikegami-t provided a fix for this. Closing bug report.

closed 2022-07-08 by igaw