linux-nvme/nvme-cli #1580: set nvme ps0 to ps4,but the current doesn't change.

cljglll, 2022-06-21:
Hi, I 'm success to change power state of nvme from 0 to 4 with set feature command,but the current of the device is same as before.

In addition,I set power state of pcie from D0 to D1, or L0 to L1,the current of the device become low.

igaw, 2022-06-21:
Which devices? Could you post the commands and log form the shell. And if there are kernel logs add them also.

cljglll, 2022-06-22:
> Which devices? Could you post the commands and log form the shell. And if there are kernel logs add them also.

[image]
[image]

keithbusch, 2022-06-22:
Without seeing the power state descriptors, it's impossible to know what difference you might expect between ps0 and ps4.

cljglll, 2022-06-22:
> Without seeing the power state descriptors, it's impossible to know what difference you might expect between ps0 and ps4.
[image]
[image]
[image]
the NPSS and the power state descriptors may be OK。

keithbusch, 2022-06-23:
The spec provides a lot of caveats that allow a controller to ignore a user request to a non-operational power state.

cljglll, 2022-06-24:
> The spec provides a lot of caveats that allow a controller to ignore a user request to a non-operational power state.

I see the spec of nvme with V1.16,don't find the caveats.
NOPPME?this is set to 0.

cljglll, 2022-06-24:
> > The spec provides a lot of caveats that allow a controller to ignore a user request to a non-operational power state.
> 
> I see the spec of nvme with V1.16,don't find the caveats. NOPPME?this is set to 0.
In addition,NOPPME is 0 , but the power consumption exceeds the limit of ps1 and ps2 when I set the power state to 1 or 2.

keithbusch, 2022-06-24:
It sounds like your device may not be implementing the capability correctly. I think the hardware vendor may be in the best position to clarify.

cljglll, 2022-06-24:
> It sounds like your device may not be implementing the capability correctly. I think the hardware vendor may be in the best position to clarify.

I see.
Thank you for your suggestion.

closed 2022-06-24 by igaw

keithbusch, 2022-06-24:
> I see the spec of nvme with V1.16,don't find the caveats.

The statements saying the controller can ignore user requested power states to support background activity pretty much gives the controller full discretion to ignore this feature.