linux-nvme/nvme-cli #1598: JSON serialization does not properly escape non-printable characters, as required by JSON spec

ids1024, 2022-07-07:
For some reason `id-ctrl` is including a non-printable character on my SK Hynix SSD, and this then fails to parse with Rust's `serde_json`.

As mentioned in https://github.com/linux-nvme/nvme-cli/issues/1374, really the correct solution here is to use a good, conformation JSON implementation to serialize output.

igaw, 2022-07-08:
Indeed, it's time to rip out our home grown JSON implementation. I give it a go.

ids1024, 2022-07-08:
Okay, actually seems to be fixed in `master`. The version in the Ubuntu repositories (1.16) must be before the switch to `json-c`, and based on that comment on the other issue, and the fact that there's still json serialization code in `util/json.c` that doesn't implement this kind of escaping.