linux-nvme/nvme-cli #1612: enable

cljglll, 2022-07-15:


cljglll, 2022-07-15:
I enable the HMB,but the read/write rate does not improve.
[image]

igaw, 2022-07-18:
Could you post the output from `nvme id-ctrl -H /dev/nvme0?`

The interesting values are:

```
hmpre     : 0
hmmin     : 0
```

cljglll, 2022-07-22:
I found another problem。my cell has 4 SSD device.the HMB setting of these devices is the same.The reasonable?
[image]

igaw, 2023-03-01:
The host buffer allocation is done via `dma_alloc_coherent()` and as far I understand this interface, it's supposed give unique dma address from the PoV from the process.

The kernel code tells the device via the set feature command the address of the buffer:

```c
static int nvme_set_host_mem(struct nvme_dev *dev, u32 bits)
{
	u32 host_mem_size = dev->host_mem_size >> NVME_CTRL_PAGE_SHIFT;
	u64 dma_addr = dev->host_mem_descs_dma;
	struct nvme_command c = { };
	int ret;

	c.features.opcode	= nvme_admin_set_features;
	c.features.fid		= cpu_to_le32(NVME_FEAT_HOST_MEM_BUF);
	c.features.dword11	= cpu_to_le32(bits);
	c.features.dword12	= cpu_to_le32(host_mem_size);
	c.features.dword13	= cpu_to_le32(lower_32_bits(dma_addr));
	c.features.dword14	= cpu_to_le32(upper_32_bits(dma_addr));
	c.features.dword15	= cpu_to_le32(dev->nr_host_mem_descs);

	ret = nvme_submit_sync_cmd(dev->ctrl.admin_q, &c, NULL, 0);

```

I am bit confused, why we are passing in the dma address (virtual memory address) and no the physical address.

keithbusch, 2023-03-01:
dma_alloc_coherent gives you two addresses: one for use by the host, another for use by the device (sometimes they're the same value). For this feature, the host should never access the address after handing it to the device. The dma_addr_t is the address that the device can use to access the host memory, so that's what the command sends to the device.

igaw, 2023-03-01:
Thanks for explaining and I see how I got confused. I read the return value is assigned to `host_mem_descs_dma` but it's the `descs_dma` variable. 

The physical address might also be MMIO thus it's possible to see the same address twice, no?

keithbusch, 2023-03-01:
I'm not sure I understand what you mean by seeing the same address twice.

I suppose you could use MMIO space for this feature, but that's not a "Host Memory Buffer" anymore (not that the device could tell the difference).

igaw, 2023-03-02:
I understood that 'HMB setting of these devices is the same.' means that several devices got the same host memory buffer address. And I am trying to figure out if this could be a valid configuration. But maybe I misunderstood the report.

keithbusch, 2023-03-02:
Oh, no, that would not be expected unless you have an IOMMU that provides a remapped address specific to the device using it.

Perhaps I misunderstood something earlier. Comment said "I enable the HMB", but you're never supposed to touch that. This is the kernel's responsibility, the user should never muck with this setting. Is the user actually setting this feature from user space?

igaw, 2023-03-03:
I don't know. Anyway, this is not really a bug in nvme-cli. These resources are under control by the kernel. So nothing we can do here.  Closing report.

If there is still an issue I suggest to report to the nvme kernel mailing list.

closed 2023-03-03 by igaw