linux-nvme/nvme-cli #1613: Support for NVMe-MI transport

jk-ozlabs, 2022-07-15:
We started some discussion (in https://github.com/linux-nvme/nvme-cli/pull/1609) about how we could add support for a MI transport, but this is probably out of scope for that PR, so moving to a separate issue.

We should be able to abstract the details of the direct-ioctl interface provided by `libnvme`, allowing for other transport types.

As discussed in #1609:
 
> In terms of future work on this series: the main design decision will be how we allow different transports. Because the libnvme-mi API is split:
> ```c
>     int nvme_identify(int fd, ...);
> ```
> vs:
> ```c
>     int nvme_mi_identify(nvme_mi_ep_t ep, ...);
> ```
> We would need to split the library calls in the nvme-cli code, along the lines of:
> ```c
>     if (dev->type == NVME_DEV_DIRECT) {
>         nvme_identify(dev->fd, ...);
>     } else if (dev->type == NVME_DEV_MI) {
>         nvme_mi_identify(dev->ep, ...);
>     }
> ```
> ... but we could neaten this by adding suitable wrappers using the nvme_cli_ namespace:
> ```c
>     int nvme_cli_identify(struct nvme_dev *dev, ...)
> ```
> which handles the dev types appropriately.

I have a branch that implements this, using some Identify commands as an example. This is up at https://github.com/CodeConstruct/nvme-cli/commits/mi . The wrapper support is in [nvme-wrap.c](https://github.com/CodeConstruct/nvme-cli/blob/85b6a7476dc189f9752adbf564488f7edb47b37a/nvme-wrap.c).

This `mi` branch is a collection of a few things:

 * The `fixes` branch, already covered by https://github.com/linux-nvme/nvme-cli/pull/1608
 * A [`more-fixes` branch](https://github.com/CodeConstruct/nvme-cli/compare/fixes...CodeConstruct:nvme-cli:more-fixes), with a couple of minor tweaks before we start messing with the interfaces. No PR yet, but I'll submit one pending the outcome of `fixes`.
 * The [`dev-struct` branch](https://github.com/CodeConstruct/nvme-cli/compare/more-fixes...CodeConstruct:nvme-cli:dev-struct), which moves the majority of the fd handing to a new `struct nvme_dev`.
 * The [`mi` branch](https://github.com/CodeConstruct/nvme-cli/compare/dev-struct...CodeConstruct:nvme-cli:mi), which adds the MI support and the first set of wrappers

Let me know what you think; if this suits I can submit a PR and start expanding on the wrappers. Otherwise, happy to incorporate any feedback.

jk-ozlabs, 2022-07-19:
Sounds good to me.

My motivation here is to shift from adding all of the MI functionality to the `mi-mctp` example in the libnvme repo, to instead using the "proper" nvme-cli tool for this.

I'll keep working on my own branch in the meantime, as this helps to validate all the MI work in libnvme. The branches mentioned above might get a few updates, and will need to be tweaked during rebases to suit changes in the main nvme-cli branch.

jk-ozlabs, 2022-08-03:
OK, I see 2.1 is out - I've updated those four branches to the 2.1 base. Let me know if you'd like a PR.

igaw, 2022-08-03:
Yes, please send the PRs and let's integrate them early in this cycle so we have time to sort out possible fallouts.

jk-ozlabs, 2022-08-05:
OK., the `fixes` one should be good to go then; as #1608. I'll get working on the others.

jk-ozlabs, 2022-08-10:
I've simplified the branches a little:

* the [`fixes` branch](https://github.com/CodeConstruct/nvme-cli/tree/fixes) now includes the commits from `more-fixes`. This has just updated #1608 with two new commits.
* I've unified all of the dev-struct changes and MI support into a single (`mi`)[https://github.com/CodeConstruct/nvme-cli/tree/mi] branch, and created #1630

jk-ozlabs, 2022-08-25:
Closing, as the MI support is merged. Thanks!

closed 2022-08-25 by jk-ozlabs

igaw, 2022-08-25:
I have to thank. It's a real nice experience working with you together.