linux-nvme/nvme-cli #1615: power state:How can I get the current values of idle_power?

fantasy520, 2022-07-19:
ps    0 : mp:15.00W operational enlat:0 exlat:0 rrt:0 rrl:0
          rwt:0 rwl:0 idle_power:- active_power:-
ps    1 : mp:8.00W operational enlat:0 exlat:0 rrt:1 rrl:1
          rwt:1 rwl:1 idle_power:- active_power:-

ikegami-t, 2022-07-20:
From the code implemented as below it looks as IPS: Idle power scale is reported as 00b: Not reported for this power state for both PSs on the device. So it seems that it is not able to get the values on the device.
```
enum nvme_psd_ps {
	 NVME_PSD_PS_NOT_REPORTED	= 0,
...
static void print_ps_power_and_scale(__le16 ctr_power, __u8 scale)
{
	__u16 power = le16_to_cpu(ctr_power);

	switch (scale & 0x3) {
	case NVME_PSD_PS_NOT_REPORTED:
		/* Not reported for this power state */
		printf("-");
		break;
```

igaw, 2022-07-21:
Thanks @ikegami-t for answering the question.

closed 2022-07-21 by igaw