linux-nvme/nvme-cli #1616: [feature request] To prompt the user before zeroing out the whole device

frankang, 2022-07-19:
Recently I wiped out a disk using the command ` nvme format /dev/nvme0n1 -l 1 ` by mistake. (the correct device should be /dev/nvme1n1)
As SSD data is mostly unrecoverable, it would be better to prompt the user to confirm the target device name before executing this irreversible command.

birkelund, 2022-07-21:
I believe that nvme-cli already does that. Keith added this back in 2019.

igaw, 2022-07-21:
https://github.com/linux-nvme/nvme-cli/blob/f2cc94275081454acc0ae8fa21a093a8e3f00769/nvme.c#L4800

Yep, we have this already implemented in nvme 1.x and nvme 2.x.

closed 2022-07-21 by igaw

ikegami-t, 2022-07-21:
Sorry the issue was closed but if possible let me confirm below about this issue to make sure.
Is this issue to display the warning message only for the format command?
Since for example the sanitize command and the delete namespace command, other commands to set and write, delete also seem delete the SSD data.
So is it not necessary to display the same warinig message and add the force option?
I am not sure if reallly only the format command is required the confirmation mentioned and implemented.

frankang, 2022-07-22:
> https://github.com/linux-nvme/nvme-cli/blob/f2cc94275081454acc0ae8fa21a093a8e3f00769/nvme.c#L4800
> 
> Yep, we have this already implemented in nvme 1.x and nvme 2.x.

Hi, I tested this format command again but did not get any warning message or prompt, I'm using nvme-cli version=1.5-1ubuntu1.2 on Ubuntu 18.04. Could you confirm the minimum version that supports this precaution.

igaw, 2022-07-22:
We don't maintain downstream versions of nvme-cli. In this case you need to talk to your distribution vendor. Usually, there are backports available which ship newer versions of nvme-cli.

frankang, 2022-07-25:
I understand, I was thinkg

> Sorry the issue was closed but if possible let me confirm below about this issue to make sure. Is this issue to display the warning message only for the format command? Since for example the sanitize command and the delete namespace command, other commands to set and write, delete also seem delete the SSD data. So is it not necessary to display the same warning message and add the force option? I am not sure if really only the format command is required the confirmation mentioned and implemented.

This issue was intended for the format command only, but of course it would be better if other delete-related commands are covered by the same warning procedure.

igaw, 2022-07-25:
IMO, it's a low level tool and you should know what you are doing. And the question is where to draw the line anyway.  E.g. `nvme write` can do damage. So following the old unix style like `rm`, `dd`.

keithbusch, 2022-07-25:
Agreed, and you need sudo/root to use this utility for any commands that touch the hardware. This tool was intended to be easily used in more complex scripts, so every interactive warning makes that harder.

igaw, 2022-07-25:
`rm` does not ask for confirmation unless explicitly asked to do so (see -i options).
 
The format command is the big hammer, that's why it is handled differently. A single `nvme write` might destroy overwrite some data but not all. If don't like to be asked use `nvme format --force`.

keithbusch, 2022-07-25:
I was initially resistant to doing this for the format command, but it is special because it is a media changing command that can be sent to either a block or char handle. The linux kernel seriously botched the naming convention for these when multipathing was introduced, making simple mistakes way too common, and this tool was incorrectly blamed for the fallout.

`sanitize` is indeed similar, and perhaps could benefit with the same warning as `format`. That's just uncommonly supported in hardware, though, so no one's complained about sanitize accidents so far.