linux-nvme/nvme-cli #1618: nvme connect-all fails to process dhchap_ctrl_key

martin-gpy, 2022-07-20:
Turns out nvme connect-all processes the dhchap_key, but not the dhchap_ctrl_key. For e.g.

```
# nvme connect-all -t tcp -a 10.228.36.89 -w 10.228.32.190 -S DHHC-1:00:QaxYTIQLMUZOE+\/dmIForZfP5Mf9OX0Hiy8+uhG8kdOjrVxN: -C DHHC-1:00:2hfbQDMe7VIZngfJM6DbQ+5DKNzgUOjymfxl9LYs2wcZ7Qym:

# nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme1n1          /dev/ng1n1            80A2rNT89y0FAAAAAAAB NetApp ONTAP Controller                  1         524.29  MB / 524.29  MB      4 KiB +  0 B   FFFFFFFF

# cat /sys/class/nvme-subsystem/nvme-subsys1/nvme1/dhchap_secret
DHHC-1:00:QaxYTIQLMUZOE+/dmIForZfP5Mf9OX0Hiy8+uhG8kdOjrVxN:

# cat /sys/class/nvme-subsystem/nvme-subsys1/nvme1/dhchap_ctrl_secret
none
```

martin-gpy, 2022-07-20:
But this works for nvme connect though. For e.g. 

```
# nvme connect -t tcp -a 10.228.36.89 -w 10.228.32.190 -n nqn.1992-08.com.netapp:sn.94d3e4c5f3ed11eca5871c666dbe2b2a:subsystem.ss_tcpnvme_2_1 -S DHHC-1:00:QaxYTIQLMUZOE+\/dmIForZfP5Mf9OX0Hiy8+uhG8kdOjrVxN: -C DHHC-1:00:2hfbQDMe7VIZngfJM6DbQ+5DKNzgUOjymfxl9LYs2wcZ7Qym:

# nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            80A2rNT89y0FAAAAAAAB NetApp ONTAP Controller                  1         524.29  MB / 524.29  MB      4 KiB +  0 B   FFFFFFFF

# cat /sys/class/nvme-subsystem/nvme-subsys0/nvme0/dhchap_secret
DHHC-1:00:QaxYTIQLMUZOE+/dmIForZfP5Mf9OX0Hiy8+uhG8kdOjrVxN:

# cat /sys/class/nvme-subsystem/nvme-subsys0/nvme0/dhchap_ctrl_secret
DHHC-1:00:2hfbQDMe7VIZngfJM6DbQ+5DKNzgUOjymfxl9LYs2wcZ7Qym:
```

martin-gpy, 2022-07-20:
This is a little more complicated that I thought. 

It looks like for nvme connect-all to handle this, we would have to pass ctrlkey from nvmf_discover() -> __discover() -> nvmf_connect_disc_entry(), and eventually call nvme_ctrl_set_dhchap_key() here just before the call to nvmf_add_ctrl() within nvmf_connect_disc_entry().

So that means the above APIs i.e. nvmf_discover() & __discover() in nvme-cli's fabrics.c as well as nvmf_connect_disc_entry() in libnvme's fabrics.c would have have to be modified for handling the extra ctrlkey param. So would that be acceptable? I can create separate pull requests for the above if you folks agree.

igaw, 2022-07-20:
We really should try not to break the API at this point.

martin-gpy, 2022-07-20:
> We really should try not to break the API at this point.

Thought so. But then, how else can we fix this?

igaw, 2022-07-25:
I think we could place the key in `struct nvme_fabric_config`. As long we get the padding and aliment correct it should backwards compatible. Luckily the structure has a nice layout:

```
struct nvme_fabrics_config {
        char *                     host_traddr;          /*     0     8 */
        char *                     host_iface;           /*     8     8 */
        int                        queue_size;           /*    16     4 */
        int                        nr_io_queues;         /*    20     4 */
        int                        reconnect_delay;      /*    24     4 */
        int                        ctrl_loss_tmo;        /*    28     4 */
        int                        fast_io_fail_tmo;     /*    32     4 */
        int                        keep_alive_tmo;       /*    36     4 */
        int                        nr_write_queues;      /*    40     4 */
        int                        nr_poll_queues;       /*    44     4 */
        int                        tos;                  /*    48     4 */
        _Bool                      duplicate_connect;    /*    52     1 */
        _Bool                      disable_sqflow;       /*    53     1 */
        _Bool                      hdr_digest;           /*    54     1 */
        _Bool                      data_digest;          /*    55     1 */
        _Bool                      tls;                  /*    56     1 */

        /* size: 64, cachelines: 1, members: 16 */
        /* padding: 7 */
};
```

This means we should be able to use the same approach as we have for the `struct *_args`.

igaw, 2022-07-26:
The idea to extend `struct nvme_fabrics_config` is not so great. Unfortunately, we forgot to add the 'size_args' member and that means we can't really tell what version of nvme_fabrics_config is passed in. I played with the idea to sneak in a version field after the last bool (tls) but this is a huge hack and I don't think we should go down this route.

My next attempt was to introduce a `struct nvme_fabrics_config2` which wrapps round the old one.  This is also not so great because we have to function which use currently `struct nvme_fabrics_config`. So we would get `nvmf_default_config2`, `nvmf_update_config2`, `nvmf_add_ctrl2`... https://github.com/linux-nvme/libnvme/commit/d06407d522b20616087f18e5226f0dd06d93e2e1

My next idea is to introduce just `nvmf_connect_disc_entry2` which is will accept a `struct_args`. And don't touch the
rest of the API.

hreinecke, 2022-07-26:
Please check if this isn't already resolved with PR #398.

hreinecke, 2022-07-26:
Sigh. One time I get it right.
I meant https://github.com/linux-nvme/libnvme/pull/398

igaw, 2022-07-26:
@hreinecke this bug tracking system allows editing posted comments :)

Anyway, the fix in https://github.com/linux-nvme/libnvme/pull/398 is working only when we have a JSON config file. If you provide only command line arguments I think we still need something like I am working on.

igaw, 2022-07-26:
@martin-gpy could you give #1622 a test run? Thanks.

hreinecke, 2022-07-26:
As mentioned in the other thread, the original idea was to have the controller key as part of the controller tree structure.
Why can't we use this method?

igaw, 2022-07-28:
Thanks for the quick test. Back to the drawing board.

igaw, 2022-08-01:
Unfortunately, this use case (providing the key via command line) doesn't really scale at all. In case we need to pass several controller host keys the command line configuration gets pretty fast too complex. So it's better to drop the command line argument for providing the controller host key and defer to the JSON config file configuration approach.

PR #1626 (2022-08-01): fabrics: Remove dhchap-ctrl-secret from discover/connect-all
The dhchap-ctrl-secret is per controller but discover and connect-all
operate on several controllers. As this get's very fast very
complicated how to express this on the command line, we defer here to
JSON config file in this case.

Fixes: #1618
closed 2022-08-01 by igaw