linux-nvme/nvme-cli #1623: NVME_NS_FLBAS_HIGHER_MASK Define Wrong??

jeffreyalien, 2022-07-28:
I noticed this definition seems wrong while testing the Enhanced PI changes:
src/nvme/types.h:	NVME_NS_FLBAS_HIGHER_MASK	= 3 << 5,

I think it should be:
NVME_NS_FLBAS_HIGHER_MASK	= 3 << 4

Am I correct?  Or missing something?   I assume there aren't many (if any) devices that support more then 16 lbaf's yet which would be the reason no one has hit this issue yet.  If correct, I'll fix and generate a pull request.

birkelund, 2022-07-28:
No, `3 << 5` is correct.

It translates to the bitmask `1100000`. The lower 4 bits (bits 0 through 3) are the least significant 4 bits of the LBA format index, bit 4 indicates if the LBA is extended and bits 5 and 6 are the most significant 2 bits of the LBA format index. The confusion comes from the fact that the "extended LBA" bit is spliced in between the index bits.

keithbusch, 2022-07-28:
In hindsight, the original decision to place the metadata flag adjacent to the format index when it was only 3 bits lead to this ugly encoding. :)

jeffreyalien, 2022-07-29:
Thanks for clarifying.  I wasn't aware of the bit 4 (LBA is extended) definition.  

The nvme cli format command accounts for bit 4 when filling in the LBAU data but I don't think it's very clear to the user when passing the lbaf parm.  I may look into updating the man page.  I don't see any references to extended lbaf's in there.

jeffreyalien, 2022-07-29:
Closing since the definition is correct and no changes necessary.

closed 2022-07-29 by jeffreyalien